MarlinFirmware/Marlin #5354: Dual Z endstops and Z probe

Nocturnal42, 2016-12-03:
What I would like to be able to do, is essentially home twice. Once up to the dual endstop, leveling the axis, and then down again to the bed to zero Z. Now I know this could be done manually, using something like G28 Z0; G30 S-1 and then use M206 with the offset obtained. 

I'm assuming adding something like this would do the trick...
    set_home_offset(Z_AXIS, probe_pt(current_position[X_AXIS] + X_PROBE_OFFSET_FROM_EXTRUDER, current_position[Y_AXIS] + Y_PROBE_OFFSET_FROM_EXTRUDER, 0, 1));

Though I haven't checked types, so the values being passed may not be right. Is this the best / most appropriate way of going about it?

thinkyhead, 2016-12-03:
Let's see. The `probe_pt` function will return the current Z position when the probe contacts the bed. You actually need to add your `zprobe_zoffset` from that value to determine how far off the bed is from the expected position. For example, if your `zprobe_zoffset` is -2.5 and the probe result is 2.6, it means the bed is 0.1mm higher than expected. [ 2.6 + (-2.5) = 0.1 ]. Your home offset should then be set to the negative of that value, -0.1.

So… I believe this is what you need:

```cpp
set_home_offset(
  Z_AXIS,
  -(probe_pt(
      current_position[X_AXIS] + X_PROBE_OFFSET_FROM_EXTRUDER,
      current_position[Y_AXIS] + Y_PROBE_OFFSET_FROM_EXTRUDER,
      0, 1
    ) + zprobe_zoffset
  )
);
```

Nocturnal42, 2016-12-03:
Well I was close :)

I'll stick that in the relevant bit of G28 and see how it goes.

Nocturnal42, 2016-12-06:
This is what I ended up with in gcode_G28() before the home Z towards the bed...

```      if (home_all_axis || homeZ) {

         const float zpos = probe_pt(
            round(Z_SAFE_HOMING_X_POINT),
            round(Z_SAFE_HOMING_Y_POINT),
            0, 1
         );
         if (zpos != NAN) set_home_offset(Z_AXIS, -(zpos + zprobe_zoffset));
      }
```
Since it's not safe if X/Y aren't homed, I should probably have it only checking for home_all_axis, but I'm fine with it as is. I dumped the X/Y probe offsets for 2 reasons. Firstly, because my nozzle is my probe atm, so the offsets are 0. Second, a quick look at the code for probe_pt() suggests it is already doing probe offsets.

closed 2016-12-06 by Nocturnal42