MarlinFirmware/Marlin #5356: [Question] Manuel probe

boelle, 2016-12-03:
To get semi automatic bed level going, i wondered if i could make a clip-on-nozzle probe thing?

ie before bed level i clip on the endstop to the nozzle and after i remove it..

should i just configure it at a FIX_MOUNTED_PROBE ?

Nocturnal42, 2016-12-03:
I'm not an expert, but there seems to be no reason it shouldn't work. It's not that different to a sled mounted probe, or a servo mounted probe. Your just deploying it manually. The only issue I see, is that it really needs to clip onto the exact same spot every time, and it needs to be tightly fit so there is no wiggle. Otherwise your probe offset settings (and probe readings) will be off.

boelle, 2016-12-03:
hehe

i will not use it that often. Since it will be saved to eeprom i dont see the logic in doing it at every print.

http://imgur.com/a/NK0rH

heater block goes in to the bigger part and a tact switch slides in the smaller part

still need to fiddle with it

epatel, 2016-12-03:
This is what I have been using for a long time now. http://www.thingiverse.com/thing:1120142

boelle, 2016-12-03:
same princple, will print 1st version now and see how it fits

Roxy-3D, 2016-12-03:
Wookie brought up UBL without using a Z-Probe.   He followed these steps:  https://github.com/MarlinFirmware/Marlin/issues/4948#issuecomment-256203643

Rather than using the Z-Probe to initially populated the mesh with reasonable values, you assume 0.0000 is a reasonable value.   And then you start the cycle of doing a Mesh Validation Pattern and edit the locations that are off.    After 2 or 3 iterations...  You will have an accurate mesh that is alive and functioning every the printer powers up.

boelle, 2016-12-03:
i'm initial more worried about the setup in configuration.h

first print of tact switch holder was almost spot on. doing the next one now and will start to make the wiring to ramps after dinner

boelle, 2016-12-03:
@Roxy-3D read the comment and it seems like a lot of work. Cant it be automated?

boelle, 2016-12-03:
my probe... http://imgur.com/a/UWrwy

Roxy-3D, 2016-12-03:
Well...   I guess it depends on what you call a lot of work.   Given how much time it takes to put a servo controlled switch onto a print nozzle, it is less work than that.   And the benefit is your machine knows all about the print surface for future use.

boelle, 2016-12-03:
I guess i just have to configure mine as a fixed probe and run automatic correction like it was a inductive probe then? of course need to mount it on the nozzle first

with automatic mesh level on will it only execute on a G29? i still have my Z max endstop that i intend to use before each print

Roxy-3D, 2016-12-03:
> with automatic mesh level on will it only execute on a G29? i still have my Z max endstop that i intend to use before each print

Which branch are you thinking of using?  G29 behaves differently in UBL.   In fact...  I don't even have it in my startup GCode produced by the Slicer because the printer powers up and knows the shape of the bed surface.

FHeilmann, 2016-12-03:
You'd still need some means of homing Z right? (Just curious)

Roxy-3D, 2016-12-03:
Yes.   You need to be able to home the machine.   But in the case I'm suggesting, the printer will know the bed is not perfectly flat and will be able to compensate for that.    And the  UBL branch has QUICK_ACCESS_TO_Z_BABYSTEPPING available several different ways.   So lately, I've been setting my printer to home with the nozzle up a little bit higher than you would normally want it.   Maybe .25mm too high.   

Because what I do is I just press the Quick Access button (or you can double click the encoder wheel) and I just lower the nozzle to the point it needs to be.   But of course...  You can have it just home to that location too.

boelle, 2016-12-03:
"You'd still need some means of homing Z right? (Just curious)"

max endstop :-D

boelle, 2016-12-03:
"Which branch are you thinking of using?"

RCBugfix

boelle, 2016-12-03:
done probe minus wires: http://imgur.com/a/yT2mY

boelle, 2016-12-04:
my config so far: http://pastebin.com/1YTfbxUq

still need to find the cords where it can go without hitting clips etc

boelle, 2016-12-04:
http://pastebin.com/sE6uuZEv

that is my config as far i can get it. Any obvious errors?

NOTE: my probe offset is really 0.48. I figured it by homing and then lowered Z until i get a triggered state and then up again until it just changes state.

boelle, 2016-12-04:
@thinkyhead @Roxy-3D 

http://pastebin.com/3yQDw6iK

my config above... G29 is almost "normal", i have a suspission that i have made a config error

and the reason is that the first probe is done 20mm from the endstop where i would like it to be done.

I have tried a few things but came up empty

boelle, 2016-12-04:
`//    +-- BACK ---0
//    |           |
//  L |    (+)  | R
//  E |           | I
//  F | (+) PN (-) | G       X endstop
//  T |           | H
//    |    (-)    | T
//    |           |
//    +-- FRONT --+`

Y Endstop

Roxy-3D, 2016-12-04:
.002 mm standard deviation is good.  But I'm not sure that number is really what you are going to see.  Because you clip the probe on and off...  You are going to get extra error from that operation that is not being measured by the M48.  The M48 is only measuring the error once the probe is in place.

But...  You will know soon enough if you are getting extra error.    And in fact, any additional error will be because the Z-Offset is a little too high or low.   That can be corrected with the Z-BabyStepping while the skirt is being printed.

boelle, 2016-12-04:
And can you shine light on why it starts by moving 20mm away from the X endstop?

boelle, 2016-12-04:
but yes... my plan is to use this only once in a while and the Z-offset will be adjusted to give a perfect first layer

thinkyhead, 2016-12-05:
If your 0,0 is in the back-right corner, then when looking at your nozzle from the front, this setting means your probe should be on the right side of the nozzle:
```
#define X_PROBE_OFFSET_FROM_EXTRUDER -20
```
Is that correct?

boelle, 2016-12-05:
nope, its on the left

but when i change that to +20 i get:

 "The given LEFT_PROBE_BED_POSITION can't be reached by the Z probe."

since the nozzle is at -2 when homed and the switch is at 18 its over the bed

boelle, 2016-12-05:
on ascii drawing it looks like this: http://imgur.com/a/EKU0G
and on a picture: http://imgur.com/a/J5ikk

x endstop is in top right corner. Z max is at top of the Z spindle most far away.

probe switch is just to the left of the nozzle

thinkyhead, 2016-12-05:
> The given LEFT_PROBE_BED_POSITION can't be reached by the Z probe

The minimum position your probe can reach, therefore is 18.

boelle, 2016-12-05:
ahh i just read this as regarding the nozzle

// Set the boundaries for probing (where the probe can reach).

ie numbers where the nozzle are that the probe can reach. Not sure if it can be rephrased so dummies get it :-D

boelle, 2016-12-05:
Q: is M48 P1 a "reasoble" way to mesure Z height on a single spot?

I want to use it as a tool to get my 2 Z spindles in sync and X axis parallel to the bed at the 2 fixed screws at the back of the bed

boelle, 2016-12-05:
never mind... i ended up adjusting with feeler blade 

and came up with this:

Bed Level Correction Matrix:
+1.000000 +0.000000 +0.000540
-0.000003 +0.999989 +0.004732
-0.000540 -0.004732 +0.999989

the first line with 3 numbers are that also the first spots on the plate ? or is it in reverse?

Roxy-3D, 2016-12-05:
> Q: is M48 P4 a "reasonable" way to measure Z height on a single spot?

No.   M48 doesn't care about where the Zero is.   It pretty much is just looking to see if the same numbers are returned with each probe.   

> the first line with 3 numbers are that also the first spots on the plate ? or is it in reverse?

> on the other hand i could view this as its almost same height at the 4 corners.. -0.000540 and +0.000540 is 0 half way between them. +1.000000 and +0.999989 is as close to the same number as a human can adjust.

> but am i right or totally wrong?

No.  That matrix has almost nothing to do with the height of individual points on the bed.  It is a 'Translation Matrix' that gets multiplied by a 'Coordinate Matrix' to produce a rotation into a new coordinate system.    That diagonal line of numbers that is almost equal to 1.000 is saying take mostly the original component (x, y or z) but add in a little bit of some of the other components to form the new coordinate.

thinkyhead, 2016-12-05:
> It is a 'Translation Matrix'

A 'Transformation Matrix' actually. In this case a 3-dimensional rotation matrix. See https://en.wikipedia.org/wiki/Transformation_matrix

boelle, 2016-12-05:
and why is it that the Z spindles does not move after a G29  and M500 ?

thinkyhead, 2016-12-05:
Due to spindle malfunction.

boelle, 2016-12-05:
no they work fine... G0 Z10 does move it down if its above 10mm 

what i mean is that if i move X or Y    Z does not move

boelle, 2016-12-05:
dooh

http://marlinfw.org/docs/features/auto_bed_leveling.html

last section gives the reason :-(

will that ever change? ie save the results

Roxy-3D, 2016-12-05:
I believe the mesh is now saved for Bi-Linear in RCBugFix.   And UBL allows multiple Mesh to be saved saved because some people use several different pieces of glass to print on.

thinkyhead, 2016-12-06:
> I believe the mesh is now saved for Bi-Linear in RCBugFix. 

No, it is not. I will have to implement that later.

boelle, 2016-12-06:
any time frame for that?

thinkyhead, 2016-12-06:
No there is not.

boelle, 2016-12-06:
oki, closing this one as i got it as close to "fixed" as i could get.

closed 2016-12-06 by boelle