MarlinFirmware/Marlin #5361: Z (nozzle) height "wrong" after G29

Ianmcmill, 2016-12-03:
**Initial configuration:**
XY mechanical endstops
`M206 X-8.00 Y-13.00 Z0`

Using inductive probe:
`#define FIX_MOUNTED_PROBE`

Z endstop = Inductive probe; no mechanical endstop installed:
`#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

Probe offset from extruder:
`#define X_PROBE_OFFSET_FROM_EXTRUDER -32  // X offset: -left  +right  [of the nozzle]`
`#define Y_PROBE_OFFSET_FROM_EXTRUDER 6  // Y offset: -front +behind [the nozzle]`
`#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]`

Auto Bed Leveling method:
`#define AUTO_BED_LEVELING_BILINEAR`

**Problem:**
After issuing G29 and a completed BILINEAR bed probing the nozzle does not touch the bed anymore

**Behavior and offset configuration process:** (how I did it)
After issuing G28 the printer correctly hits X and Y endstops, travels to the middle of the bed (Z_SAFE_HOMING enabled) and perfoms a Z homing using an inductive probe.
Nozzle is not touching bed, i.e. no layer adhesion.
Then Z offset is being set via:
`M206 Z0.55` empiric value by issuing G1 Z0.1(+) in 0.1 steps
`M500`
When issuing `G1 Z0`  the paper test is fine. Paper fits between nozzle and bed for good layer adhesion.
After that I issue `G29`. Once the leveling is completed I move back to the center of the bed and perfom a `G1 Z0`. But the nozzle is too far away from the bed. 
If I issue `G1 Z-0.55` which is the offset I set with `M206 Z0.55` the nozzle touches the bed perfectly.

What am I doing wrong or overlooking?

Attached is my Configuration.h and a G28 / G29 debug log and an image of my extruder build.

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/628865/Configuration.h.txt)
[G28_G29_debug_log_rc7bugfix_dec2016.txt](https://github.com/MarlinFirmware/Marlin/files/628866/G28_G29_debug_log_rc7bugfix_dec2016.txt)
[image]

Ianmcmill, 2016-12-03:
Oh did I mess up Home offset (M206) and Z probe offset (M851)???

Ianmcmill, 2016-12-03:
When I offset "Home Z" with `M206 G0.55 M500 `the nozzle moves closer to the bed.
When I offset "Home Z" with `M206 G-0.55 M500` the nozzle moves away from the bed.

Shouldn't this be just the other way around?

Ianmcmill, 2016-12-03:
Okay I did mess up M206 and M851. 
This might be kind of confusing for some people. And me. 

My current working configuration is:
`Home offset (mm)`
`M206 X-8.00 Y-13.00 Z0.55`
`Z-Probe Offset (mm):`
`M851 Z-0.55`

It is working but I am not sure i.e. confused why this works with Home offset set to Z 0.55 and the Z-Probe offset is set to Z -0.55.

Maybe this could be mentioned or explained more clearly in the comments?

thinkyhead, 2016-12-03:
You shouldn't need `M206` if your value is correct with `M851`. Are you homing exclusively with the Z probe? Perhaps it's a bug in the way that homing with a probe is done. Or maybe the M851 offset is wrong.

Enable `DEBUG_LEVELING_FEATURE` and then before you do `G28` issue `M111 S247` to enable all logging. Start out with `M206 Z0` and assuming the probe distance is correct, also do `M851 Z-0.55` Do your `G28` followed by `G29 V4`. Then paste the log output here or on a site like pastebin for examination.

thinkyhead, 2016-12-03:
Are you using the tagged `1.1.0-RC7` in the `RC` branch or are you using the `RCBugFix` branch?

You also shouldn't need to use `M206` for X or Y. You can just configure your home position to accommodate that situation. For example, if the nozzle is left of the bed by 8mm when you home X to the minimum endstop, then just configure your `X_MIN_POS` to -8. To make sure the probe is over the bed when you home Z, use the `Z_SAFE_HOMING` option.

Ianmcmill, 2016-12-03:
I am using RCBugFix 1st Dec 2016! Not RC7.

I don't think it's a bug. I rather think it's just misunderstood by me.
I am going with the X_MIN_POS as offset the next time I am going to update the firmware.
Thank you thinkyhead!

thinkyhead, 2016-12-04:
@Ianmcmill I hope it makes more sense as you get it configured. Calibration and configuration is always a bit of a hassle to get dialed-in, but once set up the way you like it, hopefully never have to think about it again.

Ianmcmill, 2016-12-04:
Yeah it makes more sense now. And that ABL is somehow cool. Albeit it was a big hurdle for me to get it going. But that G29 magic-movey-movey stuff is just too cool :)

thinkyhead, 2016-12-04:
I'm going to close this, but if you have more comments, you can continue to post.

closed 2016-12-04 by thinkyhead