MarlinFirmware/Marlin #5366: Out of a sudden: Error: No line /w checksum + resend + x-carriage crash into MAX X pos

Ianmcmill, 2016-12-03:
Hello,

after alot of flashing to get ABL running (10-20 flash cycles) I suddenly get this error log. I never had this error before with other firmwares:
1.1.0 RCBugfix as of 1st Dec. 2016
extract:
`22:23:14.384 : Error:No Line Number with checksum, Last Line: 2568`
`22:23:14.509 : Error:checksum mismatch, Last Line: 2570`
`22:23:14.509 : Resend: 2571`
`22:23:14.514 : Error:No Line Number with checksum, Last Line: 2570`
`22:23:15.102 : Setting Print and Travel Acceleration: 800.00`
`22:23:15.148 : X:110.12 Y:102.20 Z:1.04 E:0.00 Count X: 9189 Y:10940 Z:6308`
`22:23:16.012 : Error:checksum mismatch, Last Line: 2590`
`22:23:16.012 : Resend: 2591`
`22:23:16.017 : Error:Line Number is not Last Line Number+1, Last Line: 2590`
`22:23:16.017 : Resend: 2591`
`22:23:21.820 : Printer reset detected - initalizing` <- manually pressed reset button on RAMPS 1.4

Just before the reset detected debug the printhead runs straight into max X and crashes into the X-Idler not slowing down. Full force. Only a reset stops the motor.

I was printing two small z-wobble things.

Gcode, repetier-host log and marlin config attached.
What could this be? I have seen posts about those line number things which are/were related to a bug. Don't know if this applies here too.

[z_screw_isolator.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/629141/z_screw_isolator.gcode.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/629147/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/629148/Configuration_adv.h.txt)
[crash_log.txt](https://github.com/MarlinFirmware/Marlin/files/629139/crash_log.txt)

Ianmcmill, 2016-12-03:
Update:

Seems like lowering the baudrate from 250000 to 115200 fixed it.
No more Errors and no more crashes.

thinkyhead, 2016-12-04:
That's the key. Usually this kind of error is caused by a noisy USB connection.

closed 2016-12-04 by thinkyhead