MarlinFirmware/Marlin #5367: Extruder motor sometimes doesnt run with SWITCHING extruder

alexxy, 2016-12-03:
Hi!

I try to setup Dondolo switching extruder with current (RCBugfix, also tryed RC branch) Marlin. However sometimes i see that when priner switched to T1 extruder motor doesnt move. My board is MKS Base2 (same with MKS Base but with different stepper drivers). I use 996R MG servo with extrnal 5v power suply to switch heads.

thinkyhead, 2016-12-04:
@alexxy Interesting. Just sometimes? But most of the time it switches direction and keeps going properly…. Do you find the problem exists in general testing with no filament inserted (using `M302 S0` to allow cold extruding)? Or does it seem to occur only when running a print job?

thinkyhead, 2016-12-04:
Just checking through the code for `SWITCHING_EXTRUDER` and it's very simple. I'm inclined to think that what you're seeing is potentially due to the stepper driver overheating. Do you have a fan on your electronics and heatsinks on your stepper drivers?

Roxy-3D, 2016-12-04:
>  I'm inclined to think that what you're seeing is potentially due to the stepper driver overheating. Do you have a fan on your electronics and heatsinks on your stepper drivers?

Can you turn down the power to your stepper motors?  Or do you need all the power you are giving them?   Because having the power turned up high is how you get the thermal overload protection on the stepper motor chips to cut out at times.

alexxy, 2016-12-04:
I can lower power to extruder stepper and try again. But its quite surprising that it sometimes doesnt turn only in one direction (inverse one)...

alexxy, 2016-12-04:
I checked temperature of stepper, its quite cold. So its not related to overheating...

thinkyhead, 2016-12-04:
Well the thing is, there's hardly anything to the switching behavior. Here's the sum total of all the logic that is used to move the stepper in one direction versus the other:
```cpp
#if ENABLED(SWITCHING_EXTRUDER)
  #define E_STEP_WRITE(v) E0_STEP_WRITE(v)
  #define NORM_E_DIR() E0_DIR_WRITE(current_block->active_extruder ?  INVERT_E0_DIR : !INVERT_E0_DIR)
  #define  REV_E_DIR() E0_DIR_WRITE(current_block->active_extruder ? !INVERT_E0_DIR :  INVERT_E0_DIR)
```

thinkyhead, 2016-12-04:
If the stepper was mis-wired that would potentially cause trouble also. But when mis-wired, usually the stepper will only move in one direction, then makes chucking noises when you try to move the other. So I suppose that's not it either…

Post your `Configuration.h` file and maybe there's something in there that will give us some clue. Given how simple this function is, I still tend to think it has something to do with wiring, noise, or something like that.

You might also try flipping the plug on your E stepper to see if it goes in the opposite direction. I don't know if that will reveal anything, but it's worth a try just to see if any new anomalies pop up.

alexxy, 2016-12-04:
My configuration.

https://gist.github.com/alexxy/8142ef7135176e875181252af48cdc8b

Now i'm lowered power of motors. Another thing is that i hear noise from MG 996R servo.

thinkyhead, 2016-12-04:
@alexxy Try disconnecting the servo. The extruder won't switch sides, obviously, but you can check to see whether the stepper motor moves.

alexxy, 2016-12-04:
Ok, lets see what will heppened

Blue-Marlin, 2016-12-04:
A 'noisy' servo could be a sign for - it can't reach the ordered angle. That would mean it's drawing a lot of current - maybe enough to let drop the power supply voltage - what could cause any error.
Thy to adjust the angles until the servo is not 'noisy' any more.

alexxy, 2016-12-04:
Servo uses separate 5v psu, its rated as 5A. It also poweres rpi board that runs octoprint. I'll check servo angles and ajust them.

boelle, 2016-12-08:
are the ground level of the different PSU's the same?

mattgaspar, 2017-07-11:
@alexxy Did you ever figure this out?
I'm have the same/similar issue.

I just modded my prusa clone to dual filament using single nozzle and single extruder (switching extruder and Y filament merger).  Everything seems to work with the tool change script, the first filament gets retracted, the servo switches to the second filament, it gets pushed down to the hotend and then the extruder stops responding (no errors, x/y moves like it's still working but no filament comes out)

Upon further testing, the "second extruder" (T1) only accepts 1 extrude command.  It won't move the extruder again until the system is restarted.  If I switch to T0, the servo switches and the same extruder now works again, but switching back to T1 doesn't work until reboot.

I just updated to Marlin 1.1.4 and same issue.

Here's the relevant parts of the extruder section from my Marlin config
`// @section extruder

// This defines the number of extruders
// :[1, 2, 3, 4, 5]
#define EXTRUDERS 2

// For Cyclops or any "multi-extruder" that shares a single nozzle.
#define SINGLENOZZLE

// A dual extruder that uses a single stepper motor
#define SWITCHING_EXTRUDER
#if ENABLED(SWITCHING_EXTRUDER)
  #define SWITCHING_EXTRUDER_SERVO_NR 1
  #define SWITCHING_EXTRUDER_SERVO_ANGLES { 30, 75 } // Angles for E0, E1[, E2, E3]
  #if EXTRUDERS > 3
    #define SWITCHING_EXTRUDER_E23_SERVO_NR 1
  #endif
#endif
`

Switching servo is set to 1 because BLTouch is on servo 0 (these are both working)

I tried setting the E1 pins to the E0 values and it didn't make a difference (I assume enabling SWITCHING_EXTRUDER was already doing this for me)

Let me know if I'm missing any relevant details, I'm not familiar with any of the Marlin debugging functions.  I'm a software developer but haven't really looked through any of the Marlin code yet!

alexxy, 2017-07-20:
Well, I updated my setup, and run agian into this issue.

Interesting that if i manualy issue T0/T1 and then extrude from octoprint, both extruders work. However second one cannot print. I cant figure why

alexxy, 2017-07-22:
Some update: 

Now electronics in printer is MKS Gen 1.4
Symptoms:
I can  manualy extrude with octoprint (switch to T1, and extrude)
However when printing its not possible for T1 (head moves, but extruder stepper doesnt)

mattgaspar, 2017-07-22:
That was it!  Set DISABLE_INACTIVE_EXTRUDER to false and it's working correctly now.  
I thought I had tried that and then I noticed there is another setting I might have tried in the Configuration_adv.h file
#define DISABLE_INACTIVE_E true
Don't these settings control the same behavior?  Right now I have them set to opposite values and the switching extruder is working!

Seems like there should be some conditional statements added to set these automatically or at least a sanity check.

FYI - I'm using an MKS board as well (Gen 1.5).  I was about to try switching to a plain Ramps board setup this weekend, thankfully I don't need to do that now!

mattgaspar, 2017-07-25:
Now that the switching extruder is working I have been trying to test with a second extruder, setting the extruders to 3 or 4.  I have not been able to get the second extruder to run (with SWITCHING_EXTRUDER & SINGLE_NOZZLE enabled).

If I disable SWITCHING_EXTRUDER and set extruders to 2, the extruder on E1 works fine but if I enable SWITCHING_EXTRUDER and set extruders to 3 which should have T2 using the extruder on E1 nothing happens.  The T2 command returns active extruder 2 but then if I try to issue a command to extrude or retract (with cold extruder checking disabled) it appears to accept the command but nothing happens and no error messages.

I looked through some of the SWITCHING_EXTRUDER parts of the source code and I couldn't see any problems from the parts I could understand.
Is there anything I could do to debug the problem or anything I'm missing?  The configuration seems pretty straightforward since I know E0 and E1 work fine without switching extruder enabled and the only configuration change was to enable switching extruder and increase the extruder count to 3 or 4.

wbarber69, 2018-01-10:
i know this is an old thread but did any of you think about temperatuer logic?  maybe the second extruder isnt working because the temperature for T1 is not past 180

thinkyhead, 2018-02-11:
@mattgaspar — We haven't done a patch yet to make sure `DISABLE_INACTIVE_EXTRUDER` doesn't interfere with `SWITCHING_EXTRUDER`. And we might still need to investigate your issue when setting `EXTRUDERS` to 3 or more.

But before we proceed, please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we accidentally fixed these issues. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

mattgaspar, 2018-02-12:
@thinkyhead - I just tested with both `bugfix-1.1.x` and `bugfix-2.0.x`
Still encountered the same problem as I described above (Jul 25).

This can be easily tested without a switching extruder (it should just reverse the motor when using T1/T3)
```
#define EXTRUDERS 4
#define SINGLENOZZLE
#define SWITCHING_EXTRUDER
#define SWITCHING_EXTRUDER_SERVO_ANGLES { 0, 90, 0, 90 }
#define DISABLE_INACTIVE_EXTRUDER false
#define NUM_SERVOS 2
#define SERVO_DELAY { 300, 300 }
```

No filament loaded and disable cold extrusion checking (M302 P1)
Try extruding with T0-T3
On my ramps board T0/T1 work but T2/T3 do nothing (E0 & E1 extruders work fine without switching extruder enabled)

thinkyhead, 2018-02-19:
I will look into this more closely in my copious spare time. Since I don't have this kind of setup I will just scan the code and try to imagine what it does.

thinkyhead, 2018-02-19:
A cursory look at `Conditionals_LCD.h` shows that two switching extruders is unsupported at this time.
```cpp
#if ENABLED(SWITCHING_EXTRUDER) || ENABLED(MIXING_EXTRUDER)   // Unified E axis
  #if ENABLED(MIXING_EXTRUDER)
    #define E_STEPPERS  MIXING_STEPPERS
  #else
    #define E_STEPPERS  1                                     // One E stepper
  #endif
  #define E_MANUAL      1
  #define TOOL_E_INDEX  0
#else
  #define E_STEPPERS    EXTRUDERS
  #define E_MANUAL      EXTRUDERS
  #define TOOL_E_INDEX  current_block->active_extruder
#endif
```
The `SWITCHING_EXTRUDER` option forces `E_STEPPER 1`, `E_MANUAL 1`, and `TOOL_E_INDEX 0`.

thinkyhead, 2018-02-19:
@mattgaspar — I've made some tweaks to the code, so maybe it will work now. Test `bugfix-1.1.x` or `bugfix-2.0.x` whenever you have the time.

mattgaspar, 2018-02-20:
@thinkyhead It works! (tested with bugfix-1.1.x)
I'll have to get back to designing/printing a better switching extruders to make it actually work now! (problems with not enough tension)

I looked through the stepper_indirection.h file several times and couldn't find any problems.  I thought I did a global search but might have skipped files labeled LCD!  Why is that logic mixed in with the LCD?

Thanks for getting this fixed!

thinkyhead, 2018-02-20:
I'm thrilled it works! It's bizarre, but although all the code has been in place for ages the conditional stuff was just never done.

> Why is that logic mixed in with the LCD?

`Conditionals_LCD.h` was named at a time when it was more about LCD conditionals. Now that file contains lots of other things that need to be defined at the same time.

closed 2018-02-20 by thinkyhead