MarlinFirmware/Marlin #5369: Help: I want to setup dual X and Y endstops (for MPCNC)

paulpenney, 2016-12-04:
Hello,

I am trying to replicate the Z_DUAL_ENDSTOPS code to work with my X2 and Y2 Steppers for an MPCNC. I have gotten to the point where i've replicated the obvious code for Z_DUAL_ENDSTOPS, and I have X2_min and Y2_min reporting properly in M119 with no conflict. I am not an incredibly talented coder, but I feel like I have duplicated the Z DUAL ENDSTOPS code fairly carefully, but for some reason when I go to do a G28 I still cannot get it working properly. When I try a G28 on say, X and X2, Pressing the Xmin endstop stops both motors, and pressing the X2min endstop does nothing (despite it registering correctly with an M119 report). Z DUAL ENDSTOPS works fine on its own. 

I am running the latest RC7

I am sure there is something below the surface i'm missing, can anyone help? Any ideas? Or can anyone shed light on where the deeper picture is when a Z DUAL ENDSTOP homing is happening?

Thanks

Paul

thinkyhead, 2016-12-04:
Endstops trigger based on the given axis moving. I don't believe the axis is required to be moving in the direction of the specific endstop (i.e., min or max). In any case, I would recommend starting from `RCBugFix` and applying your changes to that branch, because it has slightly newer endstop code, and issues related to dual Z were fixed at some point.

Once you have the code modified as far as you can, if it still doesn't function correctly, then post a link to your branch and someone will give it a review and perhaps have some advice about what needs to be changed.

paulpenney, 2016-12-04:
thinkyhead,

Thanks for your response! I am indeed starting from the RCBugfix version, the latest build, and have tested the Z_DUAL_ENDSTOP functionality I am trying to duplicate to make sure it works (which it does). After duplicating as much code as I can and working out conflicts, the functionality still isn't working for the X and Y axis. 

Sorry I am new to Github and am having trouble understanding how to upload my files. I have my Marlin RCBugFix build that is modified for dual X and Y Endstops locally on my computer, how and where should I upload it?

Thanks for your help, I think this will be a tremendous addition if we can get it working for everyone who is building an MPCNC or any other application where you have two axis that need to be squared before cutting / printing.

Paul

paulpenney, 2016-12-04:
Re: being new to GitHub, I am trying to figure it out now doing research on the net, hopefully I can get my code uploaded soon. 

Paul

paulpenney, 2016-12-04:
Ok, I have uploaded my changed files to: https://github.com/paulpenney/Marlin-Dual-X-Y-Endstops

Is this something you guys can work with? (AKA, did I upload it properly?)

Paul

thinkyhead, 2016-12-05:
You should make a Fork of Marlin and then create a new branch in your fork as a copy of the RCBugFix branch. Then work with that. It's better than creating a whole new repository unconnected to this one.

thinkyhead, 2017-10-31:
Congratulations, this is finally implemented! It will be included in Marlin 1.1.7, coming soon…

closed 2017-10-31 by thinkyhead