MarlinFirmware/Marlin #5370: Question about LCD and COREXY

cfelicio, 2016-12-04:
Hi All,

I'm currently running Marlin 1.0.2, but as RC8 is approaching, I saw some info about optimizations for the LCD, and digging further, I found out some people have issues with the full graphics LCD when using calculation intensive printers (delta, corexy, etc).

I haven't experienced these issues myself, but wondering what is the best strategy:

- Should I ditch the LCD? I currently print using Octoprint via USB
- Should I switch to the 2004 text LCD? That one seems to be less calculation intensive
- Should I reduce the refresh rate of my LCD? Currently sitting at 100 (ms?), how much can I change this before it becomes unusable?

I'm currently printing below 50mm/s, which is probably why I haven't had issues yet :-)

I only use the LCD for basic stuff like homing and adjusting the bed, sometimes to fix jams, etc. All these are performed when I'm not printing, and I do not use the LCD when I'm printing.

Thinking this way, would be cool if there was a way to set the LCD to be very responsive when the printer is not busy, and put it in "hibernation mode" when the printer is actively printing!

Roxy-3D, 2016-12-04:
I don't think a Core-XY is that bad, especially compared to a Delta or a Scara.   But the Graphics LCD's do consume a lot of processing power and for very little benefit.   For $10 (including shipping) you can get a 20x4 LCD on eBay.   Personally...  I think it makes a lot of sense to consider doing that.

thinkyhead, 2016-12-04:
@cfelicio Here's a description of the issue…

The planner buffer empties out more quickly when there are several short moves. If you have a Graphical Display, it may take longer to update the display than for the planner buffer to empty, and as a result the machine stutters because it's stopping very briefly between moves. (It's especially bad on Viki displays, because the display is actually rotated 180° by default, and U8glib is slower in that case.)

We've attacked this problem on a couple of fronts. First, we adopted an option to delay screen updates when the planner buffer has only short moves. That feature essentially eliminates the issue, but it almost completely stops screen updates on a delta (which always does very short moves).

Graphical displays are much more demanding than character-based LCDs, and each screen update requires 2, 4, or 8 loops, updating just a portion of the screen. Up to now, the code did not check to see which part of the screen was being updated, so everything was being drawn each time through that loop.

Last week we implemented "selective rendering" to reduce the amount of redundant drawing and processing on graphical displays. And the results turned out great.

Our first attempt reduced processing time by ~2ms (32,000 cycles). With some more adjustments were able to save ~17ms more (272,000 cycles)! The author of U8glib was also extremely helpful.

We merged these optimizations into the `RCBugFix` branch, so that users of graphic displays should experience much better printing from now on.

cfelicio, 2016-12-04:
Thanks for the information! I guess the best strategy at this point is try the RCBugfix  with my current LCD, while I wait for the 2004LCD to arrive :-)

I'm wondering if there is any object that can be used to "stress test" short moves, and how to visualize / monitor the planner buffer (if it's something an end user can do), to see if a CoreXY actually even suffers from this problem?

Thanks!

Roxy-3D, 2016-12-04:
Very soon I'll have code in the UBL branch to monitor the depth of the Planner (and Serial) queue.   I've got an 8x8 LED matrix attached to my printers so I can turn LED's on and off for more debug info.   You can buy one for $2.50 on eBay (including shipping).   http://www.ebay.com/itm/MAX7219-LED-Dot-Matrix-Display-Module-8x8-Board-for-Arduino-DIY-/201671195758?hash=item2ef48a3c6e:g:klgAAOSwMgdX1jm7

Blue-Marlin, 2016-12-04:
How about activating ADVANCED_OK?

Roxy-3D, 2016-12-04:
Yeah...  I guess I could.  But its going to be fun to watch the LED's.

closed 2016-12-05 by cfelicio