MarlinFirmware/Marlin #5372: On Compile "DIODIO52_DDR was not declared in this scope"

lynspyre, 2016-12-04:
Arduino: 1.6.13 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\MarlinConfig.h:26:0,

                 from sketch\Marlin.h:36,

                 from sketch\temperature.cpp:27:

sketch\temperature.cpp: In member function 'void Temperature::init()':

fastio.h:74: error: 'DIODIO52_DDR' was not declared in this scope

 #define _SET_OUTPUT(IO) do {DIO ##  IO ## _DDR |=  MASK(DIO ## IO ## _PIN); } while (0)

                             ^

sketch\fastio.h:97:25: note: in expansion of macro '_SET_OUTPUT'

 #define SET_OUTPUT(IO)  _SET_OUTPUT(IO)

                         ^

sketch\fastio.h:108:28: note: in expansion of macro 'SET_OUTPUT'

 #define OUT_WRITE(IO, v) { SET_OUTPUT(IO); WRITE(IO, v); }

                            ^

sketch\temperature.cpp:1022:7: note: in expansion of macro 'OUT_WRITE'

       OUT_WRITE(SCK_PIN, LOW);

       ^

In file included from sketch\MarlinConfig.h:26:0,

                 from sketch\Marlin.h:36,

                 from sketch\temperature.cpp:27:

fastio.h:74: error: 'DIODIO52_PIN' was not declared in this scope

 #define _SET_OUTPUT(IO) do {DIO ##  IO ## _DDR |=  MASK(DIO ## IO ## _PIN); } while (0)

                                                         ^

sketch\fastio.h:38:28: note: in definition of macro 'MASK'

   #define MASK(PIN)  (1 << PIN)

                            ^

sketch\fastio.h:97:25: note: in expansion of macro '_SET_OUTPUT'

 #define SET_OUTPUT(IO)  _SET_OUTPUT(IO)

                         ^

sketch\fastio.h:108:28: note: in expansion of macro 'SET_OUTPUT'

 #define OUT_WRITE(IO, v) { SET_OUTPUT(IO); WRITE(IO, v); }

                            ^

sketch\temperature.cpp:1022:7: note: in expansion of macro 'OUT_WRITE'

       OUT_WRITE(SCK_PIN, LOW);

       ^

[... 1049 lines cut ...]

sketch\fastio.h:89:23: note: in expansion of macro '_WRITE'

 #define WRITE(IO, v)  _WRITE(IO, v)

                       ^

sketch\fastio.h:108:44: note: in expansion of macro 'WRITE'

 #define OUT_WRITE(IO, v) { SET_OUTPUT(IO); WRITE(IO, v); }

                                            ^

sketch\temperature.cpp:1024:7: note: in expansion of macro 'OUT_WRITE'

       OUT_WRITE(MISO_PIN, HIGH);

       ^

exit status 1
'DIODIO52_DDR' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

thinkyhead, 2016-12-04:
You're trying to use a thermocouple -3 or -2 and there is no `MISO_PIN`, `MOSI_PIN` or `SCK_PIN`  set for some reason.

What version of Marlin are you trying to compile?

thinkyhead, 2016-12-04:
It looks like you've set your pins to something like:
```
  #define SCK_PIN         SCK
  #define MISO_PIN        MISO
  #define MOSI_PIN        MOSI
```

…which is bad. The pins are set automatically for you in `pins.h`. If you try to set them like that, they end up being set to these prefixes:

```
  #define SCK         DIO52
  #define MISO        DIO50
  #define MOSI        DIO51
  #define SS          DIO53
```

…which are just names used for substitution. They are not intended to be used as pins.

lynspyre, 2016-12-12:
Yes, i have a thermocouple, and i didn't modified none of the parameters on pin configuration, i just grabbed the latest RC and edited with my values.

I solved the problem downloading the latest RCBugFix branch.

closed 2016-12-12 by lynspyre