MarlinFirmware/Marlin #5373: Weird characters when connecting to any host

lynspyre, 2016-12-04:
This is showed up in Arduino IDE -> Serial Monitor

`ánP<nVH5Œèk<H.]áNp|$nHÌÈ+.p|H.8]«ˆZŒŒ%ˆh€ONL}H+MJm>&~‹H'ˆHÈxñJ3ð~èŠ
@“mó7—è-'
&§}+Mo6D«P|ÂóHHvf±J¤JJˆJˆHKHJàˆIñ:&ho^7l

gJÄZäaŽJÿ!Ž
ŽJ¿HJ<~©HŠ$P|þaÿAþ%H%ˆxñé+^yÀN66KÂ^aHMÈv«P<þ!HJâaHJàAHJp|`

This is showed when connecting to Repetier 1.6.2
`03:00:44.970 : No start signal detected - forcing start
03:00:44.992 : N1 M110*34
03:00:44.992 : N2 M115*36
03:00:44.992 : N3 M105*36
03:00:44.992 : N4 M114*35
03:00:44.998 : N5 M111 S6*98
03:00:45.019 : N6 T0*60
03:00:45.022 : N7 M80*28
03:00:45.022 : N8 M105*47
03:00:48.034 : N9 M105*46
03:00:51.095 : N10 M105*22
`

This is showed up in Simplify3D
`Attempting connection at \\.\COM5...
RECEIVED: ánp|n6HÈp|Hn8]ÈZÌK%hÀonL]H+mJM^&ëHÈHÈxñJð^èJ@óm³7÷è-L
RECEIVED: &]+mBOVwD«P|Â³HsH6FñJJJJKHJàIFñ:fH/^7l
RECEIVED: *'JZäAJÿaJ
RECEIVED: ¿HJP|ék^>H$p|þaÿ!þ%H%x±é^YÀNVVâ>aHM6«P|þaHJâaHJà!HJ|@_afF·ÈMoM6
RECEIVED: IM^J«vl*IðXH^¬4Ä&óH
RECEIVED: I&HN~L
RECEIVED: 	À&8nL*xñ
RECEIVED: ZIÈjüJHJ8JÎJ_xñ
RECEIVED: m!H'
RECEIVED: JI¨jNXñ
  Testing plaintext communication protocol...
  Testing binary communication protocol...
  Testing alternate communication protocols...
  Attempting RTS reset and trying again...
  Connection failed.`

Kaibob2, 2016-12-04:
Is your [baud rate in Marlin](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration.h#L125) set to match with your host? For example both 115200

thinkyhead, 2016-12-04:
Some hosts actually start at a high rate and try to connect, then keep going down until they are successful. But most make you set the baud rate to match what is set in your `Configuration.h` `BAUDRATE` setting.

lynspyre, 2016-12-12:
I was using the printer in another PC and i had set baudrate to 250000, but the pc i'm working on with now doesn't accepted 25000, so i had to rebuild the firmware and match it to 115200.

P.S.: I didn't had the previous raw firmware, so i had to configure everything again =(

closed 2016-12-12 by lynspyre