MarlinFirmware/Marlin #5375: ShieldBuddy TC275

Tannoo, 2016-12-04:
This guy looks very exciting. Any thoughts?

http://au.rs-online.com/web/p/processor-microcontroller-development-kits/1245257/

Blue-Marlin, 2016-12-04:
https://github.com/MarlinFirmware/Marlin/issues/3851#issuecomment-239571106
https://github.com/MarlinFirmware/Marlin/issues/3851#issuecomment-263597652
expensive, overpowered, not enough pins for the power it has.

The also overpowered NUCLEO-F429ZI (http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-nucleo/nucleo-f429zi.html) is about 20€ + tax + shipping, but at least has enough pins for now - even if not directly compatible with MEGA/DUE shields.

bobc, 2016-12-06:
I have a ShieldBuddy and will look at implementing a HAL for it, but that would probably be after Re-ARM. The ShieldBuddy runs at 5V, so it might be possible it can run with RAMPS 1.4. ShieldBuddy is probably too expensive except for some niche applications.

There are lots of cheap CPU boards like Nucleo, but there is really no point considering them if there is no printer shield for them.

hitexuk, 2016-12-06:
Hi, we have been approached to port the Marlin printer app onto the Shieldbuddy.  This looks feasible but there is some AVR assembler code to change into Aurix.  

Unlike the Nucleo the ShieldBuddy price is not subsidised by the processor manufacturer.  However as the volume goes up, the price will go down!

We could put an extra row of pins inside the Arduino connectors if that would make it more useful.

hitexuk, 2016-12-06:
Btw. The Shieldbuddy is GBP79 or Euro99!

Blue-Marlin, 2016-12-06:
> There are lots of cheap CPU boards like Nucleo, but there is really no point considering them if there is no printer shield for them.

I know cnc-shield is not as convenient as a RAMPS or other dedicated 3D-printer-board, but for a developer it works and provides a lot of freedom, if you have a bit of space on the desk.
```
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * CNCSHIELD pin assignments
 *
 * http://blog.protoneer.co.nz/arduino-cnc-shield/
 * More meant as a relative universal development platform, then for a serious 3D-Printer.
 * Very suitable for a tiny CNC-machine.
 * The shield has a UNO pinout, so will fit on about anything what claims to be Arduino hardware compatible.
 */

//#ifndef __AVR_ATmega2560__
//  #error "Oops!  Make sure you have 'Arduino Mega' selected from the 'Tools -> Boards' menu."
//#endif
// TODO: pin names for other platforms like mbed.

#define BOARD_NAME         "CNC-Shield"
#define LARGE_FLASH        true

//
// Arduino pin numbering
//

//
// Limit Switches
//
#define X_STOP_PIN          9
#define Y_STOP_PIN         10
#define Z_STOP_PIN         11

//
// Steppers
//
#define X_STEP_PIN          2
#define X_DIR_PIN           4
#define X_ENABLE_PIN        8 // common enable pin

#define Y_STEP_PIN          3
#define Y_DIR_PIN           5
#define Y_ENABLE_PIN        8 // common enable pin

#define Z_STEP_PIN          4
#define Z_DIR_PIN           6
#define Z_ENABLE_PIN        8 // common enable pin

#define E0_STEP_PIN        12
#define E0_DIR_PIN         13
#define E0_ENABLE_PIN       8 // common enable pin

// Everything else has to share the 6 analog in pins on the CNC-Shield labeled as
//                    all Arduino Uno Mega Due  ST-NUCLEO F429ZI F401/411RE
// Reset/Abort        A0  D14/D54 PC0 PF0  A.16           PA3    PA0
// Feed Hold          A1  D15/D55 PC1 PF1  A.24           PC0    PA1
// Cycle Start/Resume A2  D16/D56 PC2 PF2  A.23           PC3    PA4
// Coolant Enable     A3  D17/D57 PC3 PF3  A.22           PF3    PB0
// SDA                A4  D18/D58 PC4 PF4  A.6            PF5    PC1
// SLC                A5  D19/D59 PC5 PF5  A.4            PF10   PC0
// or you may use other uncovered by the shield pins on your processor board.
//

//
// Temperature sensors
//
// Connect thermistors via something like http://reprap.org/wiki/Temperature_Sensor_2_0
// (VCC) O -- 4K7Ohm -- O (Signal) -- 10µF -- O (GND)
//                      I                     I
//                      I----- thermistor ----I
//
// or some thermocoupler based SPI device like MAX6675, MAX31855, MAX31856, ...
// or AD595, ...
//
#define TEMP_0_PIN          5 // analog numbering (D59)
#define TEMP_BED_PIN        4 // analog numbering (D58)

//
// Heaters / Fans
// Connect heaters via something like http://arduino-info.wikispaces.com/Brick-4ChannelPowerFetSwitch
//
#define HEATER_0_PIN       57 // EXTRUDER 1
#define HEATER_BED_PIN     56

#define FAN_PIN            55

#define SERVO0_PIN         54
```
stil unpublished except of here.

closed 2018-02-26 by thinkyhead