MarlinFirmware/Marlin #5376: Help: Z-Probe, mix between Servo Probe and Allen-Key Probe.

zelogik, 2016-12-04:
Hi,

I'm more a electronic/mechanical man, and I need to improve my C++ coding style. (but it's not the problem here) and just before I start to add/modify some lines I need to be sure that the feature don't already exist to make the idea available to the reprap world. :-)

So I have designed a kind of Allenkey probe without an allen-key but with a Bic Clic Pen. The feature to move X/Y/Z for deploy -> probing -> stow the probe work perfectly.
But the downside is that I need an extra space for deploying the probe inside the printing area.
I want to remove the "object" that for the moment is inside the printing area with a moving part (it's where the servo probe come in action).
My dream feature when I type G29, I have a thing like: Move X-Y-Z at the right place -> move the servo back and forth to deploy the probe ->  make the probing action ->  move the X-Y-Z at the right place -> and move the servo back and forth to stow the probe.

I have quickly looked at Marlin_main.cpp and seem like I need to modify these lines no?:

        #elif HAS_Z_SERVO_ENDSTOP && DISABLED(BLTOUCH)
          servo[Z_ENDSTOP_SERVO_NR].move(z_servo_angle[deploy ? 0 : 1]);
        #elif ENABLED(Z_PROBE_ALLEN_KEY)
          deploy ? run_deploy_moves_script() : run_stow_moves_script();
        #endif

Regards.

thinkyhead, 2016-12-09:
We already have a facility to provide enough vertical space to deploy. If you need XY space also, then the allen key code is a good model for that, as you seem to have discovered.

zelogik, 2016-12-12:
Hi,

Yes all facility was there! (and thanks!)

I was able to make my setup working without too much trouble but with very ugly coding style.

    #elif HAS_Z_SERVO_ENDSTOP && DISABLED(BLTOUCH) ||  ENABLED(Z_PROBE_ALLEN_KEY)
      deploy ? run_deploy_moves_script() : run_stow_moves_script();
      servo[Z_ENDSTOP_SERVO_NR].move(z_servo_angle[deploy ? 0 : 1]);
      servo[Z_ENDSTOP_SERVO_NR].move(z_servo_angle[deploy ? 1 : 0]);
    #endif

And disabled some condition, so I can enable both Z_SERVO_ENDSTOP and Z_PROBE_ALLEN_KEY

I will try to make a better coding style when I will make a better and easier Clic-Probe for everyone :)

closed 2017-02-11 by thinkyhead