MarlinFirmware/Marlin #5377: Broken movement with bilinear ABL

MoonshineSG, 2016-12-05:
My aluminium bed is not entirely flat and although I get decent results using AUTO_BED_LEVELING_LINEAR I was interested in trying the AUTO_BED_LEVELING_BILINEAR 

All went well, until after running the G29 I noticed that long moves become split in smaller segments. For example, moving from X0 to X200 should be one single smooth move, but after the bilenear ABL, it becomes more like X0-X50, X50-X180, X180-X200 (not the exact values, just as example). The split seams to be consistent, so the same X0-X200 is always split in what is looks like similar fragments. 

I'm not sure if this is normal, or known problem or just specific to my printer.

Blue-Marlin, 2016-12-05:
That's more or less the expected behaviour.
Every move is sub-devided in moves where it is crossing the net the probe points build. X/Y-speed should be constant, but the z-speed and/or direction changes to follow the bed.

MoonshineSG, 2016-12-05:
For the AUTO_BED_LEVELING_LINEAR, there's a gradual shift of z to follow the bed, if I am not wrong, but for AUTO_BED_LEVELING_BILINEAR is "move x-y, change z, move x-y" kinda approach ?

so the more test point, the more broken the line, (and basically, the slower the print) ?
 
If that the same for manual mesh ?

Blue-Marlin, 2016-12-05:
AUTO_BED_LEVELING_LINEAR assumes a tilt but flat bed. Z-speed for a move is constant because the assumed bed inclination is everywhere the same.
AUTO_BED_LEVELING_BILINEAR assumes a not flat bed. to follow the bed it needs to adjust  z. On some places more on some places less.
The print-speed should not be affected unless z becomes the 'leading' axis. Means, when the height differences become very big, or your z-axis has much more steps/mm then the others.

MoonshineSG, 2016-12-05:
I understand now. If this is the expected behaviour, I have no more comments :D 

I might give it a try. My bed is alright for small and medium size prints. The ones that cover most of the bed surface have some issues, but luckily I don't print those often.  

I can only imagine the calculations and coding that goes in to support badly manufactured or "tortured" beds. :D

MoonshineSG, 2016-12-06:
I tested the `AUTO_BED_LEVELING_BILINEAR` and really can't tell that the print is slower nor can the segmented movement be visually noticed. 

The first layer comes out much nicer, without the "squashed" 1st layer I used to get on the uneven areas...

The only noticeable negative effect is that the printer becomes slightly noisier due to the micro z axis  adjustments, but that is considerably outweighed by the positive effects...

closed 2016-12-06 by thinkyhead