MarlinFirmware/Marlin #5382: GT2560 with Marlin on a Printrbot simple metal with Z Probe

jleurond, 2016-12-05:
Hi there,

Last week my printrboard seems to die during a print after 1 year and a half of good job... I have an other board available at home which is the GT2560 Rev A from geeetech. I downloaded marlin V1.

So I wired all the motors, extruder, endstops and thermistor to the GT2560. To fit to the configuration of motors i modified the configuration.h.

The changes I made to do working the code for the probe is:
- MOTHERBOARD ULTIMAKER (as requested for this board)
- I pluged the probe on the Z min PIN (obliged to put differents resistances to fit to triggered when detecting metal and not triggered when not detecting)
- Activated the probe as a servo:

`#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command

// Servo Endstops
//
// This allows for servo actuated endstops, primary usage is for the Z Axis to eliminate calibration or bed height changes.
// Use M206 command to correct for switch height offset to actual nozzle height. Store that setting with M500.
//
#define SERVO_ENDSTOPS {-1, -1, 0} // Servo index for X, Y, Z. Disable with -1
#define SERVO_ENDSTOP_ANGLES {0,0, 0,0, 70,0} // X,Y,Z Axis Extend and Retract angles`

The problem is about the Z probe: I can home it one time and after the probe is always triggered (tested with M119)... Moreover I diagnose that when I home all axis, then I try to do a G29 with 3 points, the probe goes well to the XY coordinates, but the Z axis is always lifting and not going to detect the bed.

I don't know how to fix this error can you help?

Here is my configuration.h attached:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/631100/Configuration.h.txt)

Thanks a lot

thinkyhead, 2016-12-05:
We recommend switching to a newer version of Marlin, preferably `RCBugFix` which is soon to be released as `1.1.0-RC8`. It makes this kind of configuration easier.

jleurond, 2016-12-05:
Ok I'll try asap

jleurond, 2016-12-05:
Ok, so with the RCBugFix, it seems the firmware doesn't see the probe... It's the good pin (pin 30 worked on the V1.0) and now if I put metal bellow it and do a M119, it's never triggered...

Any suggestion?

thinkyhead, 2016-12-06:
Pin 30 is the `Z_MIN_PIN`, so you should make sure that you have `USE_ZMIN_PLUG` enabled. And also enable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`. If you want to use the probe with a separate pin from your Z endstop, it defaults to the `Z_MAX_PIN` (32), and then you would use `Z_MIN_PROBE_ENDSTOP` instead of `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`.

Roxy-3D, 2016-12-06:
> Last week my Printrboard seems to die during a print after 1 year and a half of good job... I have another board available at home which is the GT2560 Rev A from Geeetech. I downloaded Marlin V1.

How did it die?   I've had my printer 'die' during the middle of the print.   One of the things I do I start slicing the .STL file I'm trying to print.    And in PronterFace I wait for the GCode file to show up in the 'Load File' box.    If I click too quickly before the entire file gets written by Slic3r, I can have the print 'die' in the middle because PronterFace did not load the entire file.

fiveangle, 2016-12-06:
If it truly is a Printrbot Simple Metal 1403, give the attached configs a try with RCBugFix.  Not only do you have to configure the Z_MIN_PIN to the appropriate dual-usage, you have to set the logic states for both of those usages independent of each other even though they are the same (even if you set Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN).

@thinkyhead - there is surely an opportunity to programatically prevent users from falling into this trap which left me scratching myhead for at least an hour of consecutive FW flashes, by adding a logic check to SanityCheck.h, but then at some point you can't hand-hold it for every little thing ;)

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/635004/Archive.zip)

thinkyhead, 2016-12-07:
Good idea!
```cpp
    #elif Z_MIN_ENDSTOP_INVERTING != Z_MIN_PROBE_ENDSTOP_INVERTING
      #error "Z_MIN_ENDSTOP_INVERTING must match Z_MIN_PROBE_ENDSTOP_INVERTING when using Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN."
```

jleurond, 2016-12-07:
@Roxy-3D - Hi Roxy, the printrbot was printing from the sd card and stopped in the middle of the print. Since this, the printrboard refuses to connect to pc anymore. I tried to reset it with the jumper on the "boot pins" and push on the reset button but alway nothing. I tried several differents micro usd cables, different pc, reinstalling the teensy driver etc... Always nothing. So for me I think the motherboard died... BUT if you have a trick I am interested to know if there is an other option.

@thinkyhead - Yes I enabled USE_ZMIN_PLUG and Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN. I only have a sensor for Z so can't use a "dual endstop". With the RCBugFIX it doesn't "recognize" the sensor: never triggered...

Here is the configuration.h of the RCBugFix:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/636224/Configuration.h.txt)

Did I missed something?

I tried another thing with Marlin V1, I have a endstop available and wired it on the Z_MIN_PIN (30), with the good OFFSET and changed #define min_software_endstops true to false. Like that when the endstop touch the bed, the offset will be took in account and I can print. But it doesn't resolve the problem... It's only a temporary solution...

thinkyhead, 2016-12-07:
Just to check one thing in your config… Make sure these settings match:
```cpp
  //#define ENDSTOPPULLUP_ZMIN
  #define ENDSTOPPULLUP_ZMIN_PROBE
```
They should either both be on or both be off.

jleurond, 2016-12-07:
Oh... Ok I didn't know that the both should be on if I use a probe...

Ok I'll try when I'll go back home.

thinkyhead, 2016-12-07:
When you use a probe on the Z min endstop pin it takes on the role of the Z min endstop as well.

jleurond, 2016-12-07:
With these settings enabled:

`#define ENDSTOPPULLUP_ZMIN
 #define ENDSTOPPULLUP_ZMIN_PROB`

Same problem than before: Never triggered..

thinkyhead, 2016-12-07:
I see. And with both _disabled_?

> the probe is always triggered

> Same problem than before: Never triggered..

So with older Marlin, always triggered. With new Marlin, never triggered.

Ideally, you want the probe to at least change state. If it doesn't then it's possibly connected to the wrong pins.

jleurond, 2016-12-08:
Ok so:
- Both disabled, change nothing, never triggered.
- On V1 Marlin, the probe trigger only one time, after it's always triggered. I'll do a video when i'll be back home

thinkyhead, 2016-12-24:
If it's always triggered, it does imply some kind of hardware-related issue. Perhaps the probe input needs a pullup or pulldown resistor to keep it from floating.

mrkaroshi, 2016-12-24:
Can you post a wiring diagram or pinout diagram of how you've got it wired @jleurond ?

By my understanding you've got the COM or main sensor pin connected directly to your limit switch input right? Where are you drawing power from?

The SN04-P induction sensor used in the Printrbot Metal Plus (the one I have) uses a sourcing PNP transistor, so the voltage on the COM line will go from 0V to VCC when it switches, I haven't found any data on whether the input pins on the GT2560 are 12V tolerant but given they're designed to work with simple, normally open, PTM switches I doubt it!

Blue-Marlin, 2016-12-24:
Voltage divider:
https://www.youtube.com/watch?v=EcGFLwj0pnA

mrkaroshi, 2016-12-24:
Wouldn't a much simpler way to handle this just be using a relay? Without knowing the tolerances of the GT2560 aren't you just running 12V from the probe straight into the IO of the controller via the pull ups on the limit switch inputs? I'm going to try using a 5V relay as an intermediary to prevent the probe from frying the switch inputs - this method also means you don't need to use any unusual settings in the Configuration.h, and if anything goes wrong you'll just blow the relay not the controller. I'll let you know how it goes.

mrkaroshi, 2016-12-24:
@Blue-Marlin that's a really wasteful way of doing this isn't it? Not to mention the noise from the induction sensor entering the controller inputs?

Blue-Marlin, 2016-12-24:
@mrkaroshi 
No it is not. I cant suggest a cheaper and better solution than just two resistors or one potentiometer (adjustable for 5 V and 3.3V). It just works on 20+ sensors here.

mrkaroshi, 2016-12-24:
@Blue-Marlin personally I'm used to interfacing anything 12V or above with a microcontroller via something like a relay or MOSFET, that's just how I've done it in the past with CNC projects etc. I get that this is more about grassroots/simplistic/approachable solutions, and solving things via software, but my priority is not cheapness, but robustness.

Blue-Marlin, 2016-12-24:
> I'm used to interfacing anything 12V or above with a microcontroller via something like a relay or MOSFET

Upwards! Downwards, i'd suggest a optocoupler if a voltage divider seem not to be save enough.

mrkaroshi, 2016-12-24:
@Blue-Marlin good shout! I wish I had one of those to hand actually.

closed 2017-04-02 by thinkyhead