MarlinFirmware/Marlin #5384: Z_DUAL problem

Dreamer68, 2016-12-05:
Hello,

On my 3d printer with a ramps 1.4

In the Marlin program, I removed the #define Z_DUAL_STEPPER_DRIVERS to set e1 in z2. In manual control everything works.

But when I ask him to calibrate the position zero, there is that e1 (z2) which moves without z1.

And even when I ask to calibrate only the axis X, there is that e1 (z2) that moves.

While before activating the Z_dual, it works correctly.

Can you help me ?

thinkyhead, 2016-12-05:
Are you using the `RCBugFix` branch?

Dreamer68, 2016-12-05:
No, it changes what?

thinkyhead, 2016-12-05:
It changes almost everything.

Dreamer68, 2016-12-05:
Should I switch to the RCBugFix branch ?

thinkyhead, 2016-12-05:
I would recommend that you do. It will be easier to troubleshoot.

Dreamer68, 2016-12-05:
Thanks, I'll try

Dreamer68, 2016-12-05:
What are the differences between the two versions ?

Sanitar777, 2016-12-05:
> thinkyhead commented an hour ago
It changes almost everything.

Literally - everything! Believe me, if you don't believe the developer ;-)

Dreamer68, 2016-12-05:
If I believe the developer, but I have trouble understanding why there are 2 versions so different :-)

Sanitar777, 2016-12-05:
It's very simple... it's easier to write new code than fix bugs in the old code. Since the team decided to radically revise the old firmware. They made this decision. I thank them for that! :-)

Dreamer68, 2016-12-05:
Okay, thank you for the explanation :-)
I'll see if it fixes my problem as soon as possible.

Blue-Marlin, 2016-12-05:
Marlin V1 is almost 2 years old. We have not been lazy.

Dreamer68, 2016-12-07:
So I tried RCBugFix and I still have the same problems.

thinkyhead, 2016-12-07:
I see. My first thought is that perhaps there is too much torque for the steppers to handle.

As a preliminary step, lower your Z homing speed, Z max feedrate, Z max acceleration, and Z jerk a bit to see if it makes any difference. You can use `M201 Znn`, `M203 Znn`, and `M205 Znn` to play around with these settings without re-flashing the firmware.

Dreamer68, 2016-12-07:
There, I just tried to put both Z motors on a single conductor (Z). It works, but after a few seconds, one engine stops and the other accelerates.

You have an idea?

thinkyhead, 2016-12-07:
Bad wiring?

Dreamer68, 2016-12-07:
The engines turn in the right direction. It's just that one engine stops before the other.
I do not think it comes from wiring

Dreamer68, 2016-12-07:
And when I want to calibrate X, there is the Z axis that moves 2 seconds before x is calibrated.

thinkyhead, 2016-12-07:
> one [motor] stops before the other

If you see this when the motors are both connected to the same driver, it can't possible be software.

Dreamer68, 2016-12-07:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/637545/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/637546/Configuration_adv.txt)

thinkyhead, 2016-12-07:
> when I want to calibrate X, there is the Z axis that moves 2 seconds before x is calibrated

I don't see anything in your configurations to indicate why this would occur.

I see you have hardly made any changes to the default configurations. For example, none of these have been changed:
```cpp
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 500 }
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 25 }
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }
```
Are these all just coincidentally right for your printer?

Dreamer68, 2016-12-07:
In fact, before bothering to adjust the values, I wish I could make it move correctly.
That's why I have configured the limit switches and the direction of the different motors.

thinkyhead, 2016-12-09:
You can't make it move correctly without having values tuned to your hardware.

Dreamer68, 2016-12-10:
Thank you for your help, for now everything works except the limit switches (X, Y) that are no longer recognised.

Dreamer68, 2016-12-10:
Thank you for your help, it works, it only remains to correctly adjust dimensions :)
To recover the limit switches, I went back to the RC version, because on the RCBugFix the limit switches were not detected.

thinkyhead, 2017-02-11:
Resolved?

aaronthomaswebster, 2017-02-26:
Z_DUAL_STEPPER_DRIVERS Is not working for me either, one motor just hums... It seems like I need to adjust the steps per unit for the second Z stepper. Is there a way to set the steps for the second Z motor?

james94jeans2, 2017-02-26:
@aaronthomaswebster this sounds more like a current issue with the second driver. Did you adjust the current on the second driver?

aaronthomaswebster, 2017-02-26:
I actually pulled the drivers and switched their position and still had an issue with the motor running off of E1.

james94jeans2, 2017-02-26:
okay, maybe you could try enabling the toshiba driver option in configuration.h to test if there is a problem with the frequency of steps

closed 2017-10-31 by thinkyhead