MarlinFirmware/Marlin #5391: Z Axis Don't Stop...

cgaudin, 2016-12-05:
Hello,

I have a Chinese Prusa i3 (with MKS Base v1.4) and i wanted to try Marlin Firmware devel-ubl. I compiled the firmware with the arduino ide 1.6.12. 
I have an issue with the Z-Axis. 
As soon as i move the Z-Axis, it starts to move but never stops (G0 or G28 commands).
However, X & Y axis seems to work correctly.
I use pronterface to send commands to the printer.

I don't know if i made a mistake in my configuration file or if this is a bug.

Here is my configuration file.
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/632540/Configuration.h.txt)

thinkyhead, 2016-12-06:
Are you using the `RCBugFix` branch?
What does the output of `M119` look like when you press and release your Z endstop?

Bob-the-Kuhn, 2016-12-10:
The G28 command only stops when it hits the appropriate endstop.  That's why Thinkyhead asked  you to test your Z endstop.

If you have PINS_DEBUGGING enabled (in Configuration_adv.h) then you can issue a M43 E1 command to monitor all the configured endstops.   When an endstop changes state it's displayed on the console and the system LED changes state.  This is useful for calibrating endstops.

The G0 never stopping is really strange.  The only thing I can come up with is if the steps/mm in your configuration.h file is way too high then the hardware will move a lot farther than the software thinks it will.  If the steps/mm is too high by 100 then a G0 Z-10 command will send it in the negative Z direction by 1000 mm which will bang the head into the mechanical stop which makes it seam like it never stops.

cgaudin, 2017-03-17:
Hello, 
I've forgot to change my steps/mm in my configuration file.
Thank you for your time.

closed 2017-03-17 by cgaudin