MarlinFirmware/Marlin #5392: Extruder wont heat enough ."Heating Failed"

EugeneSmith007, 2016-12-05:
hello folks.
I am fighting several days on isssue with error message "heating failed" on my TEVO Black Widow printer after temp reaches 190C or around. From 21C to 120C it goes pretty fast then slowing down. From 170C to 190C temp increasing significantly  slow. And at this moment close to 190C shows up error "Heating failed" . I have a fan 24V which blows to the heater and when i close air flow from fan , heating goes pretty fast again up to 260 without any errors. (more then 260C i didnt try) . is this fan to powerful for this 40Watt heater? 
and as well I read a lot info here and tried to fix this issues changing some parameters in Config adv.
* WATCH_TEMP_PERIOD  set to 60
* WATCH_TEMP_INCREASE set to 2
 i tried different settings for this and nothing helps.

Any ideas how to fix?

Sungod3000, 2016-12-06:
sounds like an overpowered fan. i have a torus shaped one but cant go higher than 50% before cooling down the nozzle.

I wouldn't adjust the runaway settings rather fan speed and duct.

EugeneSmith007, 2016-12-06:
Could you tell me,  can i connect my fan to E1 output (because its available)  in order regulate fan speed . if yes , how i can do it?
now its connected directly to power supply

EugeneSmith007, 2016-12-06:
I mean what i need to change in code to work with E1 output ?

Sungod3000, 2016-12-06:
In the standard configuration on Ramps the connectors between heatbed and heater are used for the model cooling fan. Check the wiring schematics for your board, its a simple hardware issue.

thinkyhead, 2016-12-07:
If there's no available fan connector and you want to connect to the free E1 heater output, just change `FAN1_PIN` to whatever your board's `HEATER_1_PIN` is currently. The firmware will then allow you to use `M106 P1 Snnn` to regulate the speed.

closed 2017-10-31 by thinkyhead