MarlinFirmware/Marlin #5396: Hotend Offset reported wrong on M501

Kaibob2, 2016-12-06:
When i send M501 Marlin responds:
```
V27 stored settings retrieved (410 bytes)
Steps per unit:
M92 X80.30 Y80.30 Z1600.70 E97.70
Maximum feedrates (mm/s):
M203 X300.00 Y300.00 Z9.00 E150.00
Maximum Acceleration (mm/s2):
M201 X1000 Y1000 Z100 E10000
Accelerations: P=printing, R=retract and T=travel
M204 P2000.00 R3000.00 T2000.00
Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E5.00
Home offset (mm)
M206 X-31.00 Y0.00 Z0.00
Hotend offsets (mm)
M218 T1 X1420 Y1428
Material heatup parameters:
M145 S0 H210 B60 F0
PID settings:
M301 P21.57 I2.86 D40.72
Filament settings: Disabled
M200 D1.75
M200 T1 D1.75
M200 D0
```
Everything fine except for 
```
Hotend offsets (mm)
M218 T1 X1420 Y1428
```
When i send `M218 T1 X29 Y0` to match my printer Marlin responds `Hotend offsets: 0.00,0.00 29.00,0.00`
Then i send M500 and get the "settings stored" confirmation.
The Offset is applied correctly and the printer behaves like expected.
When i send an M501 again it reports:
```
......
M218 T1 X1420 Y1428
.....
```
So it's more a cosmetical bug as the function itself is working as intended.

thinkyhead, 2016-12-06:
Does `M503` also say that?

thinkyhead, 2016-12-06:
Are you using `1.1.0-RC8` (`RCBugFix`)? There was an issue with that a couple days ago, but it has been patched already.

closed 2016-12-06 by thinkyhead

Kaibob2, 2016-12-06:
@thinkyhead Sorry for coming up with this. It was on RCBugfix from around 10 days ago. Just switched to RC8 and everythings fine here.