MarlinFirmware/Marlin #5399: [Question] Override next free extruder when using the dual z stepper option

boelle, 2016-12-06:
I got one of these http://reprap.org/mediawiki/images/4/4f/RAMPS_SE_X2.jpg

for my 3rd extruder and 2nd Z stepper driver

if i enable #define Z_DUAL_STEPPER_DRIVERS it would normally take the next free extruder

but what if there are none or i want to assign a stepper driver to use specific? how would i override the automation?

i ask simply because i dont know and i cant figure it

thinkyhead, 2016-12-06:
Just define the pins you want to use in your config.
```cpp
    #define Z2_STEP_PIN   36
    #define Z2_DIR_PIN    34
    #define Z2_ENABLE_PIN 30
```

boelle, 2016-12-06:
and enable the define? or just define the pins?

thinkyhead, 2016-12-06:
Yes, enable the &#$% define.

closed 2016-12-06 by thinkyhead

boelle, 2016-12-06:
Sorry for not knowing :-/