MarlinFirmware/Marlin #5401: RC8 won't compile when DISTINCT_E_FACTORS enabled

Kaibob2, 2016-12-06:
Just upgraded to RC8 and everythings fine as expected because i always kept up to date with RCBugfix.
One thing doesn't work.
If i uncomment `#define DISTINCT_E_FACTORS` and `#define EXTRUDERS ` is set to "2" then the Arduino IDE 1.6.9 compiles with the following error:

```
Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\Kai\AppData\Local\Temp\build8cd5902b42a46e270b834640dfde094e.tmp\sketch\stepper.h:46:0,

                 from C:\Users\Kai\AppData\Local\Temp\build8cd5902b42a46e270b834640dfde094e.tmp\sketch\MarlinSerial.cpp:33:

C:\Users\Kai\AppData\Local\Temp\build8cd5902b42a46e270b834640dfde094e.tmp\sketch\planner.h: In static member function 'static void Planner::set_e_position_mm(const float&)':

planner.h:353: error: invalid conversion from 'int' to 'AxisEnum' [-fpermissive]

           + active_extruder

           ^

planner.h:348: error: initializing argument 1 of 'static void Planner::set_position_mm(AxisEnum, const float&)' [-fpermissive]

     static void set_position_mm(const AxisEnum axis, const float &v);

                 ^

Bibliothek U8glib in Version 1.19.1 im Ordner: C:\Users\Kai\Documents\Arduino\libraries\U8glib  wird verwendet
exit status 1
invalid conversion from 'int' to 'AxisEnum' [-fpermissive]
```

If i set `#define EXTRUDERS ` to "1" or comment `#define DISTINCT_E_FACTORS` it compiles fine.

esenapaj, 2016-12-07:
And, additional error exists.

<details>
<summary>error message</summary>

```
\AppData\Local\Temp\arduino_build_18740\sketch\ultralcd.cpp: In function 'void _planner_refresh_e_positioning(uint8_t)':

ultralcd.cpp:1877: error: 'i' was not declared in this scope

         planner.steps_to_mm[i] = 1.0 / planner.axis_steps_per_mm[i];

                             ^
```
</details>

thinkyhead, 2016-12-07:
There! That should be the last of the bugs.

closed 2016-12-07 by thinkyhead

Kaibob2, 2016-12-08:
@thinkyhead 
I'm sorry, but latest RCBugFix still doesn't compile when `#define DISTINCT_E_FACTORS` is enabled.
I downloaded a fresh copy and only changed:
#define MOTHERBOARD BOARD_RAMPS_14_EEB
#define EXTRUDERS 2
#define DISTINCT_E_FACTORS
#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 1
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 1

With this configuration it won't compile.

```
Arduino: 1.6.9 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\Kai\AppData\Local\Temp\buildbf5a5b98588b4694630d567cdb426e9e.tmp\sketch\planner.h: In static member function 'static void Planner::set_e_position_mm(const float&)':

planner.h:353: error: invalid conversion from 'int' to 'AxisEnum' [-fpermissive]

           + active_extruder

           ^

planner.h:348: error: initializing argument 1 of 'static void Planner::set_position_mm(AxisEnum, const float&)' [-fpermissive]

     static void set_position_mm(const AxisEnum axis, const float &v);

                 ^

exit status 1
invalid conversion from 'int' to 'AxisEnum' [-fpermissive]
```

esenapaj, 2016-12-08:
It's bit strange.
On my side, now compilation error doesn't occurs.
Commit https://github.com/MarlinFirmware/Marlin/commit/01e5d46ea7d4f955864f68fff5025beac50be3d9 solved errors.

Kaibob2, 2016-12-08:
@esenapaj Which Arduino Version are you using. I just installed 1.6.13 (was 1.6.9 before).
With 1.6.13 it throws some warnings but compiles to the end and uploads fine.

These are the warnings:
```
In file included from C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\temperature.h:30:0,

                 from C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\temperature.cpp:29:

C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\planner.h: In static member function 'static void Planner::set_e_position_mm(const float&)':

C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\planner.h:353:11: warning: invalid conversion from 'int' to 'AxisEnum' [-fpermissive]

           + active_extruder

           ^

C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\planner.h:348:17: note: initializing argument 1 of 'static void Planner::set_position_mm(AxisEnum, const float&)'

     static void set_position_mm(const AxisEnum axis, const float &v);

                 ^

"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Users\Kai\Documents\Arduino\libraries\U8glib\src" "C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\twibus.cpp" -o "C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\twibus.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Users\Kai\Documents\Arduino\libraries\U8glib\src" "C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\ultralcd.cpp" -o "C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\ultralcd.cpp.o"
In file included from C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\temperature.h:30:0,

                 from C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\ultralcd.cpp:28:

C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\planner.h: In static member function 'static void Planner::set_e_position_mm(const float&)':

C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\planner.h:353:11: warning: invalid conversion from 'int' to 'AxisEnum' [-fpermissive]

           + active_extruder

           ^

C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\planner.h:348:17: note: initializing argument 1 of 'static void Planner::set_position_mm(AxisEnum, const float&)'

     static void set_position_mm(const AxisEnum axis, const float &v);

                 ^

"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10613 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "-IC:\Users\Kai\Documents\Arduino\libraries\U8glib\src" "C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\utility.cpp" -o "C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\utility.cpp.o"
In file included from C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\temperature.h:30:0,

                 from C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\utility.cpp:25:

C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\planner.h: In static member function 'static void Planner::set_e_position_mm(const float&)':

C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\planner.h:353:11: warning: invalid conversion from 'int' to 'AxisEnum' [-fpermissive]

           + active_extruder

           ^

C:\Users\Kai\AppData\Local\Temp\arduino_build_272598\sketch\planner.h:348:17: note: initializing argument 1 of 'static void Planner::set_position_mm(AxisEnum, const float&)'

     static void set_position_mm(const AxisEnum axis, const float &v);

                 ^
```

esenapaj, 2016-12-08:
I'm using 1.6.14 Hourly Build 2016/11/25 15:33.

And I also confirmed the warnings, and I submitted a patch.
https://github.com/MarlinFirmware/Marlin/pull/5431(Suppress warnings, etc)

Kaibob2, 2016-12-08:
Applied #5431 and the warnings are gone and it compiles and works. Great job. Thank you!

Kaibob2, 2016-12-08:
Maybe the :+ collision should be fixed one day.
[image]

esenapaj, 2016-12-08:
How about these solution?

```diff
-#define MSG_XSTEPS                          "X Steps/mm"
-#define MSG_YSTEPS                          "Y Steps/mm"
-#define MSG_ZSTEPS                          "Z Steps/mm"
-#define MSG_ESTEPS                          "E Steps/mm"
+#define MSG_XSTEPS                          "XSteps/mm"
+#define MSG_YSTEPS                          "YSteps/mm"
+#define MSG_ZSTEPS                          "ZSteps/mm"
+#define MSG_ESTEPS                          "ESteps/mm"
```

or

```diff
-#define MSG_XSTEPS                          "X Steps/mm"
-#define MSG_YSTEPS                          "Y Steps/mm"
-#define MSG_ZSTEPS                          "Z Steps/mm"
-#define MSG_ESTEPS                          "E Steps/mm"
+#define MSG_XSTEPS                          "Xsteps/mm"
+#define MSG_YSTEPS                          "Ysteps/mm"
+#define MSG_ZSTEPS                          "Zsteps/mm"
+#define MSG_ESTEPS                          "Esteps/mm"
```

Kaibob2, 2016-12-08:
I think the second solution looks better. What about removing the space before the colon?

Kaibob2, 2016-12-08:
I like the third version most.
A much bigger problem is, i just figured out, that when i start a print with the second extruder it produces shit. Sorry to say that, but it's hard to explain. The model is 
[image]
The GCODE is this
[warp2.zip](https://github.com/MarlinFirmware/Marlin/files/639159/warp2.zip)

With `#define DISTINCT_E_FACTORS` disabled it prints perfect what it should, with enabled `#define DISTINCT_E_FACTORS` the Y Axis makes a unhealthy sound as if the speed would be much to high but doesn't move. The Display says Y would have moved to the correct position. 
Then X moves somewhere and starts with the infill of the first layer.
It's completely messed up.

esenapaj, 2016-12-08:
Space doesn't exists between `E1` (or `E2`) and `:`.

If you'd like to keep the string likes `X Steps/mm`, another solutions are needed.

Example:
[image]

Code:
https://github.com/esenapaj/Marlin/commit/99e8f8c62b878e1539dae1d022202b384673aa49
Demo branch:
https://github.com/esenapaj/Marlin/tree/Remove-redundant-E (If you test this, please test with `language_de.h`. otherwise maybe error is thrown)

By the way, `Endstopp-...` line looks like broken?

Kaibob2, 2016-12-08:
This looks perfect!

Kaibob2, 2016-12-08:
Should i open a new issue for 

> A much bigger problem is, i just figured out, that when i start a print with the second extruder it produces shit...

esenapaj, 2016-12-08:
>Should i open a new issue for

I think that that may be better.

Kaibob2, 2016-12-08:
Will you raise a Probe for 99e8f8c ?

esenapaj, 2016-12-08:
That is just a quick demonstration code, and if you (or I) submit it as a PR, code needs further change.
Would you like to submit a PR, or you want me to finish it and submit it as a PR?

Kaibob2, 2016-12-08:
I would prefer if you finish it as you are by far the better coder :)

esenapaj, 2016-12-08:
I submitted patches.
Remove redundant "E", etc plan A by esenapaj · Pull Request #5435 · MarlinFirmware/Marlin
Remove redundant "E", etc plan B by esenapaj · Pull Request #5436 · MarlinFirmware/Marlin

> you are by far the better coder

I think that you are far more potent than me.
Because I took more than half a year to memorize the concept of git and the minimum usage of Github desktop, but you learned it overnight.
The important thing is to try what you can not do currently.

Kaibob2, 2016-12-09:
>  ...but you learned it overnight.

I wouldn't say "learned" i just followed your guideline. but it helped a lot getting started. The main problem is that my coding knowledge is based on basic from ancient C64 times. Never learned C. This makes it exremely hard to use the correct syntax. I mainly copy paste lines from here and there, but this is very unsatisfying. The next thing is time, but i think this is a very common problem :)
Anyway, thanks for your support!

thinkyhead, 2016-12-09:
You can learn the essentials of C in a day or two. Kernighan & Ritchie's original book on C is really good too.