MarlinFirmware/Marlin #5402: Z electromagnetic stop OMRON EE-SPX403

Dontcallmelater, 2016-12-06:
Dear friends hope you can help.

I am not pro about 3d. Have MKS base 1.3 X\Y - simple switch stops.
Z - electromagnetic 12-24V as specified about OMRON EE-SPX403. BUT! 
[OMRON](https://ru.aliexpress.com/item/EE-SPX403/1638640113.html?spm=2114.30010708.3.17.Xq55CX&ws_ab_test=searchweb0_0,searchweb201602_3_10065_10068_10084_10083_10080_10082_10081_10060_10061_10062_10056_10055_10037_10054_10059_10032_10099_10078_10079_10077_10093_10073_10097_10100_10096_10070_10052_10050_424_10051,searchweb201603_1&btsid=986c875b-6031-4c6a-bf55-9ad7cc677505&aff_platform=link-c-tool&sk=uR7uBY3Rz%3A&cpt=1481027100255&af=DPlnKr2ATnM&afref=&cn&cv=90001&dp=E2rrZCpoEf&aff_trace_key=6244e2cc8fbe4893b4da49e1a09d71a3-1481027100255-04514-uR7uBY3Rz)
It was working from very begining with problems on this board sometimes it is ON some times OFF.

1. I measure the output ratio it is only 5V. I DO NOT KNOW IS THIS VOULTAGE OUTPUT WAS ALWAYS 5V or changed during the exploatation?
So this OMRON cant work normally from very begining? Or 5V is also OK for it?

2. The led lits on OMRON stop when I connect it to 12V directly (Signal wire to Z stop port) - but in this mode 3d printer does not go home Z axice. Can I change the code and convert 5V into 12V output from Z stop port?

3. Or the OMRON just dead? How to check it experimentally?

4. And in case is it possible to put casual SWITCH stop in Z? Should I change arduino code?
Hard stop will be much worther? Or almost the same as electromagnet?

Thank you!

thinkyhead, 2016-12-07:
> Can I change the code and convert 5V into 12V output from Z stop port?

No. That's not possible. You would need to do something like use a relay or MOSFET.

If the device is actually 5V then perhaps it just needs a pull-up resistor, but it's hard to say. If you can find the datasheet, I would recommend studying that to see how this device is meant to be used.

You may ask for help on the RepRap forums, and some savvy electronics guru may offer help.

Dontcallmelater, 2016-12-07:
> No. That's not possible. You would need to do something like use a relay or MOSFET and then use the 5V to switch it on/off.

 Thank you my Friend about the answer

The aliexpress seller give it to me like a normal element for a Z stop.
What can you advice to use for a Z stop?

[DATASHEET](http://www.ia.omron.com/data_pdf/cat/ee-spx303n_403n_ds_e_3_2_csm2162.pdf)

NPN voltage output:
Load power supply voltage: 12 to 24 VDC
Load current: 80 mA max.
OFF current: 0.5 mA max.
80 mA load current with a residual voltage of 2.0 V max.
10 mA load current with a residual voltage of 1.0 V max.

[image]

thinkyhead, 2016-12-07:
> What can you advice to use for a Z stop?

Anything that acts like a simple switch, passing the 5V input to GND.

macst34, 2016-12-07:
Hi @Dontcallmelater,

The Omron EE-SPX403  is a slot sensing opto sensor (not magnetic).
It should be able to handle 12-24V.
It has a little LED indicator which is lit when the slot is open. If you wired it up correctly, it should be lit when there is nothing in between the sensing prongs. (There are some arrows molded on the prongs themselves that show the approximate sensing plane) If you trigger the sensor by crossing the plane...(for example, with your x-axis smooth rod) the led will go out. This visual indicator will help you troubleshoot your endstop. 

From the datasheet you referenced, I added some notes:

[image]

**!Important**! confirm your wiring correct as mine came attached with polarity reversed from convention. Mine was wired black->positive; red ->negative; yellow ->signal.  It is also in this order on the dupont connector at the other end. I could of resoldered and reshrunk to get the colors right, but I didn't think it worth the hassle since it interfaces with my control board (ramps) correctly.

Here is a picture of mine installed.  Originally, the design called for the opto-endstop to sense the x axis smooth rod, but this made manual z leveling a chore since I had to loosen two bolts to move it and I could not control how much or how little it moved. Clearly, this would not be a good solution. To resolve this design issue, I ended up mounting an adjustable M3 bolt with an acorn nut on the end onto my x-axis assembly to trip the optosensor. That way I could adjust the whole Z plane up and down using a turn of the allen wrench to move the bolt up or down..

[image]

Since I am printing at the moment, the X axis assembly has left the Min Z position. If you look closely, the little red led is lit on the omron optosensor left of the "made in japan" mark. The bolt with  acorn nut is hovering above the EE-SPX403 opto-endstop. 

Hope this helps you.

Happy hacking,

Michael

Dontcallmelater, 2016-12-12:
Thank you @macst34 

I will try this. I also will notice the polarity.
But in general it works normaly from 5v? 
If input reversed voultage can it damage the OMRON?

It should be when this Z stop is USED? Correct?
> The only thing else that you have to check is that configuration.h has
> 
> #define Z_MIN_ENDSTOP_INVERTING false

I have simalar construction, fixtures and alu frames. The OMRON only in stock stand 90 degree to yours possition. Need to try to use your HACK. GOOD IDEA!

macst34, 2016-12-12:
hi @Dontcallmelater ,

I'm not sure what reversing polarity will do to the omrom optical sensor.

The way to check if it is working is to plug it in correctly, observing polarity, and see if the red led is on.
Then block the prongs with your finger and the red led should go off.

Yes, we do seem to have the same printer...Which country are you in and where did you get your kit?
I'm in Taiwan and got mine off ruten.com. They do not seem to have this kit anymore, but I prefer it compared to some other kits. It should be more stable than acrylic frame kits if built correctly.

The wire leads from my omron x403 are from left to right black(positive), yellow(signal), red(negative):
[image] 

Here is the connection to ramps:

[image]

Michael

Dontcallmelater, 2016-12-17:
Dear @macst34 
I get my on aliexpress. I am in Ukraine.
Dimension is 300x300x200 mm
And may be you know the pins from left to right **S\-\+** ? Correct I cant find it on internet. I measure it by mutimetr but it is not defined only 2 pins like +\-. May be the slot was soldering wrong?
[image]

macst34, 2016-12-19:
hi @Dontcallmelater,

Yes, if you put a finger in between the prongs and the led is still lit, it is probably broken/defective and unusable.

Sure, you can use a simple microswitch to replace the optical sensor.
[image]

Just cut off the optoswitch and use the same wiring....make sure to wire it the same as the X and Y stops and you'll be fine. 

Michael

thinkyhead, 2017-02-11:
What's the status of this issue?

macst34, 2017-02-12:
Hi thinkyhead....

I imagine he solved it. 

Michael

Dontcallmelater, 2017-02-13:
Dear @thinkyhead and @macst34 I have been order same magnetic stop due to a more simple and precize measurement. The only ine problem with it was that in original specs th input voultage starting from 12V Thats why I think the problems occurs. Due to long time shipment I will wright to you as i install it.

Thank you for your support!

thinkyhead, 2017-04-02:
Closing this, but please add comments if you like.

closed 2017-04-02 by thinkyhead