MarlinFirmware/Marlin #5405: More ideas how to prioritize filling the planner buffer

AnHardt, 2016-12-06:
In `loop()` 
```cpp
void loop() {
  if (commands_in_queue < BUFSIZE) get_available_commands();
...
  if (commands_in_queue) {
...
        process_next_command();
...
}
  ...
  endstops.report_state();
  idle();
}
```
we currently only do one `get_available_commands()` and one `process_next_command()`.

What would happen if we replace the `if`s with `while`s ?

I have no real clue about the consequences.
This idea is in no way tested!

landodragon141, 2016-12-06:
If I understand correctly this would cause less loops to be instanced but they will run longer. Right? Will these respond to interrupts?

AnHardt, 2016-12-06:
```
while (commands_in_queue < BUFSIZE) get_available_commands();
```
is nonsense because it potentially could last for ever when nothing is available in the serial buffer. In the deeps of `get_available_commands()`. we already have a `while (commands_in_queue < BUFSIZE && MYSERIAL.available() > 0)` and/or a `while (commands_in_queue < BUFSIZE && !card_eof && !stop_buffering)`. That's all we can do. 

@landodragon141
The wanted effect would be a slight shift in the priority.
From - (get one line, process one line, idle) to (get as many lines you can get, process as many lines as you can, idle). As far i have investigated now, we already have: (get as many lines you can get,  process one line, idle) 
Interrupts are not a problem here.

justmyopinion, 2016-12-06:
We may have discussed this before Anhardt... :-)

Anyway it is one year ago I implemented this in my optimized 1.02 in my printer and no issues has been found doing this. I have also added this change to latest RC8 and sofar not found any problem doing this and I still have a smooth printer running.
I have only tested this change to fill the block:
...
//  if (commands_in_queue) {
  **while (commands_in_queue) {**
...
        process_next_command();
}

Trying to copy this is however at own risk :-)

AnHardt, 2016-12-07:
```
while (commands_in_queue)
```
has about the same problem. It needs a second condition - about free places in the planner buffer.
So why does `if (commands_in_queue)` work? Does it?
In `process_next_command()` we have two main cases. 
Ether it's a 'motion' command what has to be planed into the planner buffer. If that is full the planner calls idle().
```
  // If the buffer is full: good! That means we are well ahead of the robot.
  // Rest here until there is room in the buffer.
  while (block_buffer_tail == next_buffer_head) idle();
```
Or we have a 'immediate' command what is tried to be executed directly. 
Most of them return directly after their job is done - for example 'switching the light on'. Here a 'while' wold help. 
Some other command have to wait until other things have been done. Most common is waiting until the planner buffer is empty. Thereafter the planner buffer is mostly dry. After waiting for temperatures the planner buffer is dry. After waiting for user the planer buffer is dry. During all this waiting 'idle()' is called. 'while' could be useful to refill the planner buffer as fast as possible. 
 
When having only fast short moves with a not completely full planner buffer (the case where it matters) the 'while' loop could 'plan' `BUFSIZE` moves at once without calling `idel()`. But the now called after every command `endstops.report_state();` would also only executed after `BUFSIZE` commands - in worst case.
To correct that we could shift `endstops.report_state();` to the end of `process_next_command();`. 
```
while (commands_in_queue &&  (block_buffer_tail != next_buffer_head) ) {
...
  process_next_command();
...
}
idle();
```
could have the wanted effect, if there is more then one command in the line buffer. 
That is not the case for 'ping-pong' mode.

thinkyhead, 2016-12-07:
Some interesting ideas. We know that `idle()` will be called when the planner can't take the next move, and we know that `idle()` will be called by anything that pauses (e.g., `M1`). And we also know that general command processing itself is actually very speedy — unlikely to take so long that more than one move will get processed by the stepper ISR, except when doing both high speed and short moves. In which case, most likely `idle()` is being called continually out of the loop in `prepare_kinematic_move_to`.

The question is, how often is `loop()` otherwise called? And what is real difference? Here's what `loop()` does, in a nutshell:
- Fill up the command buffer as much as possible
- Check for SD autostart (only actually triggered on SD insert)
- Process a single command
- Report endstop state (only if endstop hit bits are set)
- idle()

The while change only affects the 3rd step:
- Process all commands in the queue

And so the net effect is that while commands are in the queue, the other items on the list will be skipped, and there will be no return back to the Arduino thread manager that calls `loop()`. So those are the cycles that will be saved.

In the case of a deltabot, the single-command-processing block will tend to always add multiple short moves to the planner, so it won't be getting starved, and as mentioned, `idle()` will be called by `planner.buffer_line` while it waits for the next available slot.

In conclusion, the `while` will work just fine, but in terms of buffered moves and keeping the planner full, it only helps when there are explicitly short moves in the GCode itself, such as on curves, and generally this will only benefit cartesians.

But that's not a bad thing. I suggest testing some rounded objects to see how it affects the performance of the graphical LCD controller updates.

justmyopinion, 2016-12-07:
Any buffer in the process flow from UART to planner is a bottleneck so it is vital that noone is causing starving the planner when stressed.
I added this "while" to 1.02 one year ago when I was having stuttering with my printer and after optimizing my CHAR LCD updating down to 3ms this was one of the additions that gave the last and it has never shown any downsides and is running on my printer at this moment and i have added this to RC8 as well without any problem. it is however not an easy task to measure the actual performance improvement, but the idea was to assure the best flow.
it is far more destructive with the latest additions of tampering the ISR's with latest additions to RC8 and this should be reconsidered in my opinion.

thinkyhead, 2016-12-07:
> it has never shown any downsides and is running on my printer at this moment and i have added this to RC8 as well without any problem. … the idea was to assure the best flow

I think it's a good thing. We might yet do it.

> it is however not an easy task to measure the actual performance improvement

Oscilloscope on a test pin FTW.

> it is far more destructive … tampering the ISR's … and this should be reconsidered in my opinion.

It's out there (on purpose) to be evaluated now, so we'll know soon enough if allowing the UART interrupt to manage the serial buffer causes trouble. #5350 sits at the ready if needed.

fiveangle, 2017-01-11:
I'm having serious troubles with Octoprint running on RPi2 stuttering badly when refreshing the OP page after it's gone into "sleep" mode ( where Octoprint stops updating the page  and displays a "Octoprint has gone to sleep please refresh" page).  It seems related to the refresh launching new haproxy instances  temporarily starving Octoprint process for resources, causing  it to not be able to service Marlin UART streaming.  This is on a Cartesian (stock Printrbot Simple Metal).  It seems this proposed change could potentially  keep the buffer  at full more often, reducing the likelihood of it  emptying  due to the host not  been able to keep up.  This is only my theory, as  my stuttering may be related to the  integrated  ISR/UART timing of RC8 ( I haven't yet  tested with the revert to eliminate that possibility). 

 I don't have an oscilloscope but I will try the simple while() change and let you know the result.

AnHardt, 2017-01-11:
Octoprint is using pingpong communication. The above suggested change will have nearly no impact.
Instead try to increase `BLOCK_BUFFER_SIZE`.
```// The number of linear motions that can be in the plan at any give time.
// THE BLOCK_BUFFER_SIZE NEEDS TO BE A POWER OF 2, i.g. 8,16,32 because shifts and ors are used to do the ring-buffering.
#if ENABLED(SDSUPPORT)
  #define BLOCK_BUFFER_SIZE 16 // SD,LCD,Buttons take more memory, block buffer needs to be smaller
#else
  #define BLOCK_BUFFER_SIZE 16 // maximize block buffer
#endif
```

fiveangle, 2017-01-11:
I already tried that and have BLOCK_BUFFER_SIZE set to 128.  Sounds like I should focus on testing older FW.  This surfaced after changing 3 things: Octoprint from 1.2.x-beta dev stream to 1.3, and increasing print speed with available improvements of linear adv.  Guess I'll do so "ugly" 100mm/s prints on the old firmware(s) to see if they too stutter.  If so, i'll have to revert back OP and test that.

```
I must change only one thing at a time.
I must change only one thing at a time.
I must change only one thing at a time.
I must change only one thing at a ti...
```

thinkyhead, 2017-04-02:
> BLOCK_BUFFER_SIZE set to 128

That's ~12K of SRAM.

closed 2018-02-26 by thinkyhead

fiveangle, 2018-02-26:
> That's ~12K of SRAM.

No power of 2 up to and including 128 for this value made any improvement.  Interestingly, setting 128 did not crash or otherwise make any observable change.

thinkyhead, 2018-02-26:
Makes sense. Thank you for the followup!