MarlinFirmware/Marlin #5411: RC8 - Losing steps?

cfelicio, 2016-12-07:
Hi All,

I went today from 1.0.2 to RC8, and wow, I really liked the new stuff available, can't wait to try new things out! For now, I decided to not enable new stuff and just go with the bare minimum to make sure things work, and I noticed an issue. 

On 1.0.2, I could print at 60mm/s with movements at 120mm/s, without any issue, a decent 3d benchy would come out.

On RC8, the 3d benchy is slowly deforming towards one corner, indicating that I'm losing X and Y steps, but I can't hear any lost steps, so probably happening at a very small rate.

For now, I just bumped up the vref on the X and Y drivers, and reduced the travel speed to 100mm/s, and currently printing a new benchy, so far seems perfect.

Just wondering if there is anything that could have caused this between 1.0.2 and RC8, I saw the new option MINIMUM_STEPPER_PULSE, could this be necessary in my case? How was it handled in 1.0.2? Any other ideas?

Thanks! :-)

cfelicio, 2016-12-07:
Update, the new benchy is also distorted, but better. I actually now think I noticed the printer is moving faster than the set speed (60mm/s print, 100mm/s travel), feels like much faster than what I'm used to! I changed the speed using the display to 50% and it seems to be printing at the speed I expected, weird! Maybe is there a setting on config.h I did wrong? Any tips?

thinkyhead, 2016-12-07:
The acceleration code was repaired since 1.0.2, and so it does tend to accelerate faster with the old settings. However, I can't say why the top speed might seem higher. I suggest adjusting all your speeds down a little until you find a setting that works best. You can tune them with `M201` and `M203` while a print is underway (or use the LCD).

thinkyhead, 2016-12-07:
> Saw the new option `MINIMUM_STEPPER_PULSE`

You shouldn't really need that. It was added for special setups where fewer steppers were connected (in my own case, Z stepping was on a whole separate board), to ensure the steppers get a long enough pulse. It's kind of "hackish" so I don't recommend it unless position is lost. And note, if you are losing steps I would expect you to see some noticeable layer shifting.

Sebastianv650, 2016-12-07:
I guess the big problem we are facing with RC8 is the changed jerk speed behaviour if the people are copy & pasting their old settings. Before the change, the start/stop move speed was jerk speed / 2. Now, its jerk speed. Especialy if someone was using already agressive settings, now the steppers can't handle that any more..

cfelicio, 2016-12-07:
Yes, I pretty much copied / pasted my old values from 1.0.2 :(

Based on the information available, what should I cut down from the numbers below?

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 500, 500, 5, 25 }

/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_MAX_ACCELERATION      { 2000, 2000, 100, 4000 }

/**
 * Default Acceleration (change/s) change = mm/s
 * Override with M204
 *
 *   M204 P    Acceleration
 *   M204 R    Retract Acceleration
 *   M204 T    Travel Acceleration
 */
#define DEFAULT_ACCELERATION          2000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   3000    // X, Y, Z acceleration for travel (non printing) moves

/**
 * Default Jerk (mm/s)
 *
 * "Jerk" specifies the minimum speed change that requires acceleration.
 * When changing speed and direction, if the difference is less than the
 * value set here, it may happen instantaneously.
 */
#define DEFAULT_XJERK                 10.0
#define DEFAULT_YJERK                 10.0
#define DEFAULT_ZJERK                  0.4
#define DEFAULT_EJERK                  5.0

Sebastianv650, 2016-12-07:
Your values are in a reasonable range. I would start with x jerk and y jerk, simply cutting them down to 5. Just to see what happens. If it's not changing anything, return to 10 and reduce travel acceleration to 1500. If that's not helping, reduce default acceleration also. When you have found the value out of this set that makes the difference, increase it until you found the highest stable value.

And report back especialy if it's jerk or acceleration, I tend to say its jerk..

cfelicio, 2016-12-07:
Awesome, will try these tonight and let you know! Thanks so much for your help!

cfelicio, 2016-12-07:
And just in case someone ends up experiencing the same issue, is this the code change that is talked about?

https://github.com/MarlinFirmware/Marlin/pull/4936

Sebastianv650, 2016-12-07:
No, as far as I see this was only cleanup.  The big change was #4997.

cfelicio, 2016-12-09:
So I did more testing after slashing all the jerk by 1/2. It prints noticeably better, but still noticed the print shifted a tiny bit towards the top of the 3dbenchy.

Then, I slashed all the acceleration values by 1/2, and I did not notice any improvements compared to the previous scenario.

Then, I reduced the print speed in the sliced stl by about 30%. Still not much better.

I guess I will just slash jerk a tad more, and keep the acceleration values the same, and see if any improvements.

I can post comparison pictures if anyone is interested (1.0.2 marlin bench vs RC8 Benchy), I also posted on the CoreXY support group, but looks like nobody else is using RC8 yet. :(

thinkyhead, 2016-12-11:
Might be a good time to re-tune the stepper current and re-tighten all the stepper pulleys too.

cfelicio, 2016-12-11:
Ok, big findings on this during the weekend. For some reason, the new Marlin does not play nice with my old Cura (15.04). Here is what I found:

- Cura 15.04 prints quite decently on Marlin 1.0.2 with my settings (some blobs here and there)
- The same Cura version with the same profile create a terrible print on Marlin RC8 (Even after cutting acceleration / jerk to super super low values, and cutting print speed by 50%)

So, I decided to try another slicer. I remember trying Slic3r a long time ago, but it printed a terrible 3d benchy on my other printer (wanhao i3), hence why I've been using cura ever since.

Anyways, after putting RC8 back to the original jerk / acceleration (same as I had on 1.0.2), and trying a print with Slic3r, holy smokes, it came out better than what I had with Cura and Marlin 1.0.2, no blobs, super tight and clean!

At this point, I will just give up using cura with this printer, and will keep using slic3r. If anyone is interested, I can post the profiles for both slicers that I used.

Oh, another thing. For some reason, I had to recalibrate my e-steps as well, on 1.0.2 I was using 146, had to up that to 160 with Marlin RC8, not sure why the difference.

I even reverted back to 1.0.2 to double check, and sure enough, if I put 160 there, it will be too much!

Cheers and thanks for all the hard work on getting this release out! Despite the issues above, I still prefer RC8, and I think in the end it came out ahead with Slic3r :-)

brainscan, 2016-12-12:
You should try prusa's slic3r, they've made some good improvements

cfelicio, 2016-12-12:
wow, pretty cool! will definitely give it a try! thanks! ;-)

closed 2016-12-12 by cfelicio

thinkyhead, 2016-12-13:
I'm not surprised there may be some numerical differences. Let's account for them as bugs in the old version of Marin, until we hear otherwise.