MarlinFirmware/Marlin #5412: Using Additional Serial ports with hardware serial

apballard, 2016-12-07:
Hi all

I managed to reference and use the other serial ports outside of the .ino file using hardwareserial.

I'm not sure if it's a good solution, but I thought I'd share it with you, it works for me.

In whatever file I want to use it in, I simply include hardwareserial.h or Arduino.h

The problem is you need to make sure the interrupt vector for serial0 is not re-used, since Marlin serial already defines an isr for it.

In the same directory that has the hardwareserial.h file you will find a file called hardwareserial0.h.
Copy that file into your project/Marlin directory. Edit it and comment out all the lines with isr related definitions and their #ifdef's. You want this file to take priority over the built in one.

That's it. Serial1,2,3 now at your finger tips.

I've used it to interface with my nextion display. I'll get some pics together and video later for you to see.

thinkyhead, 2016-12-07:
That's awesome. I seem to remember we had another thread with generally the same idea a while ago. I guess we should document this [somewhere](http://marlinfw.org/).

apballard, 2016-12-08:
Here a couple videos of what I'm doing with my display. There are a few bugs, still got to implement settings feature and so forth.  I like pretty things, and the display appeals to me, although I must tweak some graphical components. I'm not a graphics designer, so did my best.

Startup
https://youtu.be/MvUMv2IPnMw

Quick Demo
https://youtu.be/4NnOwrkCkYk

apballard, 2016-12-08:
Sorry, need to add to my original post, if you include hardwareserial.h directly, you must make sure that you include avr/io.h first or it's in scope.  The reason is that it handles the creation of macro's for registers that determine if serial 1,2,3 are available. Arduino.h handles this though, so just including Arduino.h makes it easier.

Roxy-3D, 2016-12-08:
Very nice!   And it can do quick double clicks too, right?  Because people are already starting to use that to get fast and easy access to the Z-BabyStepping!

Roxy-3D, 2016-12-08:
Did you modify ultralcd.cpp to handle the touch screen?

apballard, 2016-12-08:
@Roxy-3D I'm still trying to work out how to use git to update a fork with my code.  Mind you, I would never call myself a real developer, so I'm sure there is plenty wrong with my code, so no judging please!

I basically created a new define in configuration.h called experimental_lcd and disabled all other displays.

I then created new experimental_lcd.h and experimental_lcd.cpp and implemented the basic ultralcd functions in those files.  And then used those functions to implement communication with the screen.  The screen has its own firmware with 32bit Microcontroller and uses serial commands, so i just send seial data to the screen. I do try to limit communication to the screen, so I have a printer object that keeps the last values sent and only sends new data to the screen if the values change.  Takes a bit of memory, but it is much less than ultralcd.

Not sure if that is improving performance, have not actually tested yet. I have the serial running at 115K.  Also i limit updates to a minimum of 2Hz (500ms) and incoming commands to 5Hz (200ms).  I check for serial data received from the screen on every iteration of the loop().

I basically wanted the least impact on the native marlin code.  So all I have to do is replace each #include ultralcd.h with an #if and including the experimental_lcd.h.

I want to implement babystepping next.  I will most probably make the X,Y,Z bar on the main screen a quick button to babystepping, so one tap and you in.  I've also been toying with the idea of keeping an encoder to make adjustments, but not so keen on the idea.

I normally look at implemented logic in ultralcd and see if i copy that over.  I want to see if i can decouple baby stepping code from the lcd, so i can reuse it.  It would be nice if babystepping code was already decoupled and can be called outside ultralcd.h, maybe in functions in its own file.

For some commands from the screen I literally receive gcode and put that on the command queue (G28, G29), in other cases I modify variables directly or through functions, like temperature settings.

Also need to implement some settings, eg lin_advance K, Z-offset and others, fwretract and some others.

Also want to implement Flow adjustments.

justmyopinion, 2016-12-08:
> Startup
> https://youtu.be/MvUMv2IPnMw
> Quick Demo
> https://youtu.be/4NnOwrkCkYk

Very inspiring

apballard, 2016-12-08:
@Roxy-3D 
As I said, know little about Github, but I think here is a link to my code......

[Experimental_LCD Branch](https://github.com/apballard/Marlin/tree/RCBugFix/Marlin)

thinkyhead, 2016-12-09:
> I know little about Github, but I think here is a link to my code

With new features it's always best to make a new branch named for the feature and not to use one that has the same name as one of the main project branches. You will find it easier to keep your work up-to-date with our changing branches.

apballard, 2016-12-09:
Thanks for the advice! I'm trying to work out how to update the branch with the changes from rcbugfix. My head is swimming! Maybe I should get a drink!

apballard, 2016-12-09:
Looks like that drink helped.....

git remote add upstream git://github.com/MarlinFirmware/Marlin.git
git fetch upstream
git pull upstream rcbugfix

Then I just had to merge.  Think i'm getting the hang of this.

closed 2016-12-10 by apballard