MarlinFirmware/Marlin #5415: [Feature Request / Idea]

boelle, 2016-12-07:
Basicly just a placeholder / reminder

But for those of us that don't have a always on probe it could be nice to be able to store the bed correction info.

It would also reduce time spent as you don't have to do a G29 before every print. Good for those that have good machines and just want to fire a G29 every now and then.

Tannoo, 2016-12-07:
Enable eprom support and you'll have that option.

boelle, 2016-12-07:
i have and no its not there :-D

@thinkyhead has said so and dont have the time yet to implement it

thinkyhead, 2016-12-07:
This is a duplicate issue. You're praying for rain that will come when it comes.

closed 2016-12-07 by thinkyhead

boelle, 2016-12-07:
where is the dublicate?

thinkyhead, 2016-12-07:
Which one? This has been requested several times. It's on the list of things to do.

boelle, 2016-12-07:
i did not know that... 

one of the others could have been https://github.com/MarlinFirmware/Marlin/issues/4697