MarlinFirmware/Marlin #5416: [Feature Request / Idea] Use probe and dual Z drivers to make Z's in sync -- Make X parallel to bed

boelle, 2016-12-07:
Good for all Mendel90 printers that have 2 of 4 bed screws fixed

Normal procedure is to use a dial gauge and with motors off adjust the 2 Z's up and down until they are the same. if you dont have a dial gauge you will have to resort to the old paper method which has its flaws

oysteinkrog, 2016-12-07:
This would be cool and quite useful. There should be a warning on the option that it can cause problems if bed is very uneven as the bearings on Z axis could then become unaligned and cause binding.

boelle, 2016-12-07:
of course. on the mendel90 thou i have never experienced that at the 2 fixed screws.

oysteinkrog, 2016-12-07:
Most printers with ball bearings would not have a binding problem. Printers with bushings are more sensitive to misalignment. I use bushings and I can definitely notice it if they are misaligned.

boelle, 2016-12-07:
mine have ball bearings now, but after xmas they will have bushings from Igus.

if i on purpose get them out of sync the ball bearings will bind a bit. but gravity wins.

and the first layer will have signs of the problem.

so either way its a good one to have

thinkyhead, 2016-12-07:
The procedure might work thus:
- Move X to left end, probe down and record the position (z1)
- Move X to right end, probe down and record the position (z2)
- If z2 > z1
  - Raise Z1 by 1/2 of the difference. Lower Z2 by half the difference.
- else
  - Lower Z1 by 1/2 of the difference. Raise Z2 by half the difference.

boelle, 2016-12-07:
that would work yes

i would inject a check before rehome as home could mean moving up (at least for a stock mendel90)

thinkyhead, 2016-12-07:
> i would inject a check before rehome

I guess we only need a re-home in the case where the Z probe is used to home Z. Otherwise, we can just move XY home and re-write the Z position based on which end of X we're at.

thinkyhead, 2016-12-07:
> re-write the Z position based on which end of X we're at.

On second thought, we still lose accuracy this way. So a re-home really is required.

boelle, 2016-12-07:
hmm.... i dont know if my thought would work... 

but do a probe half way ? if X bars are parallel to the bed all 3 points should be the same distance

Roxy-3D, 2016-12-07:
I think I would change the algorithm to:
- Move X to left end, probe down and record the position (z1)
- Move X to right end, probe down and record the position (z2)
- If z2 > z1
-- Raise Z1 by 1/4 of the difference. Lower Z2 by 1/4 the difference.
- else
--Lower Z1 by 1/4 of the difference. Raise Z2 by 1/4 the difference.
- Repeat if abs(Z1-Z2) is greater than .01mm

The reason is moving one end of the X-Carraige is going to affect the Z Position of the other end.  (You are not measuring the difference in Z-Height at the threads.   Because of the probe offset, you are forced to measure it some where in the middle.   I have no experimental data to back up this claim, but my guess is correcting by 1/2 of the difference is going to be too much.

If somebody implements this...  It will be interesting to see if that guess turns out to be correct!

thinkyhead, 2016-12-08:
@Roxy-3D True that it's imperfect. When using a probe to measure the Z height, we're really only measuring the difference at the probed points. If the bed is tilted left-to-right, it will result in the Z carriages ends being slightly off. If the bed is tilted only in the back, for example, but the probing is only done towards the front, then the adjustment won't apply everywhere. To be a more meaningful measurement, it needs to probe at a few points along the left edge, and then a few points along the right edge, and come up with an average of how much it's tilted. The adjustment to the current Z position could then be based on the current X position. Or, move the X to the half-way point between the two measured points and assume that the Z at that position will barely change.

Of course all that is obviated if after the adjustment it just re-does `G28`.

Roxy-3D, 2016-12-08:
> Of course all that is obviated if after the adjustment it just re-does G28.

Yes...  But...   If the next iteration of Homing corrects some of the previous X Axis error....   The machine is going to stabilize over time...

boelle, 2016-12-08:
" If the bed is tilted only in the back, "

it takes some fantasy i will give you that.

on a mendel90 everything is reversed... so my back is the normal front and left and right changes place also

how i level it with dial gauge etc is:

at the front where the adjustmens screws are fixed in place and never changed i go left and right and adjust the Z screws by hand (after a M84). As roxy says once you touch one it changes the other, hence you have to go back and forth a lot to get it right. once that is in place i move the bed to the back where i can adjust height in the 2 corners. 

i do the bed move by hand and without move Z up and down. i know use the adjustmens screws unti the dialgauge reads the same as in the front. this is a lot faster as i dont have to go back and forth. once the 2 adjustmens screws are ok i recheck all 4 corners. if any of them differs i start over again.

dosono, 2016-12-10:
This is a geart , With auto leveling, mesh and now tilt balancing Marlin will provide an end to end print bed calibration...
It can also work for a print bed that moves on z axis. I.e. for core xy printers and other printer with a moving bed on z. The ability to control the bed tilt with 3 or 4 motors having their own dedicated driver will bring it very close to the desired "plug and print" concept.

boelle, 2016-12-11:
@dosono i think you misunderstand this one

its mostly to bring the 2 Z's in sync. and its mostly where the bed is only adjusted on 2 out of 4 screws

boelle, 2016-12-11:
https://github.com/nophead/Mendel90/blob/master/dibond/manual/Mendel90_Dibond.pdf

page 72 is how i do it

just remember when it says back its front

and left becomes right

justmyopinion, 2016-12-11:
Make it simple...
Mount a microswitch equipped with LED-resistor with 5V/0 and place mount on bed anywhere on bed.
Dive extruder into microswitch actuator and note z position where led goes on . adjust bed if different values for different positions.
You should only have 3 screws adjusting your bed. it is not convenient/impossible to use 4 :-)

boelle, 2016-12-11:
@justmyopinion "You should only have 3 screws adjusting your bed. it is not convenient/impossible to use 4 :-)"

Sadly the mendel90 has 4, but only the 2 can be adjusted.

and many kits of the mendel90 has been sold over the years so i'm not the only one 

impossible? no.... inconvenient yes, but i dont want to mod it more than i have

forReason, 2016-12-13:
I would LOVE this featue. Could possibly save me 1 or even 2 endstops.

boelle, 2017-01-07:
problem with endstops are that they can move if not mounted correctly

and yes why not use the probe if you have it :-D

boelle, 2017-01-08:
closing this one as i no longer have a interest in it

will be switching to Josef Prusa's version of marlin and get a bed made that can take advantage of his XYZ correstions

closed 2017-01-08 by boelle