MarlinFirmware/Marlin #5417: extruder vibration or waves on side

akaJes, 2016-12-07:
In my case  have some strange behavior of hotend/motor with printed models.
There is a visible vibration on the outer side and near holes
The same as at this post https://github.com/MarlinFirmware/Marlin/issues/5222#issuecomment-261058840 
I have the same or bigger effect as on these pictures

(left side)
[image]

(and right side of hole)
[image]

and this #4766 
It seems like on a braking vibration at end of path before a path turning

after digging the code i found where its calculation and its method
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/stepper.cpp#L250
and first idea was to decrease braking acceleration
here https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/planner.cpp#L177

dividing in twice -accel 
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/planner.cpp#L187
but it is only an idea and results i can get only in an evening when i try it

Could someone have other solutions?

Drakelive, 2016-12-07:
I have the same problem on different delta Kossel

stigjoergensen, 2016-12-07:
Im relativ new to 3d printing..

But i had the same "ringing" and for me the solution was a combination of belt too lose and jerk settings to high.

akaJes, 2016-12-07:
@Drakelive did you try another firmwares/options/hardware/motors to reduce this effect?
I have cartesian Graber (prusa style) RC-8 and it was with all previous marlin version
in https://github.com/MarlinFirmware/Marlin/issues/5222 is resolving extrusion amount, if i understand right it can be also an issue - i will check it, printing thin wall - and if it will have symmetrical thickness - this is extrusion

Drakelive, 2016-12-07:
@akaJes  This problem I encountered on MarlinFirmware and MarlinKimbra (8bit and 32bit), drv8825 drivers with 1/32 microstepping.
The mechanics was assembalta and calibrated very well. On my kosselMini, I installed couplings IGUS KBRM-03. The effector is very stable and I have no involuntary movement. Generally this problem I can see with the calibration cube. In particular on changes of direction. I thought it would be a wrong motion management during decelerations.

Drk

Kaibob2, 2016-12-07:
On the second picture there is ringing on the left side. This effect can only be reduced by lowering acceleration and/or jerk settings. I guess you don't ask for this, but the inconsistency when printing holes.  The reason for me to upload these pictures, especially the ones in #4766 where the inconsistent extrusion when leaving holes and continuing the outer perimeter. I haven't found a solution yet. `lin_advance` reduces it significantly but brings other issues up, especially on setups with long Bowden tubes

justmyopinion, 2016-12-07:
In my opinion, this is not a Marlin issue, but is recognized easily in RepRap forums and the answer is probably that your acceleration is far too high to the masses moved in your construction.
Try to lower your accelerations and maybe speed and avoid travel speeds to be different from printing speeds

Drakelive, 2016-12-07:
Ok, I can try.

akaJes, 2016-12-07:
@Kaibob2 @Drakelive 
i have designed a simple thin wall cube for testing http://www.thingiverse.com/thing:1947480
and will try to check if deceleration of braking  will make some changes
@justmyopinion
lowering of a travel  speed or  acceleration will not decide this issue
lowering of an accelerations(decelerations) of printing speed is the right way - but printing will be slower

justmyopinion, 2016-12-07:
> Akajes
My experience with Travel speed is with holes, and is caused when extruder is passing through the hole at higher speed and has to continue printing at lower speed, but you may see at different, no problem.

akaJes, 2016-12-07:
@justmyopinion 
sources have simple explanation https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/stepper.cpp#L250
```
/**
 *         __________________________
 *        /|                        |\     _________________         ^
 *       / |                        | \   /|               |\        |
 *      /  |                        |  \ / |               | \       s
 *     /   |                        |   |  |               |  \      p
 *    /    |                        |   |  |               |   \     e
 *   +-----+------------------------+---+--+---------------+----+    e
 *   |               BLOCK 1            |      BLOCK 2          |    d
 *
 *                           time ----->
 *
 *  The trapezoid is the shape the speed curve over time. It starts at block->initial_rate, accelerates
 *  first block->accelerate_until step_events_completed, then keeps going at constant speed until
 *  step_events_completed reaches block->decelerate_after after which it decelerates until the trapezoid generator is reset.
 *  The slope of acceleration is calculated using v = u + at where t is the accumulated timer values of the steps so far.
 */
```
all motions is working with this method
and i want to extend right sides of trapezoids
just to start braking earlier

justmyopinion, 2016-12-07:
> and i want to extend right sides of trapezoids just to start braking earlier

I may misinterprete, but to me this means lower acceleration

akaJes, 2016-12-07:
@justmyopinion lower acceleration will extend both sides of trapezoids
there is no separated options in config for it

justmyopinion, 2016-12-07:
Ok if that is what you are looking for. That would probaly need some hefty coding :-)

akaJes, 2016-12-07:
@justmyopinion NO! :)
just change one row for test https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/planner.cpp#L187
          decelerate_steps = floor(estimate_acceleration_distance(block->nominal_rate, final_rate, -accel)),
to 
          decelerate_steps = floor(estimate_acceleration_distance(block->nominal_rate, final_rate, -accel/2)),
or
          decelerate_steps = floor(estimate_acceleration_distance(block->nominal_rate, final_rate, -accel/10)),
or try another divider

Sebastianv650, 2016-12-07:
Some month ago I was also adressing this ringing issue on my printer. Mechanical axis improvements did not change anything, lowering acceleration and jerk is helping as you can read often on printer web pages.

I could write a long article, but to make it short after printing thin single line width walls and inspecting them with strong magnifiers I'm very sure now that the ringing is not realy comming from XY axis. It's looking as if the filament flow is not stable during acceleration inside the hot end. Therefore it's pulsing, resulting in a tiny variation of filament flow rate out of the nozzle.
The following is only my theory because I found no way to proove it up to now:
If you have no filament flow with a heated hot end, after some time you get a stable thermal situation. That means, there is a transition between solid and fluid filament at some point X around the heat brake area.
If we now speed up the e extruder, at first we will move down this half-solid transition point. X moves to the nozzle. This means quite a lot of friction, and as you have a not oiled joint, this movement can occur in steps (friction stops movement, force inside the filament is increasing until friction breakes free - like during an earthquake). That's the part where we see ringing.
If the extruder final speed is reached (or somewhere around this time point), the transition point X will find a thermal equilibrium again at some point below or in the heat brake. The faster we extruder, the more it will be below the heat breake - but it will be always at a stable point. In this phase, we have a stable extrusion and no (or nearly no) ringing.

No idea how that can be eleminated, but it proves that a heat brake should be polished ;)

macst34, 2016-12-08:
Hi @Sebastianv650,

That's a pretty interesting theory for the root cause of ringing. Maybe you can test it by printing a single line pattern on top of a raft (so its printing at full height) while manually tweaking the extruder feed rates.

For ringing, i'm just gonna put these here:
[https://www.simplify3d.com/support/print-quality-troubleshooting/#vibrations-and-ringing]
[http://forums.reprap.org/read.php?1,214990]
[http://support.3dverkstan.se/article/23-a-visual-ultimaker-troubleshooting-guide#ringing]

take care,

Michael

akaJes, 2016-12-08:
i have enabled a decelerate  but do not have better results
another way is dig into a filament reaction when flow speed is changing
maybe extruder's motor must start earlier than other motors?
something like a time shifting?

justmyopinion, 2016-12-08:
> akaJes
> https://www.simplify3d.com/support/print-quality-troubleshooting/
> http://forums.reprap.org/read.php?1,214990
> http://support.3dverkstan.se/article/23-a-visual-ultimaker-troubleshooting-guide#ringing
> 

May I suggest you read carefully the links that macst34 has presented, It is not always problems are found in sw . But of course you have to believe in what you see. :-)

akaJes, 2016-12-08:
@justmyopinion I know how to decrease acceleration and speed of printing :) but i dont want! I search a way how to increase it

stigjoergensen, 2016-12-08:
Its always a compromise here...

Its quality vs speed when it comes to 3d printing - at least with FDM
prints.

In my hunt for speed i have found that lowering the jerks and accelerations
provided me with the option to increase the overall print speed.- but that
only benefits large prints.. it made small prints slower.
Im hoping that one day, we might be able to select a profile that will
adjust the paramters to what we what to happen with the print..
So that we could have a profile for fast print (that have artifacts), slow
prints where the details is spot on, a profile for better strength . but
these profile might belong in the slicers, or a combination of slicer and
machine - but i dont see that happening anytime soon as the whole process
is more or less separate steps - generate stl (or what ever the standard
might end up being), slice the model, stream the sliced model, and finally
do the physical printing.

if all these steps were better integrated then the model in step 1 could
know about the limitations of the printer, the slicer could make better
judgement on how to slice it, the streaming could optimize how its send
with - and then finally the printer could do what its supposed to do.

Kaibob2, 2016-12-08:
Only for clarification. Is this thread dealing with this issue
[image]
or with this issue
[image]

The second is clearly ringing and can be reduced by lowering weight, acceleration or jerk.
These are different phenomena. Ringing is quite easy to become rid of, the reason for the issue in the first picture is still unclear.
My guess is that @akaJes originally addressed the issue in the first picture.
My #4766 ended nowhere too.

jbrazio, 2016-12-13:
Interesting discussion topic.
On my printer is very common to see artefacts like to ones on the first picture after hole on a wall.
Do you use retraction ? I somehow culprit retraction for that artefact, retracting changes the nozzle pressure [and temperature] making those layers to be slightly different.

Kaibob2, 2016-12-13:
@jbrazio There are indeed unretractions (layer start points) at this corner of the hole! Not on every layer, but this might explain the inconsistency on the right corner of the object (one thick, one thin layer).
[image]

macst34, 2016-12-13:
@stigjoergensen 

That's a great idea...though current slicers do not adjust jerk for large prints versus small details, at this moment, perhaps we can change the jerk values for certain features or layers that have finer detail. The gcode output by slicers (such as 3dsimplify) add comments so we can tell which layer it is working on...so we can manually add gcode to manipulate jerk values in-line during printing around the area of interest.

For example, we use the default marlin jerk values of

`M205 X20.00 Y20.00 Z0.4 E5.0` 

when printing part of the print, but change the jerk values for areas/layers with fine detail by
using

`M205 X10.00 Y10.00 Z0.20 E2.50`  (or any other reasonable values)

when needed and continue changing them to suit the desired print characteristics.

For easy of use, this would need to be integrated into the slicer at some point....but nothing is stopping us from doing it now by hand...

Do you think this is something worthwhile to explore?

Michael

stigjoergensen, 2016-12-13:
@macst34 it might be worth to do these manual test, to measure the improvement vs time it takes to print.

if it works and have noticeable benefits, then perhaps we can start to discuss how to implement them in marlin with look-a-head - or find a different method of acceleration. today i think its linear, but im starting to think it would be better to ramp up slower like a form of parabolic. curve

jbrazio, 2016-12-13:
> @Kaibob2 There are indeed unretractions (layer start points) at this corner of the hole! Not on every layer, but this might explain the inconsistency on the right corner of the object (one thick, one thin layer).

Good point, so this could be caused by the slicer and not by the firmware ?

macst34, 2016-12-14:
@stigjoergensen 

I think some empirical values are needed to shed some light on the effect of adjusting jerk midprint. Our goal would be to manipulate jerk by changing the values to suit the feature printed.

To start, in general the idea would be to use the default marlin jerk values for large feature printing and swap to less jerk for fine features.

I think a simple test piece would be helpful to try these changes on. Feel free to chime in and suggest changes to this one I created very quickly over at tinkercad.

[image]

https://www.tinkercad.com/things/at83QHXh2wE-jerk-testing

based on this simple print, perhaps we can get some more data.

Michael

thinkyhead, 2016-12-14:
The `LIN_ADVANCE` feature is maturing. I suggest giving it a try to see how it affects surface quality.

closed 2017-10-31 by thinkyhead