MarlinFirmware/Marlin #5418: G28, G29 double touch

BillStruve, 2016-12-07:
Is there a setting in Configure.h for G28, like there is for G29 to stop double touch?
e.g.
// Use double touch for probing
//#define PROBE_DOUBLE_TOUCH

Blue-Marlin, 2016-12-07:
No.
When probing, we know where we are and where we have to slow down.
When homing we would have to move all the way with the slow speed - what would last forever.

BillStruve, 2016-12-07:
I see, OK.

closed 2016-12-07 by BillStruve

macst34, 2016-12-07:
@BillStruve 

If you really want to prevent occuring, the function name for the 2nd touch is "home_bump_mm" in Marlin_main.cpp. If you look into it some more, maybe you can share what you learn about it and what you did to get the outcome you wanted.

Happy hacking, 
Michael

thinkyhead, 2016-12-07:
The `home_bump_mm` defaults come from here, btw:
```cpp
#define X_HOME_BUMP_MM 5
#define Y_HOME_BUMP_MM 5
#define Z_HOME_BUMP_MM 2
```

BillStruve, 2016-12-08:
I'll put it on the to-do list, thanks.