MarlinFirmware/Marlin #5421: Homing issue

NimrodQuimbus, 2016-12-07:
When the printer homes at the start of the print, and Y axis will move/home even when it is at home and the limit switch is made.
I am not sure if the same is with the x of z

macst34, 2016-12-07:
Hi @NimrodQuimbus

(I believe) this is intended behavior...When a endstop is hit during homing, the axis is backed off and slowly re-engages the endstop. This happens for all homed axes.  This "check" just ensures the accuracy of your endstops in case you have some really fast homing speeds and the endstop is slightly deflected by the printhead moving towards it at high speed.

happy hacking,
Michael

NimrodQuimbus, 2016-12-07:
No, you misread what I said.

Y axis is homing when it is already at home, which tears the belt from it mounting.
If anything, it should check to see if an axis is at home, then if so, back off until that contact is opened, then re-home.

macst34, 2016-12-08:
Hi @NimrodQuimbus,

Sorry if I misread. If the Y axis is trying to go past the Y endstop when it is already home, that does seem to be a problem. Perhaps you can shut your steppers off and home the y-axis manually and then issue a 

`M119`

to see the status of the endstops and confirm that the issue is not mechanical. Other than that, can you tell us anymore about the issue? i.e. printer type, endstop type, configuration.h settings, etc. More information would likely be helpful since we don't have much context for the issue.

Michael

thinkyhead, 2016-12-09:
Endstops aren't tested until motion starts, but they should respond so quickly that only a few steps transpire. If it's moving more than a millimeter, then we should take a look and see what's going on.

closed 2017-10-31 by thinkyhead

NimrodQuimbus, 2017-10-31:
My Y axis will rip the belt out, but the x axis is fine.I just make sure everything is off of the stops before I print. 

 

 Closed #5421.—

   

|  | Virus-free. www.avg.com  |