MarlinFirmware/Marlin #5422: expected initializer before 'homing_feedrate_mm_s'

LeChuck71, 2016-12-07:
Big Troubles compiling the latest RC8, getting the next error message.
Any tips on avoiding them? RC3 and UBL compiles without problems

----------------------------------------------------------------------------------------------

`Arduino: 1.7.10 (Windows 7), Platine: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Marlin_main.cpp:378:17: error: expected initializer before 'homing_feedrate_mm_s'

 float constexpr homing_feedrate_mm_s[] = {

                 ^

Marlin_main.cpp: In function 'float get_homing_bump_feedrate(AxisEnum)':

Marlin_main.cpp:1528:17: error: expected initializer before 'homing_bump_divisor'

   int constexpr homing_bump_divisor[] = HOMING_BUMP_DIVISOR;

                 ^

Marlin_main.cpp:1529:13: error: 'homing_bump_divisor' was not declared in this scope

   int hbd = homing_bump_divisor[axis];

             ^

Marlin_main.cpp:1535:10: error: 'homing_feedrate_mm_s' was not declared in this scope

   return homing_feedrate_mm_s[axis] / hbd;

          ^

Marlin_main.cpp: In function 'void do_blocking_move_to(const float&, const float&, const float&, const float&)':

Marlin_main.cpp:1669:43: error: 'homing_feedrate_mm_s' was not declared in this scope

       feedrate_mm_s = fr_mm_s ? fr_mm_s : homing_feedrate_mm_s[Z_AXIS];

                                           ^

Marlin_main.cpp:1681:43: error: 'homing_feedrate_mm_s' was not declared in this scope

       feedrate_mm_s = fr_mm_s ? fr_mm_s : homing_feedrate_mm_s[Z_AXIS];

                                           ^

Marlin_main.cpp: In function 'void do_homing_move(AxisEnum, float, float)':

Marlin_main.cpp:2570:149: error: 'homing_feedrate_mm_s' was not declared in this scope

     planner.buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], fr_mm_s ? fr_mm_s : homing_feedrate_mm_s[axis], active_extruder);

`

esenapaj, 2016-12-07:
If you use IDE by Arduino.cc https://www.arduino.cc/en/Main/Software instead of by Arduino.org, perhaps problem is solved.

thinkyhead, 2016-12-07:
@esenapaj is correct. You can't use the Arduino.org compiler. We'll have to figure out a way to throw an error in that case.

closed 2016-12-07 by thinkyhead

ChrisTerBeke, 2017-01-10:
If changing your IDE doesn't work, try removing your platform.txt file from the board config folder. This will make Arduino use the (correct) defaults.

Blue-Marlin, 2017-01-10:
> You can't use the Arduino.org compiler. We'll have to figure out a way to throw an error in that case.

Some problems solve themselves. From 1.8.0 on the versions are unified again.

`constexpr` came with gcc-4.9.2 into the Arduinio (cc) 1.6.10 and with the unification into what you can get from arduino.org (1.8.0)