MarlinFirmware/Marlin #5423: Discussion: Is Sanity Check of Arduino version the right thing to do?

neildarlow, 2016-12-07:
The Arduino version requirement for RC7 was 1.6.0+ and for RC8 is 1.6.8+.

This may be fine for bundled Arduino environments but users of e.g. GNU/Linux distributions may have separate Arduino and avr-gcc-c++ packages.

Taking my fedora 24 installation, I had Arduino-1.6.0 and avr-gcc-c++-4.9.3 which built RC7 fine.
My recent fedora 25 update gave me Arduino-1.6.4 and avr-gcc-c++-6.2.0 which appears to build RC8.

Is it the case that Marlin is more dependant on the version of the underlying avr-gcc-c++ rather than the Arduino IDE version?

If this is indeed the case then would it be more sensible to base the Sanity Check on a minimum avr-gcc-c++ version instead?

thinkyhead, 2016-12-07:
The codebase requires C++11 support.

neildarlow, 2016-12-08:
In RC8, the Arduino version requirement has been bumped to 1.6.8.

C++11 support was added in the GCC-4.7 series but the avr library support may lag behind that. My concern is that the decision to use a particular version of Arduino may cause issues for some users who are unable to use a particular, newer, version.

In my case I apply a local customisation patch and I've just had to extend that to patch SanityCheck.h also. Of couse adequate testing of the generated code is also required.

jbrazio, 2016-12-08:
As of today we are not a standalone C/C++ program, we rely on the Arduino core, thus the only supported method to build Marlin is to use the Arduino environment, each version of the environment (IDE) comes with a bundled toolchain.

The small percentage of users which may want to build Marlin out of the IDE to use newer versions of avr-libc or avr-gcc should be knowledge enough to bypass any possible issues an `#error` pre-processor clause may cause them.

bobc, 2016-12-08:
> My recent fedora 25 update gave me Arduino-1.6.4 and avr-gcc-c++-6.2.0

What prevents you installing latest Arduino IDE from https://www.arduino.cc/en/Main/Software?

neildarlow, 2016-12-08:
There is nothing that stops me installing the latest Arduino.

The premise of this issue is "Should it be necessary to install the latest Arduino if a packaging of Arduino has the required level of C++ support albeit at not the latest version of the IDE".

To put things in context, I produced a variant of Marlin which was configured for nophead's Mendel90 with attached (or absent) Panelolu2 controller. This had little in the way of Arduino version requirement and users were easily able to build and use the firmware with a minimum of support.

If the Marlin project takes the view that users should use the latest Arduino (which isn't always a good idea) then I'm happy to go along with that but I'd rather not have to deal with end-users questions on how to install the latest Arduino on "name your OS here" when an existing packaging of Arduino is up to the job.

fiveangle, 2016-12-09:
I haven't tracked this down yet, so didn't file any bug, but I got the version requirement not met error when trying to build printrboard target on current platform.io.

printrboard recipe from current platformio.h:

```
[env:printrboard]
platform = teensy
framework = arduino
board = teensy20pp
build_flags =  -I $BUILDSRC_DIR -D MOTHERBOARD=BOARD_PRINTRBOARD
# Bug in arduino framework does not allow boards running at 20Mhz
#board_f_cpu = 20000000L
```

I gave up and opened in current Arduino, but if I revisit I'll let you know what I find out.

thinkyhead, 2016-12-09:
> If the Marlin project takes the view that users should use the latest Arduino (which isn't always a good idea) then I'm happy to go along with that

Comment out the #error message, and if it builds, great! The end.

bobc, 2016-12-09:
> Is it the case that Marlin is more dependant on the version of the underlying avr-gcc-c++ rather than the Arduino IDE version?
> 
> If this is indeed the case then would it be more sensible to base the Sanity Check on a minimum avr-gcc-c++ version instead?

I think the answer to both questions is No, because there are also many dependencies on the Arduino builder behaviour, standard libraries, the Arduino core, as well as compiler. 

Arduino IDE 1.6.8 was released in March 2016, I don't think it an unreasonable baseline requirement. The baseline version is chosen to help developers to provide a better software for users. Given that Marlin is entirely voluntary, it is not unreasonable to expect users to provide some minimal effort of their own. I would say finding out how to upgrade their computer with free software falls within the realm of minimal effort.

neildarlow, 2016-12-09:
Thanks for your time everyone. I'll take this as closed.

closed 2016-12-09 by neildarlow