MarlinFirmware/Marlin #5425: redefinition of 'bool lcd_detected()'

LeavOA, 2016-12-07:
```
In file included from C:\Users\Leav\AppData\Local\Temp\build6805f3717b9429efbec8834515d5dff2.tmp\sketch\ultralcd.cpp:23:0:

C:\Users\Leav\AppData\Local\Temp\build6805f3717b9429efbec8834515d5dff2.tmp\sketch\ultralcd.h:50:17: note: 'bool lcd_detected()' previously defined here

     inline bool lcd_detected() { return true; }

                 ^

Using library Wire at version 1.0 in folder: C:\Users\Leav\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.14\libraries\Wire 
Using library LiquidTWI2 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidTWI2 (legacy)
exit status 1
redefinition of 'bool lcd_detected()'
```

I have been very wrong before, so i'll tread lightly:

UltraLCD.h seems to have this inline definition of lcd_detected() on line 47:
```
  #if (ENABLED(LCD_I2C_TYPE_MCP23017) || ENABLED(LCD_I2C_TYPE_MCP23008)) && ENABLED(DETECT_DEVICE)
    bool lcd_detected();
  #else
    inline bool lcd_detected() { return true; }
#endif
```

which seems to be clashing with this code from UltraLCD.cpp around line 3155:
```
  #if (ENABLED(LCD_I2C_TYPE_MCP23017) || ENABLED(LCD_I2C_TYPE_MCP23008)) && ENABLED(DETECT_DEVICE)
    bool lcd_detected() { return lcd.LcdDetected() == 1; }
  #endif

```

Hope this is a real bug...

Thanks.

thinkyhead, 2016-12-08:
What LCD have you got selected in your config?

thinkyhead, 2016-12-08:
Thanks for the report! Now patched in `RCBugFix`.

closed 2016-12-08 by thinkyhead