MarlinFirmware/Marlin #5426: Marlin RAMPS Issues

MH15, 2016-12-07:
I've been building a custom 3d printer, and have some issues that I can't seem to solve. The Full Graphic Controller says the hotend temperature is always around 173 C, and the card always says either "card inserted" or "card removed".

Any help would be immensely appreciated.

macst34, 2016-12-07:
Hi @MH15 MH15,

Is that the temperature without heating?  If so, what type of thermistor are you using and what does your configuration.h define for thermistor type?

`#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 1`

Do you have your board set correctly?

`#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB
#endif`

maybe you can include a copy of your configuration.h or copy it to dropbox and post a link back here.
(don't just copy-paste it here since the markup screws it up)

Michael

MH15, 2016-12-07:
https://gist.github.com/MH15/b6e84d5626d092009bdc58e2ffc8a7e9
The link to my config file.

brainscan, 2016-12-07:
Which version of Marlin is that? Seems very old. You need to uncomment

//#define SDSUPPORT // Enable SD Card Support in Hardware Console

https://gist.github.com/MH15/b6e84d5626d092009bdc58e2ffc8a7e9

macst34, 2016-12-08:
Hi @MH15,

Do you have another thermistor you can try?

Your config looks alright except for what @brainscan said about enabling the SDcard.

Michael

MH15, 2016-12-08:
@macst34, I did have some extra thermistors that I ordered so I took @brainscan's advice and got the new version of Marlin and everything worked with a new thermistor. Huh. But now, all 3 axes (in CoreXY) will only move in one direction, then start wailing.

Anybody ever experienced this before?

macst34, 2016-12-08:
Hi @MH15,

Are your endstops configured correctly?

it might be helpful to do a line by line compare of your old configuration.h and insert the appropriate values into the new configuration.h

Things like x,y,z,e steps need to make it over there. There's probably some feedrate and acceleration type stuff that would be helpful to get a starting point from.
Use a program like "beyond compare" and figure out what settings need to be applied to the new version of marlin that you flashed.

Good luck,

Michael

Fulg, 2016-12-08:
You have `INVERT_X_DIR` to `false` and `INVERT_Y_DIR` to `true`, is that on purpose? That can cause you to lose your mind when trying to debug CoreXY movements. Usually you want both of these to be the same, it's hard enough to debug CoreXY without having an implicit flip of one axis over the other. :)

closed 2017-10-31 by thinkyhead