MarlinFirmware/Marlin #5427: Algorithm for linear fit without iteration or linear algebra

edwilliams16, 2016-12-08:
[fit_z.pdf](https://github.com/MarlinFirmware/Marlin/files/638420/fit_z.pdf)

@Roxy-3D 
   As far as I know, everybody's build plate is nominally a plane of constant z (with error - tilts, bumps, hollows etc).  If this is so, the above algorithm should be both fast and space efficient.
No iteration or linear algebra package required.

Roxy-3D, 2016-12-08:
I only skimmed it...   But I'm going to enjoy reading this!    I was mostly skimming it to see if I saw any dx's an dy's.   (I wasn't disappointed!  I got partial derivatives of ∂E, ∂a,  ∂b,  and ∂d instead.)   I've been playing darts (which means I win a couple of beers!!!)   So I'll read it in the morning!

But I'm not sure all terms necessarily go to 0.000 at the same time.  This kind of 'feels' like an assumption.  I'm going to have to start over in the morning:

I'm referring to:  ∂E/∂a = ∂E/∂b = ∂E/∂d = 0 (3)

Specifically...  I'm thinking there might be some 'Least Squares Solutions' where not every term is 0.0000    Can you prove everything goes to 0.0000 at the same time?

edwilliams16, 2016-12-08:
A function of one variable f(x) has maxima or minima wherever its derivative (slope) vanishes i.e. df/dx=0
 A function of three variables f(a,b,d) has maxima or minima when its derivative (slope) in each direction vanishes ie  df/da=0 & df/db=0 & df/dd=0  .  Technically they are partial derivatives because they only measure slope in one direction.  You get three equations for the three unknowns, instead of one equation for one unknown - just what you need.

Roxy-3D, 2016-12-08:
ok...  but can ∂E/∂a=0 happen at a set of points?   Can it be a flat spot on the top of the rest of the surface instead of being a single point?   Because if it can be, there isn't a unique solution at that locality.

Roxy-3D, 2016-12-08:
>  You get three equations for the three unknowns, instead of one equation for one unknown - just what you need.

OK...  That will probably help me in the morning when I read this again!

edwilliams16, 2016-12-08:
The good news is the three equations have a unique solution, which I derive.

edwilliams16, 2016-12-08:
@Roxy-3D 
[fit_z.pdf](https://github.com/MarlinFirmware/Marlin/files/640563/fit_z.pdf)

 I updated this to include a write-up of the conventional approach, where the fitting plane is completely arbitrary, in my notation, which you can compare to your own method.

Roxy-3D, 2016-12-08:
> I updated this to include a write-up of the conventional approach, where the fitting plane is completely arbitrary, in my notation, which you can compare to your own method.

@edwilliams16  Please correct me if I'm wrong, but Part 2 doesn't matter if we have a 'pretty reasonable bed plane', right?

I think I understand your approach.    It would help me to see it in C.   

How do you want to proceed?   Can you code your algorithm in C ?    I'm not opposed to trying to code it in C and running some test data through it.  The problem with me doing it is it will be at least a week and a half until I can do anything with it.   (I'm going to be travelling, and I won't have access to my printers to try things.)

edwilliams16, 2016-12-08:
@Roxy-3D  
 Yes.  Part 1 is the way to go.  I could code it easily enough, but I would have to put together my own tests, which I suspect you already have in place. Plus, you'd likely have to recode it to fit in to your infrastructure.   If you code it up in the manner in which you need to interface it, I can easily check to see you've got the algorithm implemented correctly.
  No hurry.  I just think we could improve on your 15 sec benchmark I saw somewhere.

Roxy-3D, 2016-12-08:
What I have for data sets are a few different sessions where I probed my bed and saved all the information.   

I did most of the work in Microsoft's Visual Studio.   I just stuffed the coordinates into an array and called the function and then checked the results against what QR_Solve said was the 'right' answer.   It might be useful to see if I can dig that code up because then I would have some reasonable data sets along with code to check the answer.

edwilliams16, 2016-12-09:
I'm much more fluent in yorick (a C-like interpreter with vector syntax), than C itself, as I use it for all algorithm development.  I've commented where I think you might need a little interpretation.

**********
// method 1
// x y z are arrays of the coords of the sample points, all length N
@Roxy-3D 
func fitz(x,y,z){
  if (numberof(x) < 3) {
    write, "error: Less than 3 sample points";
  }
  xbar = x(avg); // computes average of the N x values.
  // ie  xbar=0; for (i=0;i<N;i++) xbar +=x[i]; xbar /=N 
  ybar = y(avg);
  zbar = z(avg);
  x -= xbar;
  y -= ybar;
  z -= zbar;
  x2bar = (x*x)(avg); // computes average of x*x
  y2bar = (y*y)(avg);
  xybar = (x*y)(avg);
  xzbar = (x*z)(avg);
  yzbar = (y*z)(avg);
  D = x2bar*y2bar - xybar*xybar;
  if (abs(D) <= 1e-15*(max(abs(x))+max(abs(y)))) {
    write, "error: x,y points are collinear";
    return;
  }
  a = (xzbar*y2bar - yzbar*xybar)/D;
  b = (yzbar*x2bar - xzbar*xybar)/D;
  d = zbar - a*xbar - b*ybar;
  E = sum((a*x+b*y-z)^2);  // diagnostic 
  rmsE = sqrt(E/numberof(x));
  return [a,b,d,rmsE];
}

// for comparison only - method 2
func fitxyz(x,y,z){
 if (numberof(x) < 3) {
    write, "error: Less than 3 sample points";
  }
  xbar = x(avg);
  ybar = y(avg);
  zbar = z(avg);
  x -= xbar;
  y -= ybar;
  z -= zbar;
  x2bar = (x*x)(avg);
  y2bar = (y*y)(avg);
  z2bar = (z*z)(avg);
  xybar = (x*y)(avg);
  xzbar = (x*z)(avg);
  yzbar = (y*z)(avg);
  M = [[x2bar, xybar, xzbar], [xybar, y2bar, yzbar], [xzbar, yzbar, z2bar]];
  // uses Singular Value decomposition to find the eigenvector corresponding to the lowest eigenvalue
  s = SVdec(M,u, vt);
  a = u(1,0);
  b = u(2,0);
  c = u(3,0);
  // eignenvector can have either sign - we want c < 0 to compare with fitz
  if (c >0) {
    a = -a;
    b = -b;
    c = -c;
  }
    rmsE = sqrt(s(0));
  d = a*xbar + b*ybar + c*zbar;
  return [a,b,-c,-d,rmsE];
 }

//  perform some tests
write, "test1 - flat plane at z=0.1\n";
x=[0.0,0.0,1.0];
y=[0.0,1.0,0.0];
z=[0.1,0.1,0.1];
write, "Expect [0,0,0.1,0]\n";
fitz(x,y,z);

write, "test2 - tilted flat plane";
x=[0.0,0.0,1.0,1.0];
y=[0.0,1.0,0.0,1.0];
z=[0.05,0.05,0.15,0.15];
write, "Expect [0.1,0,0.05,0]\n";
fitz(x,y,z);

write, "test3 - collinear in xy";
x=[0.0,1.0,2.0,3.333];
y=[0.0,0.0,0.0,0.0];
z=[-0.1,0.0,0.1,0.0];
write,"expect error";
fitz(x,y,z);

write, "test4 - 2 sample points";
x=[0.0,1.0];
y=[0.0,0.0];
z=[0.0,0.1];
write, "Expect error";
fitz(x,y,z);

write, "test5 - sinusoidal in x-direction, uniform in y";
x= [0.0, 1.0, 2.0, 3.0, 4.0, 0.0, 1.0, 2.0, 3.0, 4.0];
y= [0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 1.0, 1.0, 1.0, 1.0];
z = 0.1*sin(x*2*pi/4);
fitz(x,y,z);
write, "from fitxyz";
fitxyz(x,y,z)([1,2,4,5]);

write, "test6 - 5 by 5 xy grid random z";
x = [0.0, 1.0, 2.0, 3.0, 4.0, 0.0, 1.0, 2.0, 3.0, 4.0, 
     0.0, 1.0, 2.0, 3.0, 4.0, 0.0, 1.0, 2.0, 3.0, 4.0,
     0.0, 1.0, 2.0, 3.0, 4.0];
y = [0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 1.0, 1.0, 1.0, 1.0,
     2.0, 2.0, 2.0, 2.0, 2.0, 3.0, 3.0, 3.0, 3.0, 3.0,
     4.0, 4.0, 4.0, 4.0, 4.0];
	     z = 0.2*(random(25) - 0.5);  // uniform in +-0.1
fitz(x,y,z);
write, "from fitxyz";
fitxyz(x,y,z)([1,2,4,5]);

*********
from running the above

yorick -i fitz.i 
 Copyright (c) 2005.  The Regents of the University of California.
 All rights reserved.  Yorick 2.2.04x ready.  For help type 'help'
 test1 - flat plane at z=0.1

 Expect [0,0,0.1,0]

[-4.62223e-33,-4.62223e-33,0.1,1.38778e-17]
 test2 - tilted flat plane
 Expect [0.1,0,0.05,0]

[0.1,0,0.05,6.93889e-18]
 test3 - collinear in xy
 expect error
 error: x,y points are collinear
[]
 test4 - 2 sample points
 Expect error
 error: Less than 3 sample points
 error: x,y points are collinear
[]
 test5 - sinusoidal in x-direction, uniform in y
[-0.02,1.14239e-18,0.04,0.0565685]
 from fitxyz
[-0.020028,-0,0.040056,0.0565572]
 test6 - 5 by 5 xy grid random z
[-0.0104978,0.0146353,-0.0111636,0.0470887]
 from fitxyz
[-0.0105078,0.0146491,-0.011171,0.047081]
> 

**********
Tests passed i.e. method 1 produces very similar results to method 2
eg for the random errors on the 5x5 grid (test 6)
[a,b,d,rmserror]
= [-0.0104978,0.0146353,-0.0111636,0.0470887]  from fitz (method 1)
= [-0.0105078,0.0146491,-0.011171,0.047081] from fitxyz (method 2)

 I think you'll find it easy to port the fitz function into C for your use.

edwilliams16, 2016-12-09:
@Roxy-3D 
I'm much more fluent in yorick (a C-like interpreter with vector syntax), than C itself, as I use it for all algorithm development. This way I can be (somewhat) guaranteed of working code. :-)  I've commented where I think you might need a little interpretation.

```// method 1
// x y z are arrays of the coords of the sample points, all length N
func fitz(x,y,z){
  if (numberof(x) < 3) {
    write, "error: Less than 3 sample points";
  }
  xbar = x(avg); // computes average of x values.
  // ie  xbar=0;for (i=0;i<N;i++) xbar +=x[i];xbar /=N 
  ybar = y(avg);
  zbar = z(avg);
  x -= xbar;
  y -= ybar;
  z -= zbar;
  x2bar = (x*x)(avg); // computes average of x*x
  y2bar = (y*y)(avg);
  xybar = (x*y)(avg);
  xzbar = (x*z)(avg);
  yzbar = (y*z)(avg);
  D = x2bar*y2bar - xybar*xybar;
  if (abs(D) <= 1e-15*(max(abs(x))+max(abs(y)))) {
    write, "error: x,y points are collinear";
    return;
  }
  a = (xzbar*y2bar - yzbar*xybar)/D;
  b = (yzbar*x2bar - xzbar*xybar)/D;
  d = zbar - a*xbar - b*ybar;
  E = sum((a*x+b*y-z)^2);  // diagnostic 
  rmsE = sqrt(E/numberof(x));
  return [a,b,d,rmsE];
}

// for comparison only - method 2
func fitxyz(x,y,z){
 if (numberof(x) < 3) {
    write, "error: Less than 3 sample points";
  }
  xbar = x(avg);
  ybar = y(avg);
  zbar = z(avg);
  x -= xbar;
  y -= ybar;
  z -= zbar;
  x2bar = (x*x)(avg);
  y2bar = (y*y)(avg);
  z2bar = (z*z)(avg);
  xybar = (x*y)(avg);
  xzbar = (x*z)(avg);
  yzbar = (y*z)(avg);
  M = [[x2bar, xybar, xzbar], [xybar, y2bar, yzbar], [xzbar, yzbar, z2bar]];
  // uses Singular Value decomposition to find the eigenvector corresponding to the lowest eigenvalue
  s = SVdec(M,u, vt);
  a = u(1,0);
  b = u(2,0);
  c = u(3,0);
  // eignenvector can have either sign - we want c < 0 to compare with fitz
  if (c >0) {
    a = -a;
    b = -b;
    c = -c;
  }
    rmsE = sqrt(s(0));
  d = a*xbar + b*ybar + c*zbar;
  return [a,b,-c,-d,rmsE];
 }

write, "test1 - flat plane at z=0.1\n";
x=[0.0,0.0,1.0];
y=[0.0,1.0,0.0];
z=[0.1,0.1,0.1];
write, "Expect [0,0,0.1,0]\n";
fitz(x,y,z);

write, "test2 - tilted flat plane";
x=[0.0,0.0,1.0,1.0];
y=[0.0,1.0,0.0,1.0];
z=[0.05,0.05,0.15,0.15];
write, "Expect [0.1,0,0.05,0]\n";
fitz(x,y,z);

write, "test3 - collinear in xy";
x=[0.0,1.0,2.0,3.333];
y=[0.0,0.0,0.0,0.0];
z=[-0.1,0.0,0.1,0.0];
write,"expect error";
fitz(x,y,z);

write, "test4 - 2 sample points";
x=[0.0,1.0];
y=[0.0,0.0];
z=[0.0,0.1];
write, "Expect error";
fitz(x,y,z);

write, "test5 - sinusoidal in x-direction, uniform in y";
x= [0.0, 1.0, 2.0, 3.0, 4.0, 0.0, 1.0, 2.0, 3.0, 4.0];
y= [0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 1.0, 1.0, 1.0, 1.0];
z = 0.1*sin(x*2*pi/4);
fitz(x,y,z);
write, "from fitxyz";
fitxyz(x,y,z)([1,2,4,5]);

write, "test6 - 5 by 5 xy grid random z";
x = [0.0, 1.0, 2.0, 3.0, 4.0, 0.0, 1.0, 2.0, 3.0, 4.0, 
     0.0, 1.0, 2.0, 3.0, 4.0, 0.0, 1.0, 2.0, 3.0, 4.0,
     0.0, 1.0, 2.0, 3.0, 4.0];
y = [0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 1.0, 1.0, 1.0, 1.0,
     2.0, 2.0, 2.0, 2.0, 2.0, 3.0, 3.0, 3.0, 3.0, 3.0,
     4.0, 4.0, 4.0, 4.0, 4.0];
	     z = 0.2*(random(25) - 0.5);  // uniform in +-0.1
fitz(x,y,z);
write, "from fitxyz";
fitxyz(x,y,z)([1,2,4,5]);

```
*********
from running the above

```
yorick -i fitz.i 
 Copyright (c) 2005.  The Regents of the University of California.
 All rights reserved.  Yorick 2.2.04x ready.  For help type 'help'
 test1 - flat plane at z=0.1

 Expect [0,0,0.1,0]

[-4.62223e-33,-4.62223e-33,0.1,1.38778e-17]
 test2 - tilted flat plane
 Expect [0.1,0,0.05,0]

[0.1,0,0.05,6.93889e-18]
 test3 - collinear in xy
 expect error
 error: x,y points are collinear
[]
 test4 - 2 sample points
 Expect error
 error: Less than 3 sample points
 error: x,y points are collinear
[]
 test5 - sinusoidal in x-direction, uniform in y
[-0.02,1.14239e-18,0.04,0.0565685]
 from fitxyz
[-0.020028,-0,0.040056,0.0565572]
 test6 - 5 by 5 xy grid random z
[-0.0104978,0.0146353,-0.0111636,0.0470887]
 from fitxyz
[-0.0105078,0.0146491,-0.011171,0.047081]
> 
```

Tests passed i.e. method 1 produces very similar results to method 2
eg for the random errors on the 5x5 grid (test 6)
[a,b,d,rmserror]
= [-0.0104978,0.0146353,-0.0111636,0.0470887]  from fitz (method 1)
= [-0.0105078,0.0146491,-0.011171,0.047081] from fitxyz (method 2)

 I think you'll find it easy to port the fitz function into C for your use.

edwilliams16, 2016-12-09:
@Roxy-3D 
[fit_z.pdf](https://github.com/MarlinFirmware/Marlin/files/641255/fit_z.pdf)
Some corrections of the diagnostics.
 
 yorick is a MatLab that is built for speed.   GUIs and objects are afterthoughts. It's also free.  
[yorick](http://dhmunro.github.io/yorick-doc/)  It has a very elegant syntax for manipulating multi-dimensional arrays and great scientific graphics.  I've been a user for 20 years. 

  You only need the fitz function.  The rest is just for testing.  I think you'll find it easy enough.  Maybe you'll become a yorick user...

thinkyhead, 2016-12-09:
> Apparently three backquotes for a code block.

Three backquotes followed by `cpp` ftw!

Roxy-3D, 2016-12-12:
I am finally settled in...   If I don't get distracted, I might get this converted tonight.

edwilliams16, 2016-12-12:
@Roxy-3D   
Great!  

   It looks like  the N < 3 test is redundant as those inputs will fail the collinear test later.  It also may be better to require the sampling points to be non-collinear when they are created, rather than waiting until the best fit plane is created in this routine.  A much looser test would be better then.  The points should not even be close to collinear or fitting in the transverse direction will be horrible.

edwilliams16, 2016-12-16:
@Roxy-3D    Any snags?  Or not at the top of the stack yet?

Roxy-3D, 2016-12-16:
I've been busy skiing and that has left me so tired I never got to work on it.  But the weather is bad on the mountain today and I gave up on skiing.  I am doing the C version of your code right now...

I have an idea for you to think about.    Is it possible to easily calculate where the max difference from the calculated plane is?   I can loop through all the numbers.   But it would be interesting if your algorithm could just report where the worst case fit is.

edwilliams16, 2016-12-16:
There's no way to get the worst fit point without looping through them all, I'm afraid.  The rms error of course gives you a sort of average deviation, where the worst points are weighted more strongly, because the errors are squared.
  Should the rms error be more than (half) the acceptable variation of the nozzle height from the build plane, you can be sure you are out of luck - but you can't be sure of the converse without checking each point. 
  
   If the bumps and hollows of your build plate were genuinely random (not likely), you could assume they were Gaussian distributed and give a probability that any single deviation exceeded your tolerance.  Ignoring that caveat, you would say that the probability that a given fit point lay within 
one, two or three rms error distances are 68%, 95% and 99.7% respectively.

 Have a great ski - I'm off to Jackson Hole for three weeks next month!

Roxy-3D, 2016-12-16:
I've got the code converted, but I have a bug.   I'm not getting the same results as you.   I think I'm misinterpreting what the yorick syntax is implying somewhere.   I've added some debug printing of the various variables as it progresses through the algorithm.   Hopefully, you can tell me what part of yorick I'm translating incorrectly.

For your first data set:
```cpp
double xxx[] = { 0.0, 0.0, 1.0 };	// Expect [0,0,0.1,0]
double yyy[] = { 0.0, 1.0, 0.0 };
double zzz[] = { 0.1, 0.1, 0.1 };
```
I'm gettting:

> Hello World.    size of triples: 3
> xbar=0.333333 ybar=0.333333 zbar=0.100000
> linear_fit_compensate_matrix( x ) = {-0.333333 -0.666667 -1.000000 }
> linear_fit_compensate_matrix( y ) = {-0.233333 -0.566667 -0.900000 }
> linear_fit_compensate_matrix( z ) = {-0.100000 -0.200000 -0.300000 }
> x2bar=0.333333 y2bar=0.666667
> xybar=0.000000 xzbar=0.033333 yzbar=0.333333
> Results: a=0.100000  b=0.500000   d=-0.100000
> 

Attached are the .cpp and .h file in their current state.

[Linear_Fit.h.txt](https://github.com/MarlinFirmware/Marlin/files/658493/Linear_Fit.h.txt)
[Linear-Fit.cpp.txt](https://github.com/MarlinFirmware/Marlin/files/658494/Linear-Fit.cpp.txt)

edwilliams16, 2016-12-17:
@Roxy-3D 
   Did your dataset get truncated above on the cut and paste?  I'm only seeing 15 points.

Roxy-3D, 2016-12-17:
> Did your dataset get truncated above on the cut and paste? I'm only seeing 15 points.
> For those I get:

No...   The whole set is there.  There should be a horizontal scroll bar you can move to see the whole thing.  But even if you don't do that...  You should be able to drag across the whole set and cut and paste it.

But here it is without the C++ formatting:

double xxx[] = { 0,50,100,150,200,           20,70,120,165,195,         0,50,100,150,200,           0,55,100,150,200,           0,65,100,150,205 };
double yyy[] = { 0, 1,  2,  3, 4,            50, 51,  52,  53, 54,     100, 101,102,103,104,        150,151,152,153,154,        200,201,202,203,204 };
double zzz[] = { 0.01,.002,-.01,-.02,0,      0.01,.002,-.01,-.02,0,     0.01,.002,-.01,-.02,0,      0.01,.002,-.01,-.02,0,      0.01,.002,-.01,-.012,0.01};

edwilliams16, 2016-12-17:
@Roxy-3D   
Silly me.  For some reason I didn't try two finger swiping horizontally.
  I get 
a=-7.519e-05,b=1.36513e-05,d=0.00353229
  in agreement with you.
  How does it compare with your old algorithm?

Roxy-3D, 2016-12-17:
> How does it compare with your old algorithm?

Unknown!  But in the next day or two...  I'll be able to give you an answer.
The cool thing is this code is about the same size as the iterative LSF code I did.   But it is almost instantaneous.

edwilliams16, 2016-12-17:
OK!   I was pretty sure I could improve on 15 seconds.  We can maybe squeeze the footprint a little:

```
// ConsoleApplication1.cpp : main project file.

#include "stdafx.h"
#include <stdio.h>
#include <math.h>
#include "Linear_Fit.h"

using namespace System;
struct linear_fit_results *results;

int main(array<System::String ^> ^args)
{
	int three_size = sizeof(xxx) / sizeof(double);
    printf("Hello World.    size of triples: %d \n", three_size );

	results = linear_fit(xxx, yyy, zzz, three_size);
	
	printf("Results: a=%f  b=%f   d=%f \n", results->A, results->B, results->D);
	getchar();
    return 0;
}

struct linear_fit_results linear_fit_results;

struct linear_fit_results *linear_fit(double *x, double *y, double *z, int n) {
	double xbar, ybar, zbar;
	double x2bar, y2bar;
	double xybar, xzbar, yzbar;
	double D;
	int i;

	linear_fit_results.A = 0.0;
	linear_fit_results.B = 0.0;
	linear_fit_results.D = 0.0;

	xbar = linear_fit_average(x, n);
	ybar = linear_fit_average(y, n);
	zbar = linear_fit_average(z, n);

	linear_fit_subtract_mean( 'x', x, xbar, n);
	linear_fit_subtract_mean( 'y', y, ybar, n);
	linear_fit_subtract_mean( 'z', z, zbar, n);

	x2bar = linear_fit_average_product( x, x, n);
	y2bar = linear_fit_average_product( y, y, n);
	xybar =	linear_fit_average_product( x, y, n);
	xzbar =	linear_fit_average_product( x, z, n);
	yzbar =	linear_fit_average_product( y, z, n);

	D = x2bar*y2bar - xybar*xybar;
	for(i=0; i<n; i++) {
		if (fabs(D) <= 1e-15*( linear_fit_max_abs(x, n) + linear_fit_max_abs(y, n))) {
			printf( "error: x,y points are collinear at index:%d \n", i );
			return NULL;
		}
	}

	linear_fit_results.A = (xzbar*y2bar - yzbar*xybar) / D;
	linear_fit_results.B = (yzbar*x2bar - xzbar*xybar) / D;
	linear_fit_results.D = zbar - linear_fit_results.A*xbar - linear_fit_results.B*ybar;

	return &linear_fit_results;
}

double linear_fit_average(double *matrix, int n)
{
	int i;
	double sum=0.0;

	for (i = 0; i < n; i++)
		sum += matrix[i];
	return sum / (double) n;
}

double linear_fit_average_product(double *matrix1, double *matrix2, int n) {
        int i;
	double sum = 0.0;

	for (i = 0; i < n; i++)
		sum += matrix1[i] * matrix2[i];
	return sum / (double) n;
}  

void linear_fit_subtract_mean(char which, double *matrix, double bar, int n) {
	int i;

	for (i = 0; i < n; i++) {
		matrix[i] -= bar;
	}
	return;
}

double linear_fit_max_abs(double *matrix, int n) {
	int i;
	double max_abs = 0.0;

	for(i=0; i<n; i++)
		if ( max_abs < fabs(matrix[i]))
			max_abs = fabs(matrix[i]);
	return max_abs;
}

```

Roxy-3D, 2016-12-17:
The change in function names helps a lot to describe what is happening!!!!

edwilliams16, 2016-12-17:
Maybe also get rid of the unused "char which".  It's clear from the remaining calling arguments which mean is being subtracted.

Roxy-3D, 2016-12-18:
The "char which" was just a debug info thing.   Yes, it should be stripped out of the code.

I don't think the initialization of linear_results.A,B,D = 0.0 is redundant.   The reason is the function can terminate early in the case where things are co-linear.   So in that case...  It might be good to have some sane values.   It might be argued we should initialize them to NAN so if that happens, they can't be used for anything.

edwilliams16, 2016-12-18:
Right now the function returns a null pointer when the points are collinear, not the initialized struct.  I'm not sure you want sane values - it's not clear what they might be. The  calling code should error out and warn the user that s/he is not defining a suitable mesh.

Roxy-3D, 2016-12-18:
Agreed.  And setting those values to 0.0 was an initial attempt at keeping things clean.  But I'm leaning towards setting the coefficients to NAN.  Anyway...  The important thing is to make sure it produces the correct results in all the test cases.   After I'm done re-basing UBL to RC-8 I'm going to put the Grid Leveling feature into it and of course this code will be used to calculate the coefficients.

(And that code will definitely be checking for a failed conversion!)

edwilliams16, 2016-12-18:
OK.  Let me (edATedwilliams.org) know if some unanticipated behavior occurs - but I'm pretty confident we have it working correctly.

Roxy-3D, 2016-12-19:
So...  I really thought I would get the Iterative LSF algorithm verifying the results.   But I didn't get it done yet.   When ever a software person give you a schedule, you should double the number and increment the unit.   If they say it will be done in 2 days, the most likely answer is 4 weeks....

edwilliams16, 2016-12-19:
Maybe we need to threaten a pay cut.  ;-)

JohnOCFII, 2016-12-19:
> Maybe we need to threaten a pay cut. ;-)

Funny running into you here, Ed.  I guessed it was you when I saw your post.  (John O'Shaughnessy -- AVSIG and SMXgig fellow pilot -- AVSIG is still kicking.  You should drop by some time.)

edwilliams16, 2016-12-19:
Hi John!  Sold my C182 when I retired to Maui. Inactive CFII...  Avgas $7/gallon here.
 Programming Arduinos is different - no petaflops of CPU or terabytes of memory available.

Roxy-3D, 2016-12-22:
I had to invert the coefficients to match what qr_solve() (and the iterative version) return.   But it seems to be in agreement with the iterative solution.  I'm not sure what to do about the 'D' term.   I think that should be 0.000 but I'm not sure about that:
```cpp
Hello World.    size of triples: 25
Results from iterative linear fit:
error=0.002864000000000  delta_size=0.100000000000000  a=0.000000000000000  b=0.000000000000000 d=0.000000000000000  cnt=1
error=0.002864000000000  delta_size=0.050000000000000  a=0.000000000000000  b=0.000000000000000 d=0.000000000000000  cnt=2
error=0.002864000000000  delta_size=0.025000000000000  a=0.000000000000000  b=0.000000000000000 d=0.000000000000000  cnt=3
error=0.002864000000000  delta_size=0.012500000000000  a=0.000000000000000  b=0.000000000000000 d=0.000000000000000  cnt=4
error=0.002864000000000  delta_size=0.006250000000000  a=0.000000000000000  b=0.000000000000000 d=0.000000000000000  cnt=5
error=0.002658140625000  delta_size=0.003125000000000  a=0.000000000000000  b=0.000000000000000 d=0.003125000000000  cnt=7
error=0.002658140625000  delta_size=0.001562500000000  a=0.000000000000000  b=0.000000000000000 d=0.003125000000000  cnt=8
error=0.002658140625000  delta_size=0.000781250000000  a=0.000000000000000  b=0.000000000000000 d=0.003125000000000  cnt=9
error=0.002657170166016  delta_size=0.000390625000000  a=0.000000000000000  b=0.000000000000000 d=0.002734375000000  cnt=11
error=0.002656701721191  delta_size=0.000195312500000  a=0.000000000000000  b=0.000000000000000 d=0.002929687500000  cnt=13
error=0.002656697525024  delta_size=0.000097656250000  a=0.000000000000000  b=0.000000000000000 d=0.002832031250000  cnt=15
error=0.002656640018463  delta_size=0.000048828125000  a=0.000000000000000  b=0.000000000000000 d=0.002880859375000  cnt=17
error=0.002198349431046  delta_size=0.000024414062500  a=0.000048828125000  b=-0.000048828125000 d=0.002783203125000  cnt=20
error=0.001996974973632  delta_size=0.000012207031250  a=0.000061035156250  b=-0.000024414062500 d=-0.000964355468750  cnt=24
error=0.001968686216417  delta_size=0.000006103515625  a=0.000067138671875  b=-0.000018310546875 d=-0.002221679687500  cnt=27
error=0.001961303546590  delta_size=0.000003051757813  a=0.000073242187500  b=-0.000018310546875 d=-0.002853393554687  cnt=30
error=0.001958897909767  delta_size=0.000001525878906  a=0.000073242187500  b=-0.000015258789063 d=-0.003166198730469  cnt=33
error=0.001958372446245  delta_size=0.000000762939453  a=0.000074005126953  b=-0.000014495849609 d=-0.003323364257813  cnt=36
error=0.001958161217889  delta_size=0.000000381469727  a=0.000074768066406  b=-0.000014114379883 d=-0.003441238403320  cnt=40
error=0.001958128431955  delta_size=0.000000190734863  a=0.000074958801270  b=-0.000013923645020 d=-0.003480529785156  cnt=43
error=0.001958115672555  delta_size=0.000000095367432  a=0.000075054168701  b=-0.000013732910156 d=-0.003509902954102  cnt=47
error=0.001958112945283  delta_size=0.000000047683716  a=0.000075149536133  b=-0.000013685226440 d=-0.003524637222290  cnt=51
error=0.001958112780096  delta_size=0.000000023841858  a=0.000075173377991  b=-0.000013685226440 d=-0.003527092933655  cnt=53
error=0.001958112648182  delta_size=0.000000011920929  a=0.000075173377991  b=-0.000013661384583 d=-0.003529536724091  cnt=56
error=0.001958112606767  delta_size=0.000000005960464  a=0.000075185298920  b=-0.000013655424118 d=-0.003531378507614  cnt=60
error=0.001958112606766  delta_size=0.000000002980232  a=0.000075185298920  b=-0.000013655424118 d=-0.003531381487846  cnt=62
error=0.001958112602482  delta_size=0.000000001490116  a=0.000075188279152  b=-0.000013652443886 d=-0.003531993925571  cnt=67
error=0.001958112602235  delta_size=0.000000000745058  a=0.000075189024210  b=-0.000013652443886 d=-0.003532071411610  cnt=69
error=0.001958112601984  delta_size=0.000000000372529  a=0.000075189769268  b=-0.000013651698828 d=-0.003532224893570  cnt=74
error=0.001958112601972  delta_size=0.000000000186265  a=0.000075189769268  b=-0.000013651512563 d=-0.003532243706286  cnt=76
error=0.001958112601962  delta_size=0.000000000093132  a=0.000075189955533  b=-0.000013651419431 d=-0.003532272577286  cnt=80

Results from high speed linear fit:
                                                       a=0.000075190022350  b=-0.000013651340876 d=0.009292287550607

```

edwilliams16, 2016-12-22:
@Roxy-3D 
  My LSQ fit plane is defined as   z = ax + by + d, whereas I think yours is ax+by+z+d=0. So to match your convention we have to reverse the signs of a, b and d of my routine on output.  However, you don't quite do that.  

  linear_fit_results.D = -(zbar + linear_fit_results.A*xbar + linear_fit_results.B*ybar);

I think is what you really want.

Roxy-3D, 2016-12-22:
That makes sense...      With that change, the second case returns:

```cpp
Hello World.    size of triples: 4
Results from iterative linear fit:
error=0.009803921568627  delta_size=0.100000000000000  a=-0.100000000000000  b=-0.100000000000000 d=0.000000000000000  cnt=2
error=0.004878048780488  delta_size=0.050000000000000  a=-0.150000000000000  b=-0.050000000000000 d=0.000000000000000  cnt=5
error=0.003683241252302  delta_size=0.025000000000000  a=-0.125000000000000  b=-0.050000000000000 d=0.000000000000000  cnt=7
error=0.000925212027756  delta_size=0.012500000000000  a=-0.112500000000000  b=-0.025000000000000 d=-0.025000000000000  cnt=11
error=0.000231722859460  delta_size=0.006250000000000  a=-0.106250000000000  b=-0.012500000000000 d=-0.037500000000000  cnt=15
error=0.000057974935503  delta_size=0.003125000000000  a=-0.103125000000000  b=-0.006250000000000 d=-0.043750000000000  cnt=19
error=0.000014498742234  delta_size=0.001562500000000  a=-0.101562500000000  b=-0.003125000000000 d=-0.046875000000000  cnt=23
error=0.000003625279071  delta_size=0.000781250000000  a=-0.100781250000000  b=-0.001562500000000 d=-0.048437500000000  cnt=27
error=0.000000906391921  delta_size=0.000390625000000  a=-0.100390625000000  b=-0.000781250000000 d=-0.049218750000000  cnt=31
error=0.000000226606872  delta_size=0.000195312500000  a=-0.100195312500000  b=-0.000390625000000 d=-0.049609375000000  cnt=35
error=0.000000056652822  delta_size=0.000097656250000  a=-0.100097656250000  b=-0.000195312500000 d=-0.049804687500000  cnt=39
error=0.000000014163343  delta_size=0.000048828125000  a=-0.100048828125000  b=-0.000097656250000 d=-0.049902343750000  cnt=43
error=0.000000003540853  delta_size=0.000024414062500  a=-0.100024414062500  b=-0.000048828125000 d=-0.049951171875000  cnt=47
error=0.000000000885215  delta_size=0.000012207031250  a=-0.100012207031250  b=-0.000024414062500 d=-0.049975585937500  cnt=51
error=0.000000000221304  delta_size=0.000006103515625  a=-0.100006103515625  b=-0.000012207031250 d=-0.049987792968750  cnt=55
error=0.000000000055326  delta_size=0.000003051757813  a=-0.100003051757813  b=-0.000006103515625 d=-0.049993896484375  cnt=59
error=0.000000000013832  delta_size=0.000001525878906  a=-0.100001525878906  b=-0.000003051757813 d=-0.049996948242187  cnt=63
error=0.000000000001153  delta_size=0.000000762939453  a=-0.100000762939453  b=-0.000000000000000 d=-0.049999237060547  cnt=68
error=0.000000000000864  delta_size=0.000000381469727  a=-0.100000762939453  b=-0.000000381469727 d=-0.049999237060547  cnt=70
error=0.000000000000072  delta_size=0.000000190734863  a=-0.100000190734863  b=-0.000000000000000 d=-0.049999809265137  cnt=75
error=0.000000000000054  delta_size=0.000000095367432  a=-0.100000190734863  b=-0.000000095367432 d=-0.049999809265137  cnt=77
error=0.000000000000005  delta_size=0.000000047683716  a=-0.100000047683716  b=-0.000000000000000 d=-0.049999952316284  cnt=82
error=0.000000000000003  delta_size=0.000000023841858  a=-0.100000047683716  b=-0.000000023841858 d=-0.049999952316284  cnt=84
error=0.000000000000000  delta_size=0.000000011920929  a=-0.100000000000000  b=-0.000000011920929 d=-0.049999988079071  cnt=89
error=0.000000000000000  delta_size=0.000000005960464  a=-0.100000000000000  b=-0.000000005960464 d=-0.049999994039536  cnt=92
error=0.000000000000000  delta_size=0.000000002980232  a=-0.100000000000000  b=-0.000000002980232 d=-0.049999997019768  cnt=95
error=0.000000000000000  delta_size=0.000000001490116  a=-0.100000000000000  b=-0.000000001490116 d=-0.049999998509884  cnt=98
error=0.000000000000000  delta_size=0.000000000745058  a=-0.100000000000000  b=-0.000000000745058 d=-0.049999999254942  cnt=101
error=0.000000000000000  delta_size=0.000000000372529  a=-0.100000000000000  b=-0.000000000372529 d=-0.049999999627471  cnt=104
error=0.000000000000000  delta_size=0.000000000186265  a=-0.100000000000000  b=-0.000000000186265 d=-0.049999999813735  cnt=107
error=0.000000000000000  delta_size=0.000000000093132  a=-0.100000000000000  b=-0.000000000093132 d=-0.049999999906868  cnt=110

Results from high speed linear fit:
                                                       a=-0.100000000000000  b=-0.000000000000000 d=-0.050000000000000
```
And everything is in agreement.

edwilliams16, 2016-12-22:
OK.  Looks like we're set on that.
  What's your model for a non-linear fit - bilinear - bicubic spline?

Roxy-3D, 2016-12-22:
UBL does a bi-linear interpolation right now.   Would it be possible to do a bi-cubic spline without ridiculous amounts of floating point math?   If so...   I would change the current algorithm.

edwilliams16, 2016-12-22:
Yes, I'm not sure on how complicated it has to be.  Is the mesh to be fit always a uniform rectangular grid on x,y?  That makes things simpler.

Blue-Marlin, 2016-12-22:
Please have a look at `ABL_BILINEAR_SUBDIVISION`.
Here splines are used to subdivide the measured mesh date. This uses huge amounts of memory what would be better invested in __measuring__ more points.
Subdividing the moves into smaller bits, not only at the mesh crossings, and calculating the related z on the fly, could have a positive effect to the curve - if fast enough.

Roxy-3D, 2016-12-23:
> Yes, I'm not sure on how complicated it has to be. Is the mesh to be fit always a uniform rectangular grid on x,y? That makes things simpler.

Yes.   Most of the testing has been done on 10x10 grids.   But 15x15 is supported.   

> Here splines are used to subdivide the measured mesh date. This uses huge amounts of memory what would be better invested in measuring more points.

I was hoping rather than using more points, the bi-cubic splines would use the mesh points surrounding the current (or desired) location in the computation.   I wasn't thinking that adding more points as the way to go.  Rather...  With the points we do have, can we have a smoother (and more accurate) curve?

Roxy-3D, 2017-01-08:
FYI:   This isn't forgotten about.  But a number of things have jumped in front of it on   my "To Do" list.    I'm real bad at doing time estimates, but right now it will be 4 or 5 weeks before I can fold this code into the UBL code base.   

To Do List:
- Burn more vacation time in the mountains
- Get my new development machine running again (after frying the Arduino and RAMPS board)
- Start compiling code and loading it on 32-Bit boards
- ???

edwilliams16, 2017-01-09:
Whenever you are ready, Roxy.

Roxy-3D, 2017-03-24:
> Whenever you are ready, Roxy.

@edwilliams16 

UBL is merged into RCBugFix.   There is a lot of work to get it compliant to how other bed leveling systems have interfaced with Marlin.   But we are getting to the point where we can write the Grid base leveling of the mesh and do a least squares fit on the probed points...    Realistically...   It is still a month away until users are getting the benefit of your algorithm.   But it is close...

edwilliams16, 2017-03-24:
@Roxy-3D   
That's great! 
 I need to install a sensor first.  I have a glass bed, so I'm thinking BLTouch as the inductive and capacitive sensors don't work.  No real urgency as I'm not really having significant leveling issues.

Roxy-3D, 2017-03-24:
Two comments...   
- First...  BL-Touch is not inductive.   It has a probe pin that gets depressed and a HAL effect sensor triggers.
- If you want...  You can bring up UBL without a Z-Probe.   But that still would not enable you to do the Grid based leveling so you can play with and modify your algorithm.

edwilliams16, 2017-03-24:
@Roxy-3D 
     I said it poorly.  I'm thinking BL-Touch because  inductive/capacitive sensors that I can more simply install don't work.

Roxy-3D, 2017-04-08:
@edwilliams16    Hey Ed, quick question for you...  Can you take a quick read through your paper and see if you can find a way to do the calculations without the full data set?    Right now I need to have the X, Y and Z of each number available to do the calculations.    With 4 byte floats, a 5x5 grid ends up needing 5x5x4x3 = 300 bytes of memory.   That is about the limit of what we can assume will be there.

It would be nice if we could go up to 8x8 or maybe even a little bit higher.   But the only way we will be able to do that is if we can modify the algorithm so that it gets a result, factors in its contribution to the finished value, and then another point is acquired.

edwilliams16, 2017-04-09:
This maybe the same as you are suggesting:

  Combine the code block so that there is a single loop over the x,y,z values and read them in on the lines of:

```
struct linear_fit_results *linear_fit( int n) {
	double xbar, ybar, zbar;
	double x2bar, y2bar;
	double xybar, xzbar, yzbar;
        double max_absx, max_absy;
	double D;
	int i;

	linear_fit_results.A = 0.0;
	linear_fit_results.B = 0.0;
	linear_fit_results.D = 0.0;
        xbar = ybar = zbar = x2bar = y2bar = z2bar = 
          xybar = xzbar = yzbar = maxabs_x = max_absy = 0.0;

	for (i=0; i<n; i++){
   	  read_next(x,y,z);
	  xbar += x;
          ybar += y;
          zbar += z;
          x2bar += x*x;
          y2bar += y*y;
          z2bar += z*z;
          xybar += x*y;
          xzbar += x*z;
          yzbar += y*z;
          max_absx = (abs(x) > max_absx) ? abs(x) : max_absx;
          max_absy = (abs(y) > max_absy) ? abs(y) : max_absy;
        }
        xbar /= n;
        ybar /= n;
        zbar /= n;
        x2bar = x2bar/n - xbar*xbar;
        y2bar = y2bar/n - ybar*ybar;
        z2bar = z2bar/n - zbar*zbar;
        xybar = xybar/n - xbar*ybar;
        yzbar = yzbar/n - ybar*zbar;
        xzbar = xzbar/n - xbar*zbar;
        
         D = x2bar*y2bar - xybar*xybar;
	 if (fabs(D) <= 1e-15*( max_absx + max_absy )) {
			printf( "error: x,y points are collinear);
			return NULL;
	}
	

	linear_fit_results.A = (xzbar*y2bar - yzbar*xybar) / D;
	linear_fit_results.B = (yzbar*x2bar - xzbar*xybar) / D;
	linear_fit_results.D = zbar - linear_fit_results.A*xbar -linear_fit_results.B*ybar;

	return &linear_fit_results;
}
```

  Note the error condition shouldn't have been looped.
 read_next  is a function that fetches the next data point.  More code space but less memory as there are no arrays of length n.

Roxy-3D, 2017-04-09:
Wow!   It might be possible?   I'll look at it more carefully in the next day or two.   It would be great if we could just probe, get the data, and feed it into the calculations but not keep it around!!!!

edwilliams16, 2017-04-09:
That's the idea.  If you don't want to specify the number of points (n) in advance, you could count in the loop, but then you'd need a stop command of some kind to tell the routine you've finished sampling points and to break out of the loop. Probably unnecessarily complex.

Roxy-3D, 2017-04-09:
> That's the idea. If you don't want to specify the number of points (n) in advance, you could count in the loop, but then you'd need a stop command of some kind to tell the routine you've finished sampling points and to break out of the loop. Probably unnecessarily complex.

No...  Just like the old G29 P3 commands...  The algorithm will know how many probe points are going to be sampled.  We do know that in advance.

Roxy-3D, 2017-04-19:
@edwilliams16   @Bob-the-Kuhn 

I know its been quiet...   I've been trying to hunt down the data corruption issue that alters Marlin's Stack Pointer when too much memory is in use.

But I'm going to take a few days off and get the UBL Grid Based Leveling going...  And of course, I'll be using the algorithm published here: https://github.com/MarlinFirmware/Marlin/issues/5427#issuecomment-292757879 

To make sure I'm not aggravating the memory usage issue.   The other thing this algorithm enables is I don't need to keep the X's and Y's around either.  So that simplifies the Grid Based Probing logic Bob did.   I'm only making small changes to it at a time because I don't want to break the logic.   Soon, it won't be recognizable.  But it was very helpful as a starting platform!!!

Roxy-3D, 2017-04-20:
@edwilliams16     Hey Ed, I think I need you to jump in and figure something out...   I have the incremental linear fit code almost working.   But there are two problems.  The code is called from this snipet of code.   It resets all of the variables and the data structure.  It then feeds in each and every point.  (For now... it does it from a pre-initialized array.  but when the algorithm works, it will be probing a point and feeding those results into the algorithm.  And then after all of the points have been feed in, it completes the calculations and prints the results:
```cpp
	incremental_LSF_reset();
	printf("Reset of incremental                                   A=%3.15f  B=%3.15f D=%3.15f\n", inc_results.A, inc_results.B, inc_results.D);
	for (int i = 0; i < set_size; i++)
		incremental_LSF(xxx9[i], yyy9[i], zzz9[i]);
	finish_incremental_LSF(set_size);
	printf("Results from incremental linear fit:");
	printf("                   A=%3.15f  B=%3.15f D=%3.15f \n\n\n", inc_results.A, inc_results.B, inc_results.D);
```
The problem is, the A term is negated.   And the D term is 0.000   Can you take a look at the code below and see if you can see an obvious error?   (I can fix the A B term by flipping the subtraction, but I'm very hesitent to do that without you saying that is correct.  But I have no idea how to fix D)  Here is the output and the code:
```cpp

Doing xxx9:   size of triples: 3
Results from iterative linear fit:
error=0.002864000000000  delta_size=0.100000000000000  a=0.000000000000000  b=0.000000000000000 d=0.000000000000000  cnt=1
error=0.002864000000000  delta_size=0.050000000000000  a=0.000000000000000  b=0.000000000000000 d=0.000000000000000  cnt=2
error=0.002864000000000  delta_size=0.025000000000000  a=0.000000000000000  b=0.000000000000000 d=0.000000000000000  cnt=3
error=0.002864000000000  delta_size=0.012500000000000  a=0.000000000000000  b=0.000000000000000 d=0.000000000000000  cnt=4
error=0.002864000000000  delta_size=0.006250000000000  a=0.000000000000000  b=0.000000000000000 d=0.000000000000000  cnt=5
error=0.002658140625000  delta_size=0.003125000000000  a=0.000000000000000  b=0.000000000000000 d=0.003125000000000  cnt=7
error=0.002658140625000  delta_size=0.001562500000000  a=0.000000000000000  b=0.000000000000000 d=0.003125000000000  cnt=8
error=0.002658140625000  delta_size=0.000781250000000  a=0.000000000000000  b=0.000000000000000 d=0.003125000000000  cnt=9
error=0.002657170166016  delta_size=0.000390625000000  a=0.000000000000000  b=0.000000000000000 d=0.002734375000000  cnt=11
error=0.002656701721191  delta_size=0.000195312500000  a=0.000000000000000  b=0.000000000000000 d=0.002929687500000  cnt=13
error=0.002656697525024  delta_size=0.000097656250000  a=0.000000000000000  b=0.000000000000000 d=0.002832031250000  cnt=15
error=0.002656640018463  delta_size=0.000048828125000  a=0.000000000000000  b=0.000000000000000 d=0.002880859375000  cnt=17
error=0.002198349431046  delta_size=0.000024414062500  a=0.000048828125000  b=-0.000048828125000 d=0.002783203125000  cnt=20
error=0.001996974973632  delta_size=0.000012207031250  a=0.000061035156250  b=-0.000024414062500 d=-0.000964355468750  cnt=24
error=0.001968686216417  delta_size=0.000006103515625  a=0.000067138671875  b=-0.000018310546875 d=-0.002221679687500  cnt=27
error=0.001961303546590  delta_size=0.000003051757813  a=0.000073242187500  b=-0.000018310546875 d=-0.002853393554687  cnt=30
error=0.001958897909767  delta_size=0.000001525878906  a=0.000073242187500  b=-0.000015258789063 d=-0.003166198730469  cnt=33
error=0.001958372446245  delta_size=0.000000762939453  a=0.000074005126953  b=-0.000014495849609 d=-0.003323364257813  cnt=36
error=0.001958161217889  delta_size=0.000000381469727  a=0.000074768066406  b=-0.000014114379883 d=-0.003441238403320  cnt=40
error=0.001958128431955  delta_size=0.000000190734863  a=0.000074958801270  b=-0.000013923645020 d=-0.003480529785156  cnt=43
error=0.001958115672555  delta_size=0.000000095367432  a=0.000075054168701  b=-0.000013732910156 d=-0.003509902954102  cnt=47
error=0.001958112945283  delta_size=0.000000047683716  a=0.000075149536133  b=-0.000013685226440 d=-0.003524637222290  cnt=51
error=0.001958112780096  delta_size=0.000000023841858  a=0.000075173377991  b=-0.000013685226440 d=-0.003527092933655  cnt=53
error=0.001958112648182  delta_size=0.000000011920929  a=0.000075173377991  b=-0.000013661384583 d=-0.003529536724091  cnt=56
error=0.001958112606767  delta_size=0.000000005960464  a=0.000075185298920  b=-0.000013655424118 d=-0.003531378507614  cnt=60
error=0.001958112606766  delta_size=0.000000002980232  a=0.000075185298920  b=-0.000013655424118 d=-0.003531381487846  cnt=62
error=0.001958112602482  delta_size=0.000000001490116  a=0.000075188279152  b=-0.000013652443886 d=-0.003531993925571  cnt=67
error=0.001958112602235  delta_size=0.000000000745058  a=0.000075189024210  b=-0.000013652443886 d=-0.003532071411610  cnt=69
error=0.001958112601984  delta_size=0.000000000372529  a=0.000075189769268  b=-0.000013651698828 d=-0.003532224893570  cnt=74
error=0.001958112601972  delta_size=0.000000000186265  a=0.000075189769268  b=-0.000013651512563 d=-0.003532243706286  cnt=76
error=0.001958112601962  delta_size=0.000000000093132  a=0.000075189955533  b=-0.000013651419431 d=-0.003532272577286  cnt=80

Results from high speed linear fit:                    a=0.000075190022350  b=-0.000013651340876 d=-0.003532287550607

[... cut ...]
	x2bar = x2bar/n - xbar*xbar;
	y2bar = y2bar/n - ybar*ybar;
	z2bar = z2bar/n - zbar*zbar;
	xybar = xybar/n - xbar*ybar;
	yzbar = yzbar/n - ybar*zbar;
	xzbar = xzbar/n - xbar*zbar;

	D = x2bar*y2bar - xybar*xybar;
	if (fabs(D) <= 1e-15*( max_absx + max_absy )) {
		printf("error: x,y points are collinear\n");
		return NULL;
	}

	inc_results.A = (xzbar*y2bar - yzbar*xybar) / D;
	inc_results.B = (yzbar*x2bar - xzbar*xybar) / D;
	inc_results.D = zbar - inc_results.A*xbar - inc_results.B*ybar;

	return &inc_results;
}
```

edwilliams16, 2017-04-21:
One thing is the bug we fixed earlier:
```
linear_fit_results.D = -(zbar + linear_fit_results.A*xbar + linear_fit_results.B*ybar);
```
and incremental_LSF shouldn't return anything. It only updates the static variables xbar ... max_absy
if I follow the logic.  If there's something else, I don't see it yet.

Roxy-3D, 2017-04-21:
> One thing is the bug we fixed earlier:
```cpp
linear_fit_results.D = -(zbar + linear_fit_results.A*xbar + linear_fit_results.B*ybar);
```
Yes, I remember that error...  But the new code was derived from your post here: https://github.com/MarlinFirmware/Marlin/issues/5427#issuecomment-292757879

I changed that line and D was still 0.000   I guess I'll print out all the sub-terms and see if I can get a hint.   But did you notice that A and B are negated?  I don't see why that is happening either!

I added an extra printf to see the sub-terms:
```cpp
	inc_results.A = (xzbar*y2bar - yzbar*xybar) / DDD;
	inc_results.B = (yzbar*x2bar - xzbar*xybar) / DDD;
	inc_results.D = zbar + inc_results.A*xbar + inc_results.B*ybar;

	printf("zbar=%14.10f     inc_results.A=%14.10f  xbar=%14.10f    inc_results.B=%14.10f    ybar=%14.10f\n", zbar, inc_results.A, xbar, inc_results.B, ybar);
	return &inc_results;
```
and I get this...  It looks like A & B are correct (except negated) but xbar and ybar both equal 0.000
```cpp
error=0.001958112606767  delta_size=0.000000005960464  a=0.000075185298920  b=-0.000013655424118 d=-0.003531378507614  cnt=60
error=0.001958112606766  delta_size=0.000000002980232  a=0.000075185298920  b=-0.000013655424118 d=-0.003531381487846  cnt=62
error=0.001958112602482  delta_size=0.000000001490116  a=0.000075188279152  b=-0.000013652443886 d=-0.003531993925571  cnt=67
error=0.001958112602235  delta_size=0.000000000745058  a=0.000075189024210  b=-0.000013652443886 d=-0.003532071411610  cnt=69
error=0.001958112601984  delta_size=0.000000000372529  a=0.000075189769268  b=-0.000013651698828 d=-0.003532224893570  cnt=74
error=0.001958112601972  delta_size=0.000000000186265  a=0.000075189769268  b=-0.000013651512563 d=-0.003532243706286  cnt=76
error=0.001958112601962  delta_size=0.000000000093132  a=0.000075189955533  b=-0.000013651419431 d=-0.003532272577286  cnt=80

Results from high speed linear fit:                    a=0.000075190022350  b=-0.000013651340876 d=-0.003532287550607

Reset of incremental                                   A=0.000000000000000  B=0.000000000000000 D=0.000000000000000
zbar=  0.0000000000     inc_results.A= -0.0000751900  xbar=  0.0000000000    inc_results.B=  0.0000136513    ybar=  0.0000000000
Results from incremental linear fit:                   A=-0.000075190022350  B=0.000013651340876 D=0.000000000000000
```

Roxy-3D, 2017-04-21:
Hey @edwilliams16  !!!  It is working!    As it turns out, we were modifying the data sets in the original high speed linear fit code.  So when I fed the data set into the incremental version of the code...  Of course it was producing the wrong results.   I just changed the order I was calling each algorithm so the one that destroyed the data set was called last.
```cpp
void linear_fit_subtract_mean(double *matrix, double bar, int n) {
	int i;
	for (i = 0; i < n; i++) {
		matrix[i] -= bar;
	}
	return;
}
```
But I also had to switch these lines:
```cpp
	linear_fit_results.A = -(xzbar*y2bar - yzbar*xybar) / D;
	linear_fit_results.B = -(yzbar*x2bar - xzbar*xybar) / D;
//	linear_fit_results.D = -(zbar - linear_fit_results.A*xbar - linear_fit_results.B*ybar);
	linear_fit_results.D = -(zbar + linear_fit_results.A*xbar + linear_fit_results.B*ybar);
```
to fix the A and B being negated issue.   But now...   All three versions agree on the results!
THANK YOU FOR THE HELP!!!!    Soon, you algorithm will be alive and doing its thing inside Marlin.
```cpp
Doing xxx:    size of triples: 4
Results from iterative linear fit:
error=0.009803921568627  delta_size=0.100000000000000  a=-0.100000000000000  b=-0.100000000000000 d=0.000000000000000  cnt=2
error=0.004878048780488  delta_size=0.050000000000000  a=-0.150000000000000  b=-0.050000000000000 d=0.000000000000000  cnt=5
error=0.003683241252302  delta_size=0.025000000000000  a=-0.125000000000000  b=-0.050000000000000 d=0.000000000000000  cnt=7
error=0.000925212027756  delta_size=0.012500000000000  a=-0.112500000000000  b=-0.025000000000000 d=-0.025000000000000  cnt=11
error=0.000231722859460  delta_size=0.006250000000000  a=-0.106250000000000  b=-0.012500000000000 d=-0.037500000000000  cnt=15
error=0.000057974935503  delta_size=0.003125000000000  a=-0.103125000000000  b=-0.006250000000000 d=-0.043750000000000  cnt=19
error=0.000014498742234  delta_size=0.001562500000000  a=-0.101562500000000  b=-0.003125000000000 d=-0.046875000000000  cnt=23
error=0.000003625279071  delta_size=0.000781250000000  a=-0.100781250000000  b=-0.001562500000000 d=-0.048437500000000  cnt=27
error=0.000000906391921  delta_size=0.000390625000000  a=-0.100390625000000  b=-0.000781250000000 d=-0.049218750000000  cnt=31
error=0.000000226606872  delta_size=0.000195312500000  a=-0.100195312500000  b=-0.000390625000000 d=-0.049609375000000  cnt=35
error=0.000000056652822  delta_size=0.000097656250000  a=-0.100097656250000  b=-0.000195312500000 d=-0.049804687500000  cnt=39
error=0.000000014163343  delta_size=0.000048828125000  a=-0.100048828125000  b=-0.000097656250000 d=-0.049902343750000  cnt=43
error=0.000000003540853  delta_size=0.000024414062500  a=-0.100024414062500  b=-0.000048828125000 d=-0.049951171875000  cnt=47
error=0.000000000885215  delta_size=0.000012207031250  a=-0.100012207031250  b=-0.000024414062500 d=-0.049975585937500  cnt=51
error=0.000000000221304  delta_size=0.000006103515625  a=-0.100006103515625  b=-0.000012207031250 d=-0.049987792968750  cnt=55
error=0.000000000055326  delta_size=0.000003051757813  a=-0.100003051757813  b=-0.000006103515625 d=-0.049993896484375  cnt=59
error=0.000000000013832  delta_size=0.000001525878906  a=-0.100001525878906  b=-0.000003051757813 d=-0.049996948242187  cnt=63
error=0.000000000001153  delta_size=0.000000762939453  a=-0.100000762939453  b=-0.000000000000000 d=-0.049999237060547  cnt=68
error=0.000000000000864  delta_size=0.000000381469727  a=-0.100000762939453  b=-0.000000381469727 d=-0.049999237060547  cnt=70
error=0.000000000000072  delta_size=0.000000190734863  a=-0.100000190734863  b=-0.000000000000000 d=-0.049999809265137  cnt=75
error=0.000000000000054  delta_size=0.000000095367432  a=-0.100000190734863  b=-0.000000095367432 d=-0.049999809265137  cnt=77
error=0.000000000000005  delta_size=0.000000047683716  a=-0.100000047683716  b=-0.000000000000000 d=-0.049999952316284  cnt=82
error=0.000000000000003  delta_size=0.000000023841858  a=-0.100000047683716  b=-0.000000023841858 d=-0.049999952316284  cnt=84
error=0.000000000000000  delta_size=0.000000011920929  a=-0.100000000000000  b=-0.000000011920929 d=-0.049999988079071  cnt=89
error=0.000000000000000  delta_size=0.000000005960464  a=-0.100000000000000  b=-0.000000005960464 d=-0.049999994039536  cnt=92
error=0.000000000000000  delta_size=0.000000002980232  a=-0.100000000000000  b=-0.000000002980232 d=-0.049999997019768  cnt=95
error=0.000000000000000  delta_size=0.000000001490116  a=-0.100000000000000  b=-0.000000001490116 d=-0.049999998509884  cnt=98
error=0.000000000000000  delta_size=0.000000000745058  a=-0.100000000000000  b=-0.000000000745058 d=-0.049999999254942  cnt=101
error=0.000000000000000  delta_size=0.000000000372529  a=-0.100000000000000  b=-0.000000000372529 d=-0.049999999627471  cnt=104
error=0.000000000000000  delta_size=0.000000000186265  a=-0.100000000000000  b=-0.000000000186265 d=-0.049999999813735  cnt=107
error=0.000000000000000  delta_size=0.000000000093132  a=-0.100000000000000  b=-0.000000000093132 d=-0.049999999906868  cnt=110
Results from incremental linear fit:                   A=-0.100000000000000  B=+0.000000000000000 D=-0.050000000000000
Results from high speed linear fit:                    a=-0.100000000000000  b=-0.000000000000000 d=-0.050000000000000

Doing xxx0:    size of triples: 3
Results from iterative linear fit:
error=0.009803921568627  delta_size=0.100000000000000  a=-0.100000000000000  b=-0.100000000000000 d=0.000000000000000  cnt=2
[... cut ...]
error=0.001958112601984  delta_size=0.000000000372529  a=0.000075189769268  b=-0.000013651698828 d=-0.003532224893570  cnt=74
error=0.001958112601972  delta_size=0.000000000186265  a=0.000075189769268  b=-0.000013651512563 d=-0.003532243706286  cnt=76
error=0.001958112601962  delta_size=0.000000000093132  a=0.000075189955533  b=-0.000013651419431 d=-0.003532272577286  cnt=80
Results from incremental linear fit:                   A=+0.000075190022350 B=-0.000013651340876 D=-0.003532287550607
Results from high speed linear fit:                    a=+0.000075190022350 b=-0.000013651340876 d=-0.003532287550607
```

edwilliams16, 2017-04-21:
I see how A and B got zeroed - since the modified data had its mean subtracted out - but I'm not sure how they ended up with the wrong sign when that was fixed...  Sign errors are a PITA. Anyhow it works.

Roxy-3D, 2017-04-21:
Yeah...  and it now makes sense why D was 0.000   The mean was getting subtracted off of all the data points!

[5 more replies]

closed 2018-02-06 by thinkyhead