MarlinFirmware/Marlin #5428: Best way to upgrade from (modified) RC3 to RC8?

JohnOCFII, 2016-12-08:
Greetings,

I've got a Geeetech G2S Pro running a modified version of RC3 (located here: https://github.com/phil-hx/Marlin - which, I believe is heavily influenced by the Rich Cattell version of Marlin https://github.com/RichCattell/Marlin ).  I'm wondering what is the best approach to upgrading to RC8?  Should I start by using my existing configuration.h and configuration_adv.h?  Or is it better to start with the included default configuration.h and configuration_adv.h, modify the configuration items that are obvious and make guesses on the rest (or ask here?).  

Thanks for any suggestions,

John

emartinez167, 2016-12-08:
There is a migration tool developed by Johnny Linden. Search for it in Google+ under the 3D printing group.

Regards,

Ernesto

scott1326, 2016-12-08:
Hi John,
     I keep an original version of my current firmware and use a comparison program (eg, WinMerge, Beyond Compare) to compare that to my modified version.  Then I go through and make those changes one-by-one in the new releases.
    
     Scott

Roxy-3D, 2016-12-08:
Yeah...   I do the same thing.  Except I use ExamDiff Pro.

christianh17, 2016-12-08:
Hi!
In the past I used Meld, two weeks ago I changed to a local git and github. Every merge worked perfect. The only problem: The GUI does not support tags (at least as far as i know) so I create a branch for every version I us on my printer.
Regards
Christian

JohnOCFII, 2016-12-08:
Thanks everyone.  I'm familiar with the -diff- type process.  I'm currently doing that with TextWrangler on the Mac.  I'll take a look at the migration tool and see what it offers.  Other than that, I guess I'll pull up a couple of editor windows and sit down with a nice cup of coffee!

thinkyhead, 2016-12-09:
If you drop in your old configurations and try to build, the `SanityCheck.h` file will catch the things you need to change and print messages in the console telling you what needs to be updated (and sometimes how).

JohnOCFII, 2016-12-09:
> If you drop in your old configurations and try to build, the SanityCheck.h file will catch the things you need to change and print messages in the console telling you what needs to be updated (and sometimes how).

Even better!  Thanks.

thinkyhead, 2016-12-09:
Note that that method may not catch options that are simply missing, if they are not required to be there. But it will tell you when some option you have enabled or set needs its name or usage to be changed.

closed 2016-12-25 by JohnOCFII