MarlinFirmware/Marlin #5433: Printing with DISTINCT_E_FACTORS produces garbage

Kaibob2, 2016-12-08:
With` #define DISTINCT_E_FACTORS` disabled it prints perfect what it should, with enabled `#define DISTINCT_E_FACTORS` the Y Axis makes a unhealthy sound as if the speed would be much to high but doesn't move. The Display says Y would have moved to the correct position.
Then X moves somewhere and starts with the infill of the first layer.
It's completely messed up.

The model is
[image]
The GCODE is
[warp2.zip](https://github.com/MarlinFirmware/Marlin/files/639242/warp2.zip)
My configuration
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/639259/Marlin.zip)

realhuno, 2016-12-08:
Yes, i can confirm this.... same issue ;-( 
I crashed two times. 

X Axis shift to max and start printing in the air.

thinkyhead, 2016-12-09:
Try `M502` followed by `M500` just to make sure everything is cool.

Kaibob2, 2016-12-09:
With the values that come from RCBugfix or mine?
Other issue: 
M92 X80.30 Y80.30 Z1600.00
M203 X300.00 Y300.00 Z9.00
M201 X1500 Y1500 Z100
are missing the E values
[image]

Kaibob2, 2016-12-12:
@thinkyhead 

> Try M502 followed by M500 just to make sure everything is cool.

Did this and tested with RCBugFix as of 10.12.2016 and it still doesn't work.

thinkyhead, 2016-12-13:
Apparently the host software is filtering out `M92 Tn En` and others. That's odd.

Kaibob2, 2016-12-13:
This
```
if (!forReplay) {
      SERIAL_ECHOLNPGM("Steps per unit:");
      CONFIG_ECHO_START;
    }
    SERIAL_ECHOPAIR("  M92 X", planner.axis_steps_per_mm[X_AXIS]);
    SERIAL_ECHOPAIR(" Y", planner.axis_steps_per_mm[Y_AXIS]);
    SERIAL_ECHOPAIR(" Z", planner.axis_steps_per_mm[Z_AXIS]);
    #if DISABLED(DISTINCT_E_FACTORS)
      SERIAL_ECHOPAIR(" E", planner.axis_steps_per_mm[E_AXIS]);
    #endif
    //SERIAL_EOL;
    #if ENABLED(DISTINCT_E_FACTORS)
      for (uint8_t i = 0; i < E_STEPPERS; i++) {
        SERIAL_ECHOPAIR("  M92 T", (int)i);
        SERIAL_ECHOPAIR(" E", planner.axis_steps_per_mm[E_AXIS + i]);
      }
    #endif
    SERIAL_EOL;
```

produces this in Simplify3D

[image]

Don't know if this helps. I tried almost any location for `SERIAL_EOL;` but didn't got it working any better.

thinkyhead, 2017-02-11:
Any news on this issue?

Elenedeath, 2017-02-22:
all here, i try to know why my printer dont print on second extruder and when DISTINCT_E_FACTORS enabled, always skip origin and know middle bed are X0 Y0 and go hit the x axis +

closed 2017-10-31 by thinkyhead