MarlinFirmware/Marlin #5437: MBL starting printhead to high about 10mm above the bed

garaguido, 2016-12-08:
When I use the MBL function in Marlin RC8 , the printhead start at 10mm above the printbed , and I must corrected more than -10mm to adjust the levelhight at Z 0.1 mm . 
How can I adjust the Z hight settings , in the Marlin RC8 Configuration.h , for starting the MBL at Z 0.1mm and adjusting the 9 points  bed correction level afterwords .
This was not my problem when I use MBL in RC6 , the printhead was correct starting  at Z 0.1mm

thinkyhead, 2016-12-08:
Hi @garaguido — It's probably a bug and not a configuration error. But can you post your `Configuration.h` so that we can see where the 10mm comes from? (You might need to add `.txt` to the name for Github to allow you to drop it on the text-field.) Hopefully I can figure it out and have a patch for you by tomorrow.

garaguido, 2016-12-09:
Hi
Here are my settings in the configuration.h file
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */
[... 1371 lines cut ...]
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         1.85  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.65  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H
.txt

psavva, 2016-12-24:
@thinkyhead I can confirm the same bug, running RC8.
I'm busy testing it now with RCBugFix as of this moment (just downloaded)

JonnyLeRoy, 2016-12-26:
Thank you all, for working on this bug. I spent many a day. Not understanding why I couldn't understand simple instructions.

brainscan, 2016-12-26:
I can't reproduce this problem with RC8 and mesh levelling is working fine for me so maybe this is a configuration problem?

psavva, 2016-12-26:
@braiscan, please can you post your configuration file for us to compare?

garaguido, 2016-12-27:
@brainscan , yes maybe your configuration.h file , please, then we can compare it.

brainscan, 2016-12-29:
I'll try and post it in the next couple of days, I've not been with my pc so I don't have a copy of my configuration.h

garaguido, 2016-12-31:
@brainscan , Thanks for the effort..

JonnyLeRoy, 2017-01-30:
garaguido

Did you find a fix?

garaguido, 2017-01-30:
No , I am still waithing for the configuration.h  file from @brainscan . I uploaded the resent Marlin Bugfix file but I am not tested it yet , perhaps in a few days I will try it out  and reply my comment to you

brainscan, 2017-01-30:
My apologies for not posting my config but my wife had to have a caesarean to deliver our twin girls and they've been in special care as they were early and small, so I have been a little preoccupied, I hope you understand. They're all doing much better so once they're home I should be able to post it, although hopefully the latest RCBugFix will have already solved this problem.

garaguido, 2017-01-30:
Oh , I am sorry to hear this . My regards to you and to your wife for your new born twin girls.
I let you now about my new tests within a few days

JonnyLeRoy, 2017-01-31:
Im glad all is good to grate with your family. Sleep is just a bad ;) as you now are learning.

JonnyLeRoy, 2017-01-31:
I cant see what I changed in my settings, Auto Level is now working. Can you see what i did that caused the problem we were having?

/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */
[... 1365 lines cut ...]
 * 34 - RAMPS1.4 - uses Analog input 5 on the AUX2 connector
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H

garaguido, 2017-02-01:
Your config file is not the same as mine.
I use the Mesh bed leveling feature and not the ABL , your config in Mesch leveling was not activated.
The difference of 10mm is with the Manuel Bed Leveling.
I will test it with the latest Marlin Bugfix and let you now if the problem still occured

garaguido, 2017-02-01:
I tested today with the MarlinBugfix from 14 jan 2017 and the result is still the same 10mm to high at Z0.
When I use the Marlin software from 23 april 2016 the Z0 is at the correct  0,1mm ( =end switch position) before I use the Mesh bedleveling command.

JonnyLeRoy, 2017-02-02:
Sometimes I'm not very bright.
I had not thought to try just ABL when I was having my problems. I just got so frustrated I could not think.
I took off  the sensor and made my print bed with manual leveling. On  9mm Glass. Not a fast or simple fix, but it is a fix.
I decided I would give the Auto a fresh look. Now that all the other things were worked out. Now that I know my print bed is flat I did not try Manuel Bed Leveling. 
I'm so glad for everyone's help. Software is soooo not my thing.  It was trying to make me crazy. 
It would have if not for your help!

garaguido, 2017-02-02:
@thinkyhead , 
Have you found the sollution ( Bug) yet ? 
I have tested the latest Marlin Bugfix with NO result , it is still 10mm to high with G28 before Mesh beleveling command. 
It is working fine with the earlear Marlin Bugfix version from 14 april 2016.
Must I  use perhaps the M420 offset code and store this in the Eprom for the sollution perhaps?

thinkyhead, 2017-02-12:
At some point I modified the behavior of MBL and probably broke something while trying to deal with the `MESH_HOME_SEARCH_Z` setting, `Z_HOMING_HEIGHT`, etc.

To help locate the issue, please enable `DEBUG_LEVELING_FEATURE`, then issue `M111 S247` to enable full logging. Do a `G28` followed by `G29` and then post the log output here for examination.

garaguido, 2017-02-12:
Hi, 
G29 after G28 is this OK , I use only MBL?

garaguido, 2017-02-14:
@thinkyhead , Hi , hereby the Logfile you ask before. I hope this can solve my problem ?

Connecting...
start
Printer is now online.
echo:Marlin 1.1.0-RC8
echo: Last Updated: 2016-12-06 12:00 | Author: (none, default config)
Compiled: Feb 14 2017
echo: Free Memory: 3465  PlannerBufferBytes: 1168
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E95.00
echo:Maximum feedrates (mm/s):
echo:  M203 X200.00 Y200.00 Z8.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z200 E5000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
Mesh bed leveling:
echo:  M420 S0 X3 Y3
echo:  G29 S3 X1 Y1 Z0.00000
echo:  G29 S3 X2 Y1 Z0.00000
echo:  G29 S3 X3 Y1 Z0.00000
echo:  G29 S3 X1 Y2 Z0.00000
echo:  G29 S3 X2 Y2 Z0.00000
echo:  G29 S3 X3 Y2 Z0.00000
echo:  G29 S3 X1 Y3 Z0.00000
echo:  G29 S3 X2 Y3 Z0.00000
echo:  G29 S3 X3 Y3 Z0.00000
echo:Material heatup parameters:
echo:  M145 S0 H205 B55 F0
  M145 S1 H240 B90 F0
echo:PID settings:
echo:  M301 P18.73 I1.35 D64.90
echo:  M304 P257.75 I39.13 D424.42
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-5.11
echo:SD card ok
echo:SD card ok
>>>M111 S247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
>>>G28
SENDING:G28
echo:G28
>>> gcode_G28
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X:15 Y:-5 Z:-5.11 (Right-Front & Below Nozzle)
Mesh Bed Leveling
  current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 4.00
>>> do_blocking_move_to(0.00, 0.00, 4.00)
<<< do_blocking_move_to
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, -300.00, 0.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, 5.00, 0.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, -10.00, 25.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
<<< do_homing_move(X)
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 200.00
> home_offset[X] = 0.00
  current_position=(0.00, 0.00, 4.00) :
<<< set_axis_is_at_home(X)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
  current_position=(0.00, 0.00, 4.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
  current_position=(0.00, 0.00, 4.00) : > homeX
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, -300.00, 0.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
echo:busy: processing
echo:busy: processing
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, 5.00, 0.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, -10.00, 25.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
<<< do_homing_move(Y)
>>> set_axis_is_at_home(Y)
For Y axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 200.00
> home_offset[Y] = 0.00
  current_position=(0.00, 0.00, 4.00) :
<<< set_axis_is_at_home(Y)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
  current_position=(0.00, 0.00, 4.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
  current_position=(0.00, 0.00, 4.00) : > homeY
Z_SAFE_HOMING >>>
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
  destination=(85.00, 105.00, 4.00) : Z_SAFE_HOMING
>>> do_blocking_move_to(85.00, 105.00, 4.00)
echo:busy: processing
<<< do_blocking_move_to
>>> homeaxis(Z)
  current_position=(85.00, 105.00, 4.00) : set_probe_deployed
deploy: 1
do_probe_raise(10.00)
>>> do_blocking_move_to(85.00, 105.00, 15.11)
echo:busy: processing
<<< do_blocking_move_to
>>> do_blocking_move_to(85.00, 105.00, 15.11)
<<< do_blocking_move_to
Home 1 Fast:
>>> do_homing_move(Z, -255.00, 0.00)
  current_position=(85.00, 105.00, 0.00) : sync_plan_position
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 5.00, 0.00)
  current_position=(85.00, 105.00, 0.00) : sync_plan_position
echo:busy: processing
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -10.00, 1.00)
  current_position=(85.00, 105.00, 0.00) : sync_plan_position
echo:busy: processing
echo:busy: processing
echo:busy: processing
<<< do_homing_move(Z)
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 170.00
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> zprobe_zoffset = -5.11
> home_offset[Z] = 0.00
  current_position=(85.00, 105.00, 5.11) :
<<< set_axis_is_at_home(Z)
  current_position=(85.00, 105.00, 5.11) : sync_plan_position
  current_position=(85.00, 105.00, 5.11) : > AFTER set_axis_is_at_home
  current_position=(85.00, 105.00, 5.11) : set_probe_deployed
deploy: 0
do_probe_raise(10.00)
>>> do_blocking_move_to(85.00, 105.00, 15.11)
echo:busy: processing
<<< do_blocking_move_to
>>> do_blocking_move_to(85.00, 105.00, 15.11)
<<< do_blocking_move_to
<<< homeaxis(Z)
<<< Z_SAFE_HOMING
  current_position=(85.00, 105.00, 15.11) : > (home_all_axis || homeZ) > final
  current_position=(85.00, 105.00, 15.11) : sync_plan_position
  current_position=(85.00, 105.00, 15.11) : clean_up_after_endstop_or_probe_move
<<< gcode_G28
X:85.00 Y:105.00 Z:15.11 E:0.00 Count X: 6800 Y:8400 Z:6044
>>>G29
SENDING:G29
echo:G29
Mesh bed leveling not active.
X:85.00 Y:105.00 Z:15.11 E:0.00 Count X: 6800 Y:8400 Z:6044

garaguido, 2017-02-14:
@thinkyhead 
I have a quick look at the logfile , and what for me strange behavior is " current_position .... Z = 15.11 " minus my Z offset with -5.11 is still 10 mm to high . Why also the .... do_probe_raise (10.00) that is perhaps the problem ?

Pipshag, 2017-02-25:
It seems to me that the offset it the same as the Z_CLEARANCE_DEPLOY_PROBE value. I'm getting an offset error of 2 whereas garaguido gets 10 (which correlates with our  Z_CLEARANCE_DEPLOY_PROBE values).

thinkyhead, 2017-02-26:
As it homes using the probe, the position of the nozzle when the probe triggers is 5.11mm. You have the probe configured to do a raise of 10mm when "stowing" - or disabling. The position should be correct after a `G28` without `G29`.

MBL "hijacks" `G28` for its final move after leveling is done, and is probably setting Z without taking account of this special usage. (It's treating it like the usual `G28` and failing to account for the probe raise.)

So, could you also post a log of `G29`. The log above only includes `G28`.

Pipshag, 2017-02-26:
After G28 the displayed Z-position is a false 0. By sending G1 Z0 the bed moves up to the correct Z0 position and after that everything works as intended. Somewhere there's a mismatch of values.

james94jeans2, 2017-02-26:
This is what I mentioned in here #5885

garaguido, 2017-02-26:
@thinkyhead , 
I ame not  using ABL , you mensioned a probe ... where ...? 
With MBL there is no probe  , I only level the printhead to the bed with a feelergage  !!!

Pipshag, 2017-02-28:
@james94jeans2 Thank you for making the reference to the other issue. I will try with making M420 Z[height above MESH_HOME_SEARCH_Z] which would work to set the correct Z-height, if I understood your point correctly?

james94jeans2, 2017-03-01:
I would guess that there can't be any problem when ENABLE_LEVELING_FADE_HEIGHT is disabled, as you can not set a z_fade_height when the feature is disabled.

And yes, you understood my point correctly.

Pipshag, 2017-03-01:
@james94jeans2 I did test by disabling ENABLE_LEVELING_FADE_HEIGHT and that was the solution. A thing to note is that the LCD-display is showing the correct height (in my case it showed Z5.03) after G28 - when ENABLE_LEVELING_FADE_HEIGHT is enabled the display shows Z0 after a G28.

james94jeans2, 2017-03-01:
And the wrong Z display also happens when you don't specify a fade_height?

Pipshag, 2017-03-01:
Z0 condition after G28 was:
define ENABLE_LEVELING_FADE_HEIGHT
no usage of M420

Working condition was:
//define ENABLE_LEVELING_FADE_HEIGHT

Never had the chance to test with M420 above threshold of MESH_HOME_SEARCH_Z.

james94jeans2, 2017-03-01:
I think the Leveling Fade needs some revisiting, judging by this wierd behaviour.

garaguido, 2017-03-01:
I will try this out on my Prusa I3 .

garaguido, 2017-03-06:
@james94jeans2 ,
My MBL system is working now with the new RC8bugfix from 14-01-17 , what I dit was 👍 
- undefine ENABLE_LEVELING_FADE_HEIGHT , testing with this undefine only the MBL did not working

The MBL was well working with these extra settings ;
- undefine ENABLE_LEVELING_FADE_HEIGHT ( not the cause I think)
- undefine FIX_MOUNTED_PROBE ( I do not use ABL this was by me defined when using MBL ) perhaps there seems the problem , a bug perhaps ?
- define NOZZLE_PARK_FEATURE ( not the cause I think)

I will test this now with only FIX_MOUNTED_PROBE undefined

garaguido, 2017-03-06:
@james94jeans2 ,
My conclusion was correct , the problem seems to happen when FIX_MOUNTED_PROBE is defined when not defining ABL .
I have tested it and MBL is working fine now with the MarlinRC8 bugfix from 14 jan 2017.
The Z0 height is now correct and the leveling feature is working fine now.

thinkyhead, 2017-03-30:
@james94jeans2 I believe the Z position should now be correct in the case you cited at https://github.com/MarlinFirmware/Marlin/issues/5885#issuecomment-282537599 because there's no more `Z_HOME_SEARCH` stuff. We just disable software endstops now. If you do the same test again, you should get a proper Z position.

garaguido, 2017-05-12:
I used the new released Marlin 1.1.0 version with MBL and  I have the same problem as before "Printing 10 mm to High" after a MBL procedure.
The version Marlin-RCBugfix from 23 of April is working fine by me , what is there changed in the newer Marlin 1.1.0 about this problem ? After I flashed the firmware with the RCBugfix version of 23 april all things working fine now.

psavva, 2017-05-12:
Please use this branch
https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x?files=1

It contains the bugfixes for 1.1.0

thinkyhead, 2017-05-14:
We'll have to keep looking into it. Enable `DEBUG_LEVELING_FEATURE` and do `M111 S32` before `G29` to produce a log that includes some additional details which might be useful to narrowing the cause of this issue.

garaguido, 2017-05-24:
I can not flash my Atmega2560 any more , I tink there is a problem with the electronic UART. 
When I uploaded the sketch the Tx RX leds are blinking , but when it is done  uploading still (no errors)  there is same flashprogram as before , no changes , weard behavior ....?

Bob-the-Kuhn, 2017-05-24:
I've had that problem more than once.  It's always been because I was editing a file that was NOT in the directory being used by the Arduino IDE.  Sometimes I just grabbed the wrong file. Sometimes I selected the wrong item from the recently used list.  Rarely it's been solved by closing the Arduino IDE and then re-opening.

garaguido, 2017-05-26:
I have reflashed the Atmeg2506 with new fuses and bootloader and its working again.
Flashed the Atmega2560 with the MarlinBugfix 1.1.x and its working correct now , after a manual bedleveling the hight of the nozzle is now correct.

Thanks foor you effort.

garaguido, 2017-06-01:
I think I have a solution for my problem, I must set a G28 code in the front of my scriptfiles when I starting a print ,then  the nozzle hight is OK .  
When I starting a print without the G28 in the front of the script the nozzle is starting 10mm too high. 
Is this the normal procedure?

Roxy-3D, 2017-06-01:
Pretty much...   Every slicer will include default gcode at the front of the file.  And one of the very first things in that default gcode will be a G28.

garaguido, 2017-06-02:
Thanks for your help.

closed 2017-06-02 by garaguido

thinkyhead, 2017-06-07:
@garaguido Check the G-code on the front of your script. If it has a command like `G92 Z0` or something else which makes assumptions that the nozzle is at the home position, then it will not print correctly. Just make sure the start G-code has only the minimum commands needed to get ready for a print.