MarlinFirmware/Marlin #5451: DHT11 or DHT22 support?

Tannoo, 2016-12-10:
Would it be possible to incorperate the use of one of these?
Could have the data displayed in the 'about' screen.

Blue-Marlin, 2016-12-10:
Of cause. That is possible.

The more interesting question is, if the 'win' for a 3D-printer is big enough to justify the effort?
Marlin primary is for 3D-printers - not for weather stations.

Tannoo, 2016-12-10:
Lol.. Yes, but to monitor ambient temp and humidity at the printer.

Blue-Marlin, 2016-12-10:
Have had a look into 'DHT library'. Its full of delays and active waits - unusable for Marlin.
Ether one would have to stop printing for a measurement, or one had to rewrite the library - a major task.

For me value/effort is too low.

Tannoo, 2016-12-10:
That's ok. I will attempt something. Hopefully, I don't break it too bad. Lol

The timing issues.. I was thinking of only getting these reading while it is not printing.

closed 2016-12-11 by Tannoo