MarlinFirmware/Marlin #5458: Documentation for multiextruders

alexxy, 2016-12-10:
From config file its not clear what is the logic of multi extruder offsets, distance in what direction will be calculated?

And it seems it behave differently if i set it via M218

ekozan, 2016-12-10:
//#define EXTRUDER_OFFSET_X {0.0, 20.00}
//#define EXTRUDER_OFFSET_Y {0.0, 0.00}

It's the same axis direction of the printer but in relative coordinate

alexxy, 2016-12-11:
So formula for second extruder coordinates are

E2_X = CURR_X - EXTRUDER_OFFSET_X[1]
E2_Y = CURR_Y - EXTRUDER_OFFSET_Y[1]
E2_Z = CURR_Z - EXTRUDER_OFFSET_Z[1]

Or something different?

I'm asking because i dont see this behavior.

ekozan, 2016-12-11:
first is 0,0 second is 0+X and 0+Y

alexxy, 2016-12-11:
what about Z?

landodragon141, 2016-12-11:
The Z will always be equal because they are generally mounted together on the X or Y axis.

alexxy, 2016-12-11:
Not for switching extruder

landodragon141, 2016-12-11:
What do you mean by switching extruder?

alexxy, 2016-12-12:
Its extruders like dondolo that uses servo to switch head

landodragon141, 2016-12-12:
Well in that case X Y and Z will always be equal since the hot ends "should" be in the same place when active. These values are mostly intended for multi hotends that are fixed so that marlin knows where they are relative to each other. You could use those values to fine tune but I recommend doing as much tuning physically as you can. One thing you could do to test calibration is print two thin squares (just a couple layers) one on top of the other using the two extruders. That will help you calibrate X and Y. I may be preaching to the choir but I hope this information is helpful.

thinkyhead, 2016-12-13:
You can also define a `HOTEND_OFFSET_Z { 0, z2 }` in the case of switching extruder.

closed 2018-02-26 by thinkyhead