MarlinFirmware/Marlin #5459: Migrating from RC7 to RC8  "unable to find a register to spill in class "POINTER_REGS"

pmbroth, 2016-12-11:
ERRO MEssage

Arduino: 1.6.12 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\temperature.cpp: In function 'Temperature::manage_heater() [clone .part.4]':

temperature.cpp:797: error: unable to find a register to spill in class 'POINTER_REGS'

 }

 ^

temperature.cpp:797: error: this is the insn:

(insn 100 99 103 3 (set (reg:SF 109 [ D.2252 ])

        (mem:SF (post_inc:HI (reg:HI 169 [ ivtmp.109 ])) [7 MEM[base: _144, offset: 0B]+0 S4 A8])) sketch\temperature.cpp:684 100 {*movsf}

     (expr_list:REG_INC (reg:HI 169 [ ivtmp.109 ])

        (nil)))

sketch\temperature.cpp:797: confused by earlier errors, bailing out

lto-wrapper: C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-gcc returned 1 exit status

c:/users/pmbroth/appdata/local/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
unable to find a register to spill in class 'POINTER_REGS'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

pmbroth, 2016-12-11:
Ran it again with the show verbose"

Arduino: 1.6.12 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\PMBROTH\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\PMBROTH\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\PMBROTH\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10612 -build-path C:\Users\PMBROTH\AppData\Local\Temp\arduino_build_200635 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2 -prefs=runtime.tools.avrdude.path=C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino6 -verbose P:\DATA\3d projects\formbot\Marlin Firmware\Marlin-RC 8\Formbot_T-Rex_Plus_BR\Formbot_T-Rex_Plus_BR.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\PMBROTH\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\PMBROTH\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\PMBROTH\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10612 -build-path C:\Users\PMBROTH\AppData\Local\Temp\arduino_build_200635 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2 -prefs=runtime.tools.avrdude.path=C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino6 -verbose P:\DATA\3d projects\formbot\Marlin Firmware\Marlin-RC 8\Formbot_T-Rex_Plus_BR\Formbot_T-Rex_Plus_BR.ino
Using board 'mega' from platform in folder: C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15
Using core 'arduino' from platform in folder: C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15
Detecting libraries used...
"C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15\cores\arduino" "-IC:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15\variants\mega" "C:\Users\PMBROTH\AppData\Local\Temp\arduino_build_200635\sketch\Formbot_T-Rex_Plus_BR.ino.cpp" -o "nul"
"C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15\cores\arduino" "-IC:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15\variants\mega" "C:\Users\PMBROTH\AppData\Local\Temp\arduino_build_200635\sketch\M100_Free_Mem_Chk.cpp" -o "nul"
"C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15\cores\arduino" "-IC:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15\variants\mega" "C:\Users\PMBROTH\AppData\Local\Temp\arduino_build_200635\sketch\MarlinSerial.cpp" -o "nul"
"C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15\cores\arduino" "-IC:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15\variants\mega" "C:\Users\PMBROTH\AppData\Local\Temp\arduino_build_200635\sketch\Marlin_main.cpp" -o "nul"
"C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15\cores\arduino" "-IC:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.15\variants\m [... cut ...]
C:\Users\PMBROTH\AppData\Local\Temp\arduino_build_200635\sketch\temperature.cpp:797: confused by earlier errors, bailing out

lto-wrapper: C:\Users\PMBROTH\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-gcc returned 1 exit status

c:/users/pmbroth/appdata/local/arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
unable to find a register to spill in class 'POINTER_REGS'

bobc, 2016-12-11:
This appears to be a general bug in gcc https://gcc.gnu.org/bugzilla/show_bug.cgi?id=60040

Possibly fixed in version of gcc 6.1 or later. 6.1 was released in March 2016 but is not in an Arduino update yet.

pmbroth, 2016-12-11:
Hello, how would I update the software with the fix?

Blue-Marlin, 2016-12-11:
Would be interesting how your 'Configuration.h' and 'Configuration_adv.h' looks like - to find out what combination of features causes this error. In almost all case Marlin compiles.

pmbroth, 2016-12-11:
[my configh and advh.zip](https://github.com/MarlinFirmware/Marlin/files/644732/my.configh.and.advh.zip)

Here you go!

Thank you..

pmbroth, 2016-12-11:
I have dual extruder capability, and inductive sensor..  Heater is not controlled by mksgen, it has a separate controller.  Printer is formbot TREX +

pmbroth, 2016-12-12:
Here is the rc7 con
[rc7 config files.zip](https://github.com/MarlinFirmware/Marlin/files/644950/rc7.config.files.zip)
fig files that work

thinkyhead, 2016-12-12:
Sometimes just rearranging code in the problematic file is enough to get it to compile. Try shuffling functions around in `temperature.cpp` and possibly it will work around the compiler's bug.

esenapaj, 2016-12-12:
@pmbroth 
If you disable `PROBE_DOUBLE_TOUCH` or `Z_MIN_PROBE_REPEATABILITY_TEST` or `Z_SAFE_HOMING`, compilation error is solved.
But I'm not sure why.

pmbroth, 2016-12-12:
Hello,
I will review my file tonight, and recheck the it for accuracy (5th time) and I will try the work arounds.  I will let you know how it goes.

Than k you

Brian

Fulg, 2016-12-12:
As a workaround, you can also downgrade your Arduino IDE to 1.6.11.

pmbroth, 2016-12-13:
if I DISABLE PROBE_DOUBLE_TOUCH or Z_MIN_PROBE_REPEATABILITY_TEST    IT FAILS WITH THE SAME ERROR, IF I COMMME OUIT  Z_SAFE_HOMING IT COMPILES

pmbroth, 2016-12-13:
Any suggestions?  I do not really want to disable safe homing.

thinkyhead, 2016-12-13:
It's a compiler issue, and so all we can do is sacrifice chickens.
I suggest trying Arduino 1.6.8, 1.6.9, and 1.6.10.

pmbroth, 2016-12-13:
@thinkyhead      Hello - By using an older compiler is new functionality lost?  I have been trying the nightly builds of the compiler, no gains there.   Thank you

bobc, 2016-12-13:
I would try Arduino 1.6.8 first, that has an older version of gcc (it was updated in 1.6.9 https://www.arduino.cc/en/Main/ReleaseNotes)

If you are feeling adventurous, you might try this http://blog.zakkemble.co.uk/avr-gcc-6-1-0/ but not having tried myself I can't vouch for it.

pmbroth, 2016-12-13:
Thank you, I will try tonight when I get home from work..  I appreciate all of the help..

pmbroth, 2016-12-14:
@bobc    Hi Bob,  I tried first the **new nightly build as of 12/13 - no good**, same error, I tried to install the new avr, per the link **no good**
I also removed the software, and installed the 1.6.8 version, and it worked.  Does this mean I lose something because it is an older version?

bobc, 2016-12-14:
@pmboth I think you should be ok with 1.6.8

pmbroth, 2016-12-16:
All seems to be working except the filament run out sensor.  I opened a new ticket for that.  Thank you everyone!

closed 2016-12-16 by pmbroth