MarlinFirmware/Marlin #5460: handle_status_leds referencing deprecated variables?

danzca6, 2016-12-11:
Pardon me if this is already being addressed. I couldn't find anything about it on here yet. I was playing with the TEMP_STAT_LEDS feature today in 1.1.0 RC8 and noticed that the routine handle_status_leds is referencing STAT_LED_RED and STAT_LED_BLUE that is deprecated according to the sanity checker. This message "STAT_LED_RED/STAT_LED_BLUE are now STAT_LED_RED_PIN/STAT_LED_BLUE_PIN. Please update your pins definitions." is thrown if either of those two are defined. Shouldn't the below actually be #if PIN_EXISTS(STAT_LED_RED_PIN) for example?

**In Marlin_main.cpp sub routine handle_status_leds:**
...
      if (new_led != red_led) {
        red_led = new_led;
        #if PIN_EXISTS(**STAT_LED_RED**)
          WRITE(STAT_LED_RED_PIN, new_led ? HIGH : LOW);
          #if PIN_EXISTS(**STAT_LED_BLUE**)
            WRITE(STAT_LED_BLUE_PIN, new_led ? LOW : HIGH);
          #endif
        #else
          WRITE(STAT_LED_BLUE_PIN, new_led ? HIGH : LOW);
        #endif
      }
...

**Also it is referenced in the setup of Marlin_main.cpp:**
  #if PIN_EXISTS(**STAT_LED_RED**)
    OUT_WRITE(STAT_LED_RED_PIN, LOW); // turn it off
  #endif

  #if PIN_EXISTS(**STAT_LED_BLUE**)
    OUT_WRITE(STAT_LED_BLUE_PIN, LOW); // turn it off
  #endif

I'm asking because the LED should show blue when the extruder is <54C and red when >55C and I am seeing red when the I start warming up to 200C (when extruder is under 54C). I did check my LED setup by enabling RGB_LED as well and can get M150 R255 to show red, M150 B255 to show blue. So I know the pins are setup right. I used the same pin numbers for STAT_LED_RED_PIN and STAT_LED_BLUE_PIN for testing the TEMP_STAT_LEDS feature. Just curious if it's something with the code or the way I am incorrectly understanding how this feature is supposed to work. Thank you.

bobc, 2016-12-11:
> Shouldn't the below actually be #if PIN_EXISTS(STAT_LED_RED_PIN) for example?

The macro PIN_EXISTS() adds "_PIN", so no.

danzca6, 2016-12-11:
Shoot you're right. Then my issue isn't that. Still wondering why it's red at low temps and not blue. I will have to play with that more. Thank you for pointing that out.

thinkyhead, 2016-12-13:
Apparently it also turns red if the bed temperature is over 55. Could that be the case?

closed 2016-12-13 by thinkyhead