MarlinFirmware/Marlin #5465: Arduino: 1.6.13 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Johnwr57, 2016-12-11:
I am trying to load the original factory install sketch (PLduino_firmware) from your site on my Digital-Loggers PLC with a Mega2560 and I am getting the message below.  Can anyone help?

C:\Users\JR\Desktop\plduino_firmware\plduino_firmware.ino:4:26: fatal error: Adafruit_GFX.h: No such file or directory

 #include <Adafruit_GFX.h>

                          ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

Blue-Marlin, 2016-12-11:
Likely you are wrong here. We don't have a `plduino_firmware.ino`. In our .ino file the includes begin at line 37.

bobc, 2016-12-11:
@Johnwr57 I'm pretty sure you want https://github.com/digitalloggers/PLDuino/issues

closed 2016-12-13 by thinkyhead