MarlinFirmware/Marlin #5472: Turkish translate error

ceturan, 2016-12-12:
error turkish translation on below line and all \xc7e strings
"#define MSG_RETRACT                         "Geri \xc7ek"                                               //Geri Çek"

\xc7 mean "Ç" but \xc7e mean "~" at unicode table.  in my opinion , A distinctive sign is required between \xc7 and e.

i use space between \xc7 and e.  But it looks bad  like  //Geri Ç ek"

boelle, 2016-12-12:
why not submit a PR to RCBugfix?

yhfudev, 2016-12-12:
This file is NOT encoded in unicode, you may take a look the file language_tr_utf8.h,
or you may want to use the new UTF-8 language engine for the firmware #5441.

AnHardt, 2016-12-12:
Please try
utf8-mapper for Turkish #5475.

ceturan, 2016-12-13:
Thank You @AnHardt  for turkish mapper, it is working, you rock. and @yhfudev .

and sorry @boelle i don't known submit PR , this is first  time :-)

closed 2016-12-13 by ceturan