MarlinFirmware/Marlin #5473: Language file error in language_es.h

yhfudev, 2016-12-12:
patch:
```diff
diff --git a/Marlin/language_es.h b/Marlin/language_es.h
index 37a5b72..7af292c 100644
--- a/Marlin/language_es.h
+++ b/Marlin/language_es.h
@@ -178,16 +178,19 @@
 #define MSG_INFO_EXTRUDERS                  "Extrusores"
 #define MSG_INFO_BAUDRATE                   "Baudios"
 #define MSG_INFO_PROTOCOL                   "Protocolo"
+#if LCD_WIDTH > 19
 #define MSG_INFO_PRINT_COUNT                "Conteo de impresion"
 #define MSG_INFO_COMPLETED_PRINTS           "Completadas"
 #define MSG_INFO_PRINT_TIME                 "Tiempo total de imp."
 #define MSG_INFO_PRINT_LONGEST              "Impresion mas larga"
 #define MSG_INFO_PRINT_FILAMENT             "Total de Extrusion"
+#else
 #define MSG_INFO_PRINT_COUNT                "Impresiones"
 #define MSG_INFO_COMPLETED_PRINTS           "Completadas"
 #define MSG_INFO_PRINT_TIME                 "Total"
 #define MSG_INFO_PRINT_LONGEST              "Mas larga"
 #define MSG_INFO_PRINT_FILAMENT             "Extrusion"
+#endif
 #define MSG_INFO_MIN_TEMP                   "Temperatura minima"
 #define MSG_INFO_MAX_TEMP                   "Temperatura maxima"
 #define MSG_INFO_PSU                        "Fuente de poder"
```

boelle, 2016-12-12:
why not submit a PR to RCBugfix?

yhfudev, 2016-12-12:
Sorry, I submit it just now. see #5474

boelle, 2016-12-12:
:-D @thinkyhead will have a look at it within a few days

closed 2016-12-13 by thinkyhead