MarlinFirmware/Marlin #5477: [Potential Bug]K8200 max temp trigger at 50C

boelle, 2016-12-12:
using pronterface i can set the bed to 50C without any issue

but when i ask repetier host to print something on the K8200 i get an MAXTEMP trigger for the bed

configuration.h: http://pastebin.com/WWsiP2KZ

so it should not trigger as settings say 150. Any clues as what i should check

boelle, 2016-12-14:
@thinkyhead no rush, but do you have any idea why the printer behaves like this? what could i check to narrow it down? a little lost

thinkyhead, 2016-12-19:
Using the current `RCBugFix`?

boelle, 2016-12-22:
yep

boelle, 2017-01-07:
Made it work under repetier

i assume this is due to an undersized heatbed. It cant hold 50 degree C with a 15V 30A so my assumptions is that it can with a 24V psu of same or less rating

The stange homing issues is not present in repetier

I dont have the funds nor the lust at present to fiddle more with the printer. But next step will be to get a 24V psu and retry

closed 2017-01-08 by boelle