MarlinFirmware/Marlin #5480: #define TEMP_SENSOR_0 0

jmil, 2016-12-12:
Marlin has significant utility outside of heated extrusion; laser cutters and photopolymerization 3D printers and inkjet printers. We used to be able to turn off all temp_sensor_# values with:

Configuration.h:
#define TEMP_SENSOR_0 0
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 0

However, in RC branch, eg:
7bea5e5e5701de0b90b6c422c954337ce860bb0f
Marlin now throws all kinds of errors and won't compile due to SanityCheck.h, print_heaterstates(), etc.

There seems to be an underlying assumption that there will always be at least one temperature sensor. How can we make this conditional so that people that don't use heaters and temperature sensors can still easily use Marlin?

Please don't say to just put a 100 kOhm resistor at the temp sensor port. I want a real software solution please.

THANKS!!

--jmil

jmil, 2016-12-12:
Note also that we cannot compile Marlin with EXTRUDERS set to 0:

Configuration.h:
// This defines the number of extruders
// :[1, 2, 3, 4]
#define EXTRUDERS 1

Blue-Marlin, 2016-12-12:
The solution for the temperatures is - use the dummy sensors 999 or 998.

jmil, 2016-12-12:
Oh wow, thanks!!!! that did it. So good thank you.

closed 2016-12-12 by jmil