MarlinFirmware/Marlin #5482: MINTEMP limitation prevents starting at 0 or sub 0 temperatures.

AndKe, 2016-12-12:
I am using #define HEATER_0_MINTEMP 1
with temp sensor #define TEMP_SENSOR_0 1 (100k)

Problem is: the printer is in unheated storage (pretty automated)  so the room temperature during winter is often 0 or maybe less degree.
- so it goes into mintemp error. 

If I configure "-10" - which would solve the problem  - MINTEMP is never triggered.  It simply does not handle nagative temperatures.

-of course, I will not disable this great safety feature.
-heating the printing chamber before may solve it, (adding manual step) but still, the Marlin already stopper (and OctoPi have trouble communicating with it)
-the best solution would be to handle negative "mintemp"
-also: when mintemp message is displayed/sent , it should contain one important thing; the actual temperature that trigged the safety logic.

justmyopinion, 2016-12-12:
There is no negative definition in table for thermistor type 1, but you may try to add a few, values are interpolated buit not very accurate in that end and wwould need some caibration.

THIS IS END OF TABLE FOR TYPE 1:
...

 {       985*OVERSAMPLENR        ,       20      },
 {       993*OVERSAMPLENR        ,       15      },
 {       999*OVERSAMPLENR        ,       10      },
 {       1004*OVERSAMPLENR       ,       5       },
 {       1008*OVERSAMPLENR       ,       0       },
Try to add a few more to tabel:
 **_{       1012*OVERSAMPLENR       ,      -5       },
 {       1015*OVERSAMPLENR       ,      -10     }_** 

};
#endif

It may work but it is not tested

AndKe, 2016-12-12:
Thank you - that's interesting, I do not need any accuracy at the lower end of scale , just a difference between -10 degrees  and "sensor disconnected"
- will test.

3dGriff, 2016-12-13:
I also find myself running in cold weather, not sure how you're automating your setup but I've used an end-routine (M140) to maintain bed temp between prints as a solution in the past.

AndKe, 2016-12-13:
@justmyopinion 's suggestion worked perfectly fine.

Please pull https://github.com/MarlinFirmware/Marlin/pull/5485 , it allows user to use MINTEMP -14 degrees , and still trigger the safety shutdown when sensor is disconnected. (tested, and verified)

3dGriff, 2016-12-13:
Good to know.  Thanks!  Uploading

thinkyhead, 2016-12-13:
Merged code from #5485.

closed 2016-12-13 by thinkyhead