MarlinFirmware/Marlin #5484: Using ENDSTOP_INTERRUPTS_FEATURE on Mega + RAMPS

jrhuerta, 2016-12-13:
Hi @thinkyhead!

I've taken a look at the interrupt pins on the Mega 2560 and these are:
2, 3, 18, 19, 20, 21

Looking at the endstop pins in `pins_RAMPS.h`
```
//
// Limit Switches
//
#define X_MIN_PIN           3
#ifndef X_MAX_PIN
  #define X_MAX_PIN         2
#endif
#define Y_MIN_PIN          14
#define Y_MAX_PIN          15
#define Z_MIN_PIN          18
#define Z_MAX_PIN          19
```

if I swtich `Y_MIN_PIN` and `Z_MAX_PIN` and wire accordingly then I should be able to use this feature on my setup? Is this correct?

jrhuerta, 2016-12-13:
This was the output I got:
```
PINs causing interrups are:
D2
D3
D10
D11
D12
D13
D14
D15
D18
D19
D20
D21
D50
D51
D52
D53
A8 = D62
A9 = D63
A10 = D64
A11 = D65
A12 = D66
A13 = D67
A14 = D68
A15 = D69
Arduino pin numbering!
```
Pin 14 shows in there, does that mean that I don't even have to modify the `pins_RAMPS.h`?

Blue-Marlin, 2016-12-13:
It does mean the Arduino_pins.h for the Arduino Mega is incomplete and patched for some pins.
It does mean you don't have to alter your pins-file.

thinkyhead, 2016-12-13:
With the `PINS_DEBUGGING` feature the `M43` command prints info about the pins, though I don't recall at the moment if it marks the interrupt-able ones.

jbrazio, 2016-12-13:
@thinkyhead when `ENDSTOP_INTERRUPTS_FEATURE` do all pins have to be on interrupts or can we have a mixed scenario ?

Blue-Marlin, 2016-12-13:
Take a look into the code.
In a mixed scenario still all endstops would have to be tested - always.
The interrupts set a flag only, if to test the endstops the conventional way or not.

```
      // Not all used endstop/probe -pins can raise interrupts. Please deactivate ENDSTOP_INTERRUPTS or change the pin configuration!
      static_assert(digitalPinToPCICR(X_MAX_PIN) != NULL, "X_MAX_PIN is not interrupt-capable"); // if pin has no pin change interrupt - error
```
and so on.

thinkyhead, 2016-12-13:
That's correct. The savings is in not checking any until one triggers. It's doesn't derive as much savings if you still have to poll one or two.

closed 2017-10-31 by thinkyhead