MarlinFirmware/Marlin #5486: [Discussion] FWRetract with Z-Hop and pure Z-Moves inserted by Slicers

FHeilmann, 2016-12-13:
I recently ran into an issue with using FWRetract and Simplify3D with a post processing script.

In short, what happens if your retract using `G10`, perform a Z-Move and then unretract using `G11` the Z-Hop will be undone regardless of what the Z-Move did.

An example: 
Using 

- FWRetract with a non-zero Z-Hop
- Simplify3D with skirts enabled and no Z-Hop
- A search-and-replace script which replaces all `G1 E-xxx` retracts with `G10` and all `G1 Exxx` unretracts with `G11? , your G-Code will end up looking like this:

```G-code
[Start.gcode]
G10 ;this is always inserted by Simplify3D for whatever reason, afterwards we are at whatever Z the start code left the nozzle at plus z-hop distance
G1 Z[first_layer_height] ; after this, we are now at first layer height
[Skirt]
G10; no nozzle lift since the first retract was never undone
[Move to part]
G11; unretract which also undoes the z-hop. after this we end up at first-layer-height - z-hop
[Start first layer]
```

The first retract after the start code will cause a FWRetract and the associated nozzle lift. Simplify3D then moves to the first-layer-height which potentially undoes the nozzle lift done by the FWRetract. After printing the skirt, the requested G10 is ignored, since we didn't undo the first retract and, thus, are still in the retracted state. After moving to the part we un-retract, which will now cause the nozzle-lift to be undone, potentially driving the nozzle below Z0 or to Z0 if software endstops are active.

Of course, this is something that Slicers should avoid and I understand that this is not necessarily the responsibility of Marlin, however since FWretract, right now, is in a place where a lot of people rely on auto-detection or simple search-and-replace methods to get firmware controlled retracts I wanted to start this discussion about how to protect against situations like this.

Possible solutions for this problem:
- (specifically for Simplify3D) Add additional smarts to your replace script to remove the first `G10` which should avoid double G10 calls (my current solution)
- Modify Marlin such that if a Z-Move is issued in the retracted state, the Z-Hop will not be undone on the next unretract
- Modify Marlin such that if multiple calls to G10 are made, only the Extruder portion of the FWRetract is ignored and the Z-Hop will always be executed, even if already retracted.

manianac, 2016-12-13:
Is it possible one of your post-processing scripts are typo'd?  I use S3D with firmware retract, and checking my gcode its working properly.

First set S3D to have 1mm retract distance, no extra restart distance, no vertical lift, and 15mm/s retract speed.  These settings don't affect the print, they're only for the regex to catch.

Set your post processing commands to
``` gcode
{REPLACE "G1 E-1.0000 F900\n" "G10\n"}
{REPLACE "G1 E0.0000 F900\n" "G11\n"}
{REPLACE "G1 E0.0000 F270\n" "G11\n"}
```

So for my gcode I end up with
``` gcode
Initial setup.....
M109 S205 ; set hotend temp
G92 E0
G10
G1 Z0.240 F120
; layer 1, Z = 0.24
T0
; tool H0.240 W0.480
; skirt
G1 X109.640 Y110.209 F12000
G11
G92 E0
G1 X109.675 Y110.177 E0.0054 F1200
Printing........
```

FHeilmann, 2016-12-13:
What is your FWretract Z-Hop?

Looking at your G-Code you would have this issue also, but you may not see it if your Z-Hop is substantially lower than your first layer height. Suppose your Z-Hop was 0.2 Line 4 would perform the Nozzle lift, then you move to Z0.240 and then G11 would lower the nozzle and you would end up at Z0.04

The issue here is not catching all the retracts and unretracts with the regex. The issue is that simplify3d moves the Z-height after retracting and before unretracting.

thinkyhead, 2016-12-13:
@FHeilmann Your idea seems sound. If `G1 Z` occurs after `G10` Marlin should forget the z hop and not undo it with `G11`.

closed 2016-12-14 by thinkyhead