MarlinFirmware/Marlin #5489: Dual Steppers for each axis

forReason, 2016-12-13:
Dear Community, I have a Azteeg X3 Pro which allows for up to 8 Stepper Drives.

I build a Large Printer So I want to use two steppers for Each axis (X1/2; Y1/2; Z1/2; E1; E2)
Is this somehow possible? Repetier supports this since months.

thinkyhead, 2016-12-13:
If you have enough free stepper drivers on the board (let's call them `E2`, `E3`, and `E4`), you can use them to do dual X, dual Y, and dual Z. Just enable the options `X_DUAL_STEPPER_DRIVERS`, `Y_DUAL_STEPPER_DRIVERS`, and `Z_DUAL_STEPPER_DRIVERS`. Then attach X2 to E2. Attach Y2 to E3. Attach Z2 to E4.

You then have two steppers working together in parallel on each axis. The Dual Z axis supports an additional Z2 endstop.

forReason, 2016-12-13:
Thank you for the fast reply. There is the following Problem:
I have found the following in Configuration_adv.h
`#ifdef Z_DUAL_STEPPER_DRIVERS
  #undef EXTRUDERS
  #define EXTRUDERS 1
#endif

// Same again but for Y Axis.
//#define Y_DUAL_STEPPER_DRIVERS

// Define if the two Y drives need to rotate in opposite directions
#define INVERT_Y2_VS_Y_DIR true

#ifdef Y_DUAL_STEPPER_DRIVERS
  #undef EXTRUDERS
  #define EXTRUDERS 1
#endif
`

First of all I cannot Find any 'X_DUAL_STEPPER_DRIVERS' second the following section gives me headaches:
`#if defined (Z_DUAL_STEPPER_DRIVERS) && defined (Y_DUAL_STEPPER_DRIVERS)
  #error "You cannot have dual drivers for both Y and Z"
#endif`

Can i just uncomment the last "error" section or am I totally wron in the first place?

Kind Regards

thinkyhead, 2016-12-13:
I suggest you start with `RCBugFix`. You're using an older version of Marlin which lacks the support you're looking for.

Kaibob2, 2016-12-13:
I suppose you're on an older Marlin version as this is supported in the most recent `RCBugFix `
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration_adv.h#L252

forReason, 2016-12-13:
I used the latest Release `[1.0.2-2](https://github.com/MarlinFirmware/Marlin/releases/tag/1.0.2-2)` from 12 days ago. Should I wait until 1.10 gets a stable release or is the latest prerelease "stable"?

Kaibob2, 2016-12-13:
1.0.2-2 is extremely outdated. thinkyhead only fixed one [single thing in there.](https://github.com/MarlinFirmware/Marlin/releases/tag/1.0.2-2)
You can safely use [RC8](https://github.com/MarlinFirmware/Marlin/tree/RC) but better use the [BugFix version](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix) as there are already a few things fixed.

forReason, 2016-12-13:
Thank you very much

closed 2016-12-13 by forReason