MarlinFirmware/Marlin #5496: [QUESTION] Redundant E entries in LCD Control Menu

Kaibob2, 2016-12-13:
Are these lines
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/ultralcd.cpp#L2024
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/ultralcd.cpp#L2048
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/ultralcd.cpp#L2072

necessary for something? When `DISTINCT_E_FACTORS` is enabled there are three E entries which makes it a little confusing what to adjust.

[image]

Sebastianv650, 2016-12-13:
If you have more than one E stepper, there are also more than one max. acceleration values. One for each stepper.

Kaibob2, 2016-12-13:
I know but do we need:
E
E1
E2

What is E (without a number) in this scenario

Kaibob2, 2016-12-13:
Ah, just found out. It's the active extruder. When i'm printing with E0 and change `E` than only `E1` changes.
Hmm, i would prefer if there is only E1 and E2 (in a two extruder scenario) as i think everybody knows which is the active extruder if changing values during operation.
But maybe this has a deeper meaning i don't get right know.

Kaibob2, 2016-12-14:
What i mean is:
Can't these three lines just be deleted to make it more clear which value you're editing. I suppose nobody would miss them...

thinkyhead, 2016-12-14:
Another UI concept, when there are multiple E steppers, is to place the active E first in the list, then the others in order, so if the second extruder was active in a 3-extruder system…
```
A max Z:       100
A max E2:    10000
A max E1:    10000
A max E3:    10000
A Retract:    3000
```

Kaibob2, 2016-12-14:
@thinkyhead I would prefer the second option
```
A max Z:       100
A max E1:    10000
A max E2:   *10000
A max E3:    10000
A Retract:    3000
```
The first option makes it confusing and the third would imply that A max E2 greater 10000

But anyway, my intention was, that there are not (for example) three E entries in a two extruder scenario.
If it would be just like this
```
A max Z:       100
A max E1:   10000
A max E2:   10000
A Retract:    3000
```
i think it would be easy to understand as i don't see the advantage of seeing which extruder is active in the display. If i'm printing and there is coming filament from E1 and i want to change parameters for the active printhead i know where to change it...everywhere where an E1 appears.

closed 2017-07-26 by Kaibob2